OP-TEE/build #332: Customize with buildroot to run Fortran code

gperezalbaUGR, 2019-01-14:
Hi,
I'm trying to run a fortran file `file_name.f` in OP-TEE. Are these the steps to do it?:
1 - Add a new `BR2_TOOLCHAIN_HAS_FORTRAN` line as you say in https://github.com/OP-TEE/optee_os/issues/2632#issuecomment-436268681 and re-run make.
2 - Do I have to copy the `file_name.f` into the SD or do I have to copy the compiled (compiled in my PC) `file_name.f` ? (Maybe the right way is cross-compile it in my PC and copy the file. Sorry I'm new with this)
3 - Where do I have to place the file from step 2?.

Thanks in advance

jbech-linaro, 2019-01-31:
> I'm trying to run a fortran file `file_name.f` in OP-TEE. Are these the steps to do it?:
> 1 - Add a new `BR2_TOOLCHAIN_HAS_FORTRAN` line as you say in [OP-TEE/optee_os#2632 (comment)](https://github.com/OP-TEE/optee_os/issues/2632#issuecomment-436268681) and re-run make.

Yes

> 2 - Do I have to copy the `file_name.f` into the SD or do I have to copy the compiled (compiled in my PC) `file_name.f` ? (Maybe the right way is cross-compile it in my PC and copy the file. Sorry I'm new with this)

I have no experience with Fortran, but if that is the Fortran binary, then you need to move it to the root fs. How? Depending on what device you are using (Qemu, HiKey, FVP  ... etc).

> 3 - Where exactly in the SD do I have to place the file from step 2?.

Cannot answer, since I don't know how to run Fortran files.

jbech-linaro, 2019-08-08:
I'm closing the ticket, either because it has already been answered or that it is no longer relevant or it could be lack of response from the author. Having that said, feel free to re-open the ticket if you have more to add to the ticket.

closed 2019-08-08 by jbech-linaro