OP-TEE/build #335: tftp instructions lost from rpi3.md

msa2, 2019-01-21:
From page

  https://github.com/OP-TEE/build

following the rpi3.md link

https://github.com/OP-TEE/build/blob/master/docs/rpi3.md

the section about NFS boot has  lost  instructions about setting up TFTP boot (it used to be there)

However, the standard uboot.env created by the build process, uses also TFTP when you choose 'nsfboot' (using environment symbol 'tftpserverip' for both tftp and nfs (not 'serverip' as in documentation).

jbech-linaro, 2019-01-21:
In the past we loaded 
U-Boot -> TF-A -> OP-TEE -> Kernel
but a couple of months ago it was updated, so now it is
TF-A -> OP-TEE -> U-Boot -> Kernel

Since we cannot load TF-A and OP-TEE directly with U-Boot (without having to fatwrite to the SD-card), we decided to just get rid of the TFTP part in the instructions and just keep the NFS part.

Regarding `tftpserverip` and `serverip`. We're actually just using `nfsserverip` which should be seen just as a local variable (see https://github.com/OP-TEE/build/blob/master/rpi3/firmware/uboot.env.txt). Documentation should be updated to reflect what is in uboot.env.txt).

msa2, 2019-01-21:
> In the past we loaded
> U-Boot -> TF-A -> OP-TEE -> Kernel
> but a couple of months ago it was updated, so now it is
> TF-A -> OP-TEE -> U-Boot -> Kernel

I've no clue what TF-A actually is. But, remember I'm using "-b 3.2.0" and there the old tftpboot works, and is very handy with symlinks to build tree. So this will not work in future any more, we always have to mess up with SD card after rebuild of optee.bin or linux kernel?

> Regarding `tftpserverip` and `serverip`. We're actually just using `nfsserverip` which should be seen just as a local variable

Ok, but in 'b 3.2.0', it still is 'tftpserverip'...

jbech-linaro, 2019-01-21:
> I've no clue what TF-A actually is.

Trusted Firmware A (used to be refered to as Arm Trusted Firmware, A-TF).

> So this will not work in future any more, we always have to mess up with SD card after rebuild of optee.bin or linux kernel?

Yes, I'm not very happy about this myself, but the boot is a proper boot now and that was the price we had to pay. You can update things as in the past from U-Boot (dchp load a file, then fatwrite to the corresponding address) ... **BUT** if there is a bug in either TF-A or OP-TEE that makes it stop booting, then you have to do the "SD dance" again. 

> Ok, but in 'b 3.2.0', it still is 'tftpserverip'...

Right, it was the last version who used this, probably legacy name used there.

igoropaniuk, 2019-01-21:
@jbech-linaro 

>> So this will not work in future any more, we always have to mess up with SD card after rebuild of optee.bin or linux kernel?

> Yes, I'm not very happy about this myself, but the boot is a proper boot now and that was the price we had to pay. You can update things as in the past from U-Boot (dchp load a file, then fatwrite [...]

We started using upstream U-boot/TF-A, that's the main advantage here, I think. You can still load Linux kernel via TFTP and mount rootfs via NFS, but not TF-A with OP-TEE.

If you want to avoid messing up with SD card, Wi-Fi-enabled SD card can be a good candidate for your case (for example, https://www.amazon.com/Transcend-Wi-Fi-Class-Memory-TS32GWSDHC10/dp/B00A659ILQ). For example, U-boot maintaner is using this setup based on Wi-FI SD cards in his CI lab (https://www.konsulko.com/developing-on-hardware-without-functional-networking/)

jforissier, 2019-01-21:
@igoropaniuk 
> If you want to avoid messing up with SD card, Wi-Fi-enabled SD card can be a good candidate

That is super-cool :+1: I wasn't aware these things existed :confused:

msa2, 2019-01-22:
Looks better, checked out the latest (not b 3.2.0). But, when trying to use
`
linaro-stretch-installer-20180416-89.tar.gz
`
and copying optee (tee-supplicant, etc) files from out-br/target/ directories, then

```
root@linaro-installer:~# ldd /usr/sbin/tee-supplicant 
/usr/sbin/tee-supplicant: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /usr/sbin/tee-supplicant)
	linux-vdso.so.1 (0x0000007fb3360000)
	libteec.so.1 => /usr/lib/libteec.so.1 (0x0000007fb3321000)
	libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000007fb32f5000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007fb31ab000)
	/lib/ld-linux-aarch64.so.1 (0x0000007fb3335000)

```
however, xtest seems to be ok
```

root@linaro-installer:~# ldd /usr/bin/xtest 
	linux-vdso.so.1 (0x0000007f7fb0b000)
	libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000007f7fab4000)
	libteec.so.1 => /usr/lib/libteec.so.1 (0x0000007f7faa0000)
	libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007f7f9f5000)
	libcrypto.so.1.0.0 => /usr/lib/libcrypto.so.1.0.0 (0x0000007f7f832000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007f7f6e8000)
	/lib/ld-linux-aarch64.so.1 (0x0000007f7fae0000)
	libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000007f7f6d5000)

```

jforissier, 2019-01-22:
@msa2 

> /usr/sbin/tee-supplicant: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /usr/sbin/tee-supplicant)

This means that `tee-supplicant` was linked against a more recent version of libc than the one that is found at runtime. In other words, `libc.so` found in the toolchain is more recent than the one in `linaro-stretch-installer-20180416-89.tar.gz`.

IMO, the easiest fix is to use an older cross-compiler. First identify which version of libc comes with the Stretch distribution, then find the most recent cross compiler that has this version or older. Since identifying versions could be a bit tricky, you may as well just try your luck with increasingly older compilers. I would recommend 7.3 from https://releases.linaro.org/components/toolchain/binaries/ as a first step.

HTH.

msa2, 2019-01-22:
> IMO, the easiest fix is to use an older cross-compiler. First identify which version of libc comes with the Stretch distribution, then find the most recent cross compiler that has this version or [...]

Well, I tried another approach: log into rpi3 and
```

apt install build-essential
apt install cmake
... copy optee-client directory into linaro home account
make clean
cmake .
make
make install
tee-supplicant &
xtest
```
Took a while with apt installs, but compile was reasonably quick and did something; xtest gives in the end...

```
....
regression_2003 OK
regression_2004 OK
+-----------------------------------------------------
23304 subtests of which 0 failed
93 test cases of which 0 failed
0 test cases were skipped
TEE test application done!
```

Not sure I trust the result, but looks promising...

jforissier, 2019-01-23:
> Well, I tried another approach: log into rpi3 and
[....]

That's fine. Adds more manual steps to the build, though...

> Not sure I trust the result, but looks promising...

Why not? Seems to easy? :wink:

msa2, 2019-01-23:
> That's fine. Adds more manual steps to the build, though...

I didn't want to mess up with the build process from "upstream", it's complex enough already as is. I hope my "workaround" is just temporary, you do provide a new updated linaro-stretch rootfs eventually, I hope....?

jbech-linaro, 2019-01-23:
> I didn't want to mess up with the build process from "upstream", it's complex enough already as is. I hope my "workaround" is just temporary, you do provide a new updated linaro-stretch rootfs eventually, I hope....?

We will do something at least in the near future, since for RPi3 we have many questions almost weekly. For the core team we usually don't see the problems, but due to the amount of questions it's obviously hard to use this. So stay tuned!

jbech-linaro, 2019-01-23:
Right, one of the things that will come in future instructions is that we will say that with our OP-TEE instructions it is only possible to use a UART (cable), i.e., no X. In general be more clear what is expected to work and what is not tested/supported. This will limit possibilities and the scope, but we're a small team and we don't have enough bandwidth to support and maintain every possible combination for all sorts for devices (> 30 today), root files system (busybox, buildroot, linaro-stretch, AOSP, OpenEmbedded etc), build environments (Linux, AOSP, OpenEmbedded etc). When we published OP-TEE ~5 years ago that was possible, but now it's simply too much for us to handle on our own. Having that said, you don't have to be a Linaro guy or connected to Linaro to be a maintainer for a device. So ... if you have time and interest .... :smile:

msa2, 2019-01-28:
This issue could be closed. The instructions match the latest release for NFS setup. I can live without TFTP (although the SD shuffle in trying to test Trusted Boot is a bit tiresome :-)

jbech-linaro, 2019-01-28:
> although the SD shuffle in trying to test Trusted Boot is a bit tiresome

Cannot more than agree! Let's see if we can figure out something clever in the future that makes it less painful (again) to work with RPi3! Thanks!

closed 2019-01-28 by jbech-linaro