OP-TEE/build #338: "Trusted" boot on rpi3? Missing build instructions?

msa2, 2019-01-25:
I see some hints that there is/was some work done for a boot sequence, which includes verification by certificates for all loaded components. From console output is seems that toolchain builds components starting from "BL1" below. So, I'm looking for modification where BL1 and all subsequent modules use some certificate (possibly compiled into the code) to verify each file the load.

**However, I cannot find any instructions how to enable this. Is it possible or not?**

That I want to have trust chain from BL1 forward ready, and worry about verified BL1 as a separate issue (maybe on different hardware platform). And, I suppose, for u-boot it would mean using FIT with signatures?

```
NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v2.0(debug):v2.0-580-g023bc019
NOTICE:  BL1: Built : 09:49:08, Jan 22 2019
INFO:    BL1: RAM 0x100ee000 - 0x100f7000
INFO:    BL1: cortex_a53: CPU workaround for 843419 was applied
INFO:    BL1: cortex_a53: CPU workaround for 855873 was applied
NOTICE:  rpi3: Detected: Raspberry Pi 3 Model B (1GB, Sony, UK) [0x00a02082]
INFO:    BL1: Loading BL2
INFO:    Loading image id=1 at address 0x100b4000
INFO:    Image id=1 loaded: 0x100b4000 - 0x100b9410
NOTICE:  BL1: Booting BL2
INFO:    Entry point address = 0x100b4000
INFO:    SPSR = 0x3c5
NOTICE:  BL2: v2.0(debug):v2.0-580-g023bc019
NOTICE:  BL2: Built : 09:49:08, Jan 22 2019
INFO:    BL2: Doing platform setup
INFO:    BL2: Loading image id 3
INFO:    Loading image id=3 at address 0x100e0000
INFO:    Image id=3 loaded: 0x100e0000 - 0x100e9078
INFO:    BL2: Loading image id 4
INFO:    Loading image id=4 at address 0x10100000
INFO:    Image id=4 loaded: 0x10100000 - 0x1010001c
INFO:    OPTEE ep=0x10100000
INFO:    OPTEE header info:
INFO:          magic=0x4554504f
INFO:          version=0x2
INFO:          arch=0x1
INFO:          flags=0x0
INFO:          nb_images=0x1
INFO:    BL2: Loading image id 21
INFO:    Loading image id=21 at address 0x10100000
INFO:    Image id=21 loaded: 0x10100000 - 0x101453e0
INFO:    BL2: Skip loading image id 22
INFO:    BL2: Loading image id 5
INFO:    Loading image id=5 at address 0x11000000
INFO:    Image id=5 loaded: 0x11000000 - 0x110764a0
NOTICE:  BL1: Booting BL31
INFO:    Entry point address = 0x100e0000
INFO:    SPSR = 0x3cd
NOTICE:  BL31: v2.0(debug):v2.0-580-g023bc019
NOTICE:  BL31: Built : 09:49:08, Jan 22 2019
INFO:    rpi3: Checking DTB...
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for 843419 was applied
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied
INFO:    BL31: Initializing BL32

```

igoropaniuk, 2019-01-25:
Hi @msa2,

Just some notice in advance, that before TF-A BL1 you won't be able to get a chain of trust, as no RPi3 bootrom nor VPU-Side Bootloader (`bootcode.bin`) support this, so I assume you're working on this setup just for PoC.

Depending the boot stage:

1. SW-side initialization/booting (TF-A BL1->BL2->BL31, BL32(OP-TEE), BL33) up to U-boot (BL33. the first non-secure bootloader), you can use TBB support in TF-A (https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/trusted-board-boot.rst)

2. Starting from U-boot, you can sign and put Linux kernel, dtbs/dtb overlays. initial ramdisk etc.  into the FIT image

3. Linux kernel -> userspace (integrity checking of filesystem), you can map your rootfs via `device-mapper` and use dm-verity feature in the Linux kernel (https://www.kernel.org/doc/Documentation/device-mapper/verity.txt). Unfortunately, BCM2837B0 is missing ARM ISA’s Crypto Extensions, so performance degradation is expected.

msa2, 2019-01-25:
Thanks, I will try to digest this....

... so I cannot just put some magic "-DTRUSTED_BOARD_BOOT" in some global makefile in optee build?

The "./arm-trusted-firmware" is then not same as 

https://github.com/ARM-software/arm-trusted-firmware

If not, do you know if your build process will work, if I replace 'arm-trusted-firmware' from above git repo?

jbech-linaro, 2019-01-25:
It's a long time since we tried it (commit says: Sep 21, 2016), but what the engineer did back then to use this with OP-TEE was:
https://github.com/OP-TEE/optee_os/blob/master/documentation/optee_with_auth_framework.md

igoropaniuk, 2019-01-25:
@msa2

We've been using mainline TF-A for RPi3 setup (https://github.com/ARM-software/arm-trusted-firmware) for a while (AFAIK since 3.3.0 release), so if you just fetch the latest master of TF-A, it's expected that it will work out-of-the-box.

msa2, 2019-01-28:
I replaced the optee version of arm-trusted-firmware
`
git clone https://github.com/ARM-software/arm-trusted-firmware.git
`
It seemed to compile and work as before, so I proceeded to put some additional flags into build/Makefile `ARM_TF_FLAGS`:
```

	PLAT=rpi3 \
	MBEDTLS_DIR=$(OPTEE_PATH)/lib/libmbedtls/mbedtls/library
	TRUSTED_BOARD_BOOT=1 \
	GENERATE_COT=1 \
	ARM_ROTPK_LOCATION=devel_rsa \
	ROT_KEY=$(ARM_TF_PATH)/plat/arm/board/common/rotpk/arm_rotprivk_rsa.pem \
	RPI3_PRELOADED_DTB_BASE=0x00010000 \
	SPD=opteed

```
But, looks like I lost the optee-os in the process...

```
Terminal ready
NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v2.0(debug):v2.0-688-g7ca572d9
NOTICE:  BL1: Built : 11:12:09, Jan 28 2019
INFO:    BL1: RAM 0x100ee000 - 0x100f7000
INFO:    BL1: cortex_a53: CPU workaround for 843419 was applied
INFO:    BL1: cortex_a53: CPU workaround for 855873 was applied
NOTICE:  rpi3: Detected: Raspberry Pi 3 Model B (1GB, Sony, UK) [0x00a02082]
INFO:    BL1: Loading BL2
INFO:    Loading image id=1 at address 0x100b4000
INFO:    Image id=1 loaded: 0x100b4000 - 0x100b91e8
NOTICE:  BL1: Booting BL2
INFO:    Entry point address = 0x100b4000
INFO:    SPSR = 0x3c5
NOTICE:  BL2: v2.0(debug):v2.0-688-g7ca572d9
NOTICE:  BL2: Built : 11:12:10, Jan 28 2019
INFO:    BL2: Doing platform setup
INFO:    BL2: Loading image id 3
INFO:    Loading image id=3 at address 0x100e0000
INFO:    Image id=3 loaded: 0x100e0000 - 0x100e7078
INFO:    BL2: Loading image id 5
INFO:    Loading image id=5 at address 0x11000000
INFO:    Image id=5 loaded: 0x11000000 - 0x110764a0
NOTICE:  BL1: Booting BL31
INFO:    Entry point address = 0x100e0000
INFO:    SPSR = 0x3cd
NOTICE:  BL31: v2.0(debug):v2.0-688-g7ca572d9
NOTICE:  BL31: Built : 11:12:13, Jan 28 2019
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for 843419 was applied
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x11000000
INFO:    SPSR = 0x3c9

U-Boot 2019.01-00093-gaac0c29d4b (Jan 28 2019 - 11:12:01 +0200)

DRAM:  944 MiB
RPI 3 Model B (0xa02082)
MMC:   mmc@7e202000: 0, sdhci@7e300000: 1
Loading Environment from FAT... OK
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
starting USB...
USB0:   scanning bus 0 for devices... 3 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0 

```

igoropaniuk, 2019-01-28:
@msa2

Could you please share the full diff

msa2, 2019-01-28:
I checked out latest optee, and there was some problem when I tried to use the existing mbedtls in optee tree, so I checked out 
`
https://github.com/ARMmbed/mbedtls.git
`
above root. Made following additions to build/Makefile (my previous changes were actually faulty, lacked "\"):

```
diff --git a/rpi3.mk b/rpi3.mk
index 71abfac..120329a 100644
--- a/rpi3.mk
+++ b/rpi3.mk
@@ -66,6 +66,11 @@ ARM_TF_FLAGS ?= \
        CRASH_REPORTING=1 \
        LOG_LEVEL=40 \
        PLAT=rpi3 \
+       MBEDTLS_DIR=$(ROOT)/../mbedtls \
+       TRUSTED_BOARD_BOOT=1 \
+       GENERATE_COT=1 \
+       ARM_ROTPK_LOCATION=devel_rsa \
+       ROT_KEY=$(ARM_TF_PATH)/plat/arm/board/common/rotpk/arm_rotprivk_rsa.pem \
        RPI3_PRELOADED_DTB_BASE=0x00010000 \
        SPD=opteed

```
... after long build process and copy to SD, the boot process hangs:

```
Terminal ready
NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v2.0(debug):v2.0-580-g023bc019
NOTICE:  BL1: Built : 14:06:09, Jan 28 2019
INFO:    BL1: RAM 0x100ee000 - 0x100fa000
INFO:    BL1: cortex_a53: CPU workaround for 843419 was applied
INFO:    BL1: cortex_a53: CPU workaround for 855873 was applied
INFO:    Using crypto library 'mbed TLS'
NOTICE:  rpi3: Detected: Raspberry Pi 3 Model B (1GB, Sony, UK) [0x00a02082]
INFO:    BL1: Loading BL2
INFO:    Loading image id=6 at address 0x100b4000
INFO:    Image id=6 loaded: 0x100b4000 - 0x100b446f

```
...if this is the way it reports certificate check fail, it would be nice if it said so clearly. Anyway, I will continue checking, maybe I should start from totally clean again.

jbech-linaro, 2019-01-28:
@msa2 when I'm trying to figure out what is happening in TF-A code, then I usually also build with `LOG_LEVEL=50`, that gives lots of more debug prints. So it could be worth change that also in your build arguments.

msa2, 2019-01-28:
Thanks, changed LOG_LEVEL=50, got huge amount of log output, and it also booted to linux. Small extract of the output...

```
....
    [LV2] VA:0x3f800000 PA:0x3f800000 size:0x200000 DEV-RW-XN-PRIV-S
    [LV2] VA:0x3fa00000 PA:0x3fa00000 size:0x200000 DEV-RW-XN-PRIV-S
    [LV2] VA:0x3fc00000 PA:0x3fc00000 size:0x200000 DEV-RW-XN-PRIV-S
    [LV2] VA:0x3fe00000 PA:0x3fe00000 size:0x200000 DEV-RW-XN-PRIV-S
  [LV1] VA:0x40000000 size:0x40000000
  [LV1] (2 invalid descriptors omitted)
NOTICE:  BL2: v2.0(debug):v2.0-580-g023bc019
NOTICE:  BL2: Built : 15:13:47, Jan 28 2019
INFO:    Using crypto library 'mbed TLS'
INFO:    BL2: Doing platform setup
INFO:    BL2: Loading image id 3
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=7 at address 0x100e0000
INFO:    Image id=7 loaded: 0x100e0000 - 0x100e060e
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=9 at address 0x100e0000
INFO:    Image id=9 loaded: 0x100e0000 - 0x100e04da
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=13 at address 0x100e0000
INFO:    Image id=13 loaded: 0x100e0000 - 0x100e0430
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=3 at address 0x100e0000
INFO:    Image id=3 loaded: 0x100e0000 - 0x100eb078
INFO:    BL2: Loading image id 4
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=10 at address 0x10100000
INFO:    Image id=10 loaded: 0x10100000 - 0x101004e8
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=14 at address 0x10100000
INFO:    Image id=14 loaded: 0x10100000 - 0x101004ce
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=4 at address 0x10100000
INFO:    Image id=4 loaded: 0x10100000 - 0x1010001c
INFO:    OPTEE ep=0x10100000
INFO:    OPTEE header info:
INFO:          magic=0x4554504f
INFO:          version=0x2
INFO:          arch=0x1
INFO:          flags=0x0
INFO:          nb_images=0x1
INFO:    BL2: Loading image id 21
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=21 at address 0x10100000
INFO:    Image id=21 loaded: 0x10100000 - 0x101453e0
INFO:    BL2: Skip loading image id 22
INFO:    BL2: Loading image id 5
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=11 at address 0x11000000
INFO:    Image id=11 loaded: 0x11000000 - 0x110004ea
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=15 at address 0x11000000
INFO:    Image id=15 loaded: 0x11000000 - 0x11000440
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=5 at address 0x11000000
INFO:    Image id=5 loaded: 0x11000000 - 0x110764a0
NOTICE:  BL1: Booting BL31
INFO:    Entry point address = 0x100e0000
INFO:    SPSR = 0x3cd
VERBOSE: Argument #0 = 0x100ca650
VERBOSE: Argument #1 = 0xf1e2d3c4b5a6978
VERBOSE: Argument #2 = 0x0
VERBOSE: Argument #3 = 0x0
VERBOSE: Argument #4 = 0x0
VERBOSE: Argument #5 = 0x0
VERBOSE: Argument #6 = 0x0
VERBOSE: Argument #7 = 0x0
VERBOSE: Trusted SRAM seen by this BL image: 0x100e0000 - 0x100f6000
VERBOSE: Code region: 0x100e0000 - 0x100e8000
VERBOSE: Read-only data region: 0x100e8000 - 0x100eb000
VERBOSE: Coherent region: 0x100f5000 - 0x100f6000
mmap:
 VA:0x10000  PA:0x10000  size:0x10000  attr:0x1a  granularity:0x40000000
 VA:0x10000000  PA:0x10000000  size:0x1000  attr:0x8  granularity:0x40000000
 VA:0x100e0000  PA:0x100e0000  size:0x8000  attr:0x2  granularity:0x40000000
 VA:0x100e8000  PA:0x100e8000  size:0x3000  attr:0x22  granularity:0x40000000
 VA:0x100f5000  PA:0x100f5000  size:0x1000  attr:0x8  granularity:0x40000000
 VA:0x100e0000  PA:0x100e0000  size:0x16000  attr:0xa  granularity:0x40000000
 VA:0x10100000  PA:0x10100000  size:0xf00000  attr:0xa  granularity:0x40000000
 VA:0x3f000000  PA:0x3f000000  size:0x1000000  attr:0x8  granularity:0x40000000

VERBOSE: Translation tables state:
VERBOSE:   Xlat regime:     EL3
VERBOSE:   Max allowed PA:  0xffffffff
VERBOSE:   Max allowed VA:  0xffffffff
VERBOSE:   Max mapped PA:   0x3fffffff
VERBOSE:   Max mapped VA:   0x3fffffff
VERBOSE:   Initial lookup level: 1
VERBOSE:   Entries @initial lookup level: 4
VERBOSE:   Used 3 sub-tables out of 4 (spare: 1)
  [LV1] VA:0x0 size:0x40000000
    [LV2] VA:0x0 size:0x200000
      [LV3] VA:0x0 size:0x1000
      [LV3] (15 invalid descriptors omitted)
      [LV3] VA:0x10000 PA:0x10000 size:0x1000 MEM-RW-XN-NS
      [LV3] VA:0x11000 PA:0x11000 size:0x1000 MEM-RW-XN-NS
      [LV3] VA:0x12000 PA:0x12000 size:0x1000 MEM-RW-XN-NS
.....
```
Maybe I had some error in copying the image first time, or adding more debug makes it work...?

Running xtest -- passed all.

Next step? Tamper with something and see if it spots it?

msa2, 2019-01-28:
Oh, this is what the log says about BL1 loading...
```
    [LV2] VA:0x3fc00000 PA:0x3fc00000 size:0x200000 DEV-RW-XN-S
    [LV2] VA:0x3fe00000 PA:0x3fe00000 size:0x200000 DEV-RW-XN-S
  [LV1] VA:0x40000000 size:0x40000000
  [LV1] (2 invalid descriptors omitted)
NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v2.0(debug):v2.0-580-g023bc019
NOTICE:  BL1: Built : 15:13:45, Jan 28 2019
INFO:    BL1: RAM 0x100ee000 - 0x100fa000
VERBOSE: BL1: cortex_a53: CPU workaround for 826319 was not applied
VERBOSE: BL1: cortex_a53: CPU workaround for 835769 was not applied
VERBOSE: BL1: cortex_a53: CPU workaround for disable_non_temporal_hint was not applied
INFO:    BL1: cortex_a53: CPU workaround for 843419 was applied
INFO:    BL1: cortex_a53: CPU workaround for 855873 was applied
INFO:    Using crypto library 'mbed TLS'
VERBOSE: rpi3: mbox: Sending request at 0x100f2420
NOTICE:  rpi3: Detected: Raspberry Pi 3 Model B (1GB, Sony, UK) [0x00a02082]
INFO:    BL1: Loading BL2
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=6 at address 0x100b4000
INFO:    Image id=6 loaded: 0x100b4000 - 0x100b446f
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=1 at address 0x100b4000
INFO:    Image id=1 loaded: 0x100b4000 - 0x100c65e0
VERBOSE: BL1: BL2 memory layout address = 0x10001000
NOTICE:  BL1: Booting BL2
INFO:    Entry point address = 0x100b4000
INFO:    SPSR = 0x3c5
VERBOSE: Argument #0 = 0x0
VERBOSE: Argument #1 = 0x10001000
VERBOSE: Argument #2 = 0x0
VERBOSE: Argument #3 = 0x0
VERBOSE: Argument #4 = 0x0
VERBOSE: Argument #5 = 0x0
VERBOSE: Argument #6 = 0x0
VERBOSE: Argument #7 = 0x0
VERBOSE: Trusted SRAM seen by this BL image: 0x10001000 - 0x100ee000
VERBOSE: Code region: 0x100b4000 - 0x100c1000
VERBOSE: Read-only data region: 0x100c1000 - 0x100c6000
VERBOSE: Coherent region: 0x100d1000 - 0x100d1000
mmap:
 VA:0x20000  PA:0x20000  size:0x1e0000  attr:0x12  granularity:0x40000000

```

msa2, 2019-01-28:
Actually, next step should be getting the u-boot to verify linux kernel? I assume it is not done by the configuration so far?

msa2, 2019-01-28:
Actually, I'm not quite sure what I achieved... from Makefile `BL33=$(U-BOOT_BIN)`, I would guess the u-boot is BL33. Hower, in console log I see no mention of loading and checking of BL33. The last entry before u-boot is...
```

D/TC:0 0 check_ta_store:533 TA store: "Secure Storage TA"
D/TC:0 0 check_ta_store:533 TA store: "REE"
D/TC:0 0 mobj_mapped_shm_init:709 Shared memory address range: 11600000, 13600000
I/TC: Initialized
D/TC:0 0 init_primary_helper:1033 Primary CPU switching to normal world boot
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x11000000
INFO:    SPSR = 0x3c9
VERBOSE: Argument #0 = 0x0
VERBOSE: Argument #1 = 0x0
VERBOSE: Argument #2 = 0x0
VERBOSE: Argument #3 = 0x0
VERBOSE: Argument #4 = 0x0
VERBOSE: Argument #5 = 0x0
VERBOSE: Argument #6 = 0x0
VERBOSE: Argument #7 = 0x0

U-Boot 2019.01-00093-gaac0c29d4b (Jan 28 2019 - 15:13:35 +0200)

DRAM:  944 MiB
RPI 3 Model B (0xa02082)

```

msa2, 2019-01-29:
Just to verify that I don't do work that someone has already done...

Optee build does not have any ready support for enabling u-boot linux image verification?

I found one link
https://blog.crysys.hu/2018/06/verified-boot-on-the-raspberry-pi/
which is for non-TF-A version.  Instructions need to be tweaked for TF-A version, but is this pretty much what I would have to do? Design and write my own kernel.its file containing entries for kernel8 and bcm2710-rpi-3-b.dtb, and build kernel8.fit (or something?).

jbech-linaro, 2019-01-29:
> Just to verify that I don't do work that someone has already done...
> 
> Optee build does not have any ready support for enabling u-boot linux image verification?

True, I cannot remember anyone that has been doing that. The closest we did related to this was done ~5 years ago and is still at the old Linaro wiki: https://wiki.linaro.org/WorkingGroups/Security/Verified-U-boot. However, due to the age, it's probably very much out of date. So it's probably wise to not follow that blindly.

msa2, 2019-01-29:
Puzzled now, replacing replacing "kernel8.img" with "image.fit" seems to cause boot to hang on "Loading BL2"? But, after I increased boot partition size, and left kernel8.img along with image.fit, it does boot to u-boot? I thought "kernel8.img" was only loaded by u-boot?

I'm also unsure about the address I should use for the image.fit, but loading it does something...
I suppose, tinker with CONFIG_SYS_BOOTM_LEN next?
```
U-Boot> run nfsboot                                                                                   
15098986 bytes read in 624 ms (23.1 MiB/s)
## Loading kernel from FIT Image at 02000000 ...
   Using 'config-1' configuration
   Verifying Hash Integrity ... OK
   Trying 'kernel-1' kernel subimage
     Description:  default kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x020000c0
     Data Size:    15071744 Bytes = 14.4 MiB
     Architecture: AArch64
     OS:           Linux
     Load Address: 0x00080000
     Entry Point:  0x00080000
     Hash algo:    sha1
     Hash value:   d100cea6cb0a51efde88f859929ad83cb4606418
   Verifying Hash Integrity ... sha1+ OK
## Loading fdt from FIT Image at 02000000 ...
   Using 'config-1' configuration
   Verifying Hash Integrity ... OK
   Trying 'fdt-1' fdt subimage
     Description:  device tree
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x02e5fba8
     Data Size:    25374 Bytes = 24.8 KiB
     Architecture: AArch64
     Hash algo:    sha1
     Hash value:   a0e703e76371dbd7076f99a4fa63fa0b1c89a20e
   Verifying Hash Integrity ... sha1+ OK
   Booting using the fdt blob at 0x2e5fba8
   Loading Kernel Image ... Image too large: increase CONFIG_SYS_BOOTM_LEN
Must RESET board to recover

```

msa2, 2019-01-29:
...also, doubtfull about what should the load/entry addresses be in the .its file (now just copied 0x00080000 from the reference blog page).

msa2, 2019-01-29:
After setting CONFIG_SYS_BOOTM_LEN
```
diff --git a/include/configs/rpi.h b/include/configs/rpi.h
index 9ce41767a9..81f1f24c4c 100644
--- a/include/configs/rpi.h
+++ b/include/configs/rpi.h
@@ -81,7 +81,7 @@
 #define CONFIG_SETUP_MEMORY_TAGS
 #define CONFIG_CMDLINE_TAG
 #define CONFIG_INITRD_TAG
-
+#define CONFIG_SYS_BOOTM_LEN (16 << 20)
 /* Environment */
 #define ENV_DEVICE_SETTINGS \
        "stdin=serial,usbkbd\0" \

```
...it seems to be verifying and booting...
```
U-Boot> boot
15098986 bytes read in 628 ms (22.9 MiB/s)
## Loading kernel from FIT Image at 02000000 ...
   Using 'config-1' configuration
   Verifying Hash Integrity ... OK
   Trying 'kernel-1' kernel subimage
     Description:  default kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x020000c0
     Data Size:    15071744 Bytes = 14.4 MiB
     Architecture: AArch64
     OS:           Linux
     Load Address: 0x00080000
     Entry Point:  0x00080000
     Hash algo:    sha1
     Hash value:   d100cea6cb0a51efde88f859929ad83cb4606418
   Verifying Hash Integrity ... sha1+ OK
## Loading fdt from FIT Image at 02000000 ...
   Using 'config-1' configuration
   Verifying Hash Integrity ... OK
   Trying 'fdt-1' fdt subimage
     Description:  device tree
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x02e5fba8
     Data Size:    25374 Bytes = 24.8 KiB
     Architecture: AArch64
     Hash algo:    sha1
     Hash value:   a0e703e76371dbd7076f99a4fa63fa0b1c89a20e
   Verifying Hash Integrity ... sha1+ OK
   Booting using the fdt blob at 0x2e5fba8
   Loading Kernel Image ... OK
   reserving fdt memory region: addr=8000000 size=2000000
   reserving fdt memory region: addr=0 size=1000
   Loading Device Tree to 000000003ab36000, end 000000003ab3f31d ... OK

Starting kernel ...

I/TC: Dynamic shared memory is disabled
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.56-v8 (msa@kone2) (gcc version 8.2.1 20180802 (GNU Toolchain for the A-profile Architecture 8.2-2018-08 (arm-rel-8.23))) #2 SMP PREEMPT Mon Jan 28 14:41:29 EET 2019
[    0.000000] Boot CPU: AArch64 Processor [410fd034]
[    0.000000] Machine model: Raspberry Pi 3 Model B
```
I think I have all pieces now? Except the oddity of needing "kernel8.img" present...
My changes to build are now at this state:

```
diff --git a/kconfigs/u-boot_rpi3.conf b/kconfigs/u-boot_rpi3.conf
index 304f129..750e0d8 100644
--- a/kconfigs/u-boot_rpi3.conf
+++ b/kconfigs/u-boot_rpi3.conf
@@ -1 +1,5 @@
 CONFIG_SYS_TEXT_BASE=0x11000000
+CONFIG_FIT=y
+CONFIG_FIT_SIGNATURE=y
+CONFIG_RSA=y
+CONFIG_OF_CONTROL=y
diff --git a/rpi3.mk b/rpi3.mk
index 71abfac..aa5b4e8 100644
--- a/rpi3.mk
+++ b/rpi3.mk
@@ -64,8 +64,13 @@ ARM_TF_FLAGS ?= \
        DEBUG=1 \
        V=0 \
        CRASH_REPORTING=1 \
-       LOG_LEVEL=40 \
+       LOG_LEVEL=50 \
        PLAT=rpi3 \
+       MBEDTLS_DIR=$(ROOT)/../mbedtls \
+       TRUSTED_BOARD_BOOT=1 \
+       GENERATE_COT=1 \
+       ARM_ROTPK_LOCATION=devel_rsa \
+       ROT_KEY=$(ARM_TF_PATH)/plat/arm/board/common/rotpk/arm_rotprivk_rsa.pem \
        RPI3_PRELOADED_DTB_BASE=0x00010000 \
        SPD=opteed

```

igoropaniuk, 2019-01-29:
yeah, and try to remove this line also from `config.txt` (VC4 bootcode.bin expects that kernel8.img exists in the root of partition):
```
kernel_address=0x02000000
```

msa2, 2019-01-29:
I changed that line in u-boot to
`
load_kernel=fatload mmc 0:1 ${kernel_addr_r} image.fit
`
and in nfsboot, the last entry into
`
nfsboot=run load_kernel; run set_bootargs_tty set_bootargs_nfs set_common_args; bootm ${kernel_addr_r}
`
It boots just fine as is now....

msa2, 2019-01-29:
> yeah, and try to remove this line also from `config.txt` (VC4 bootcode.bin expects that kernel8.img exists in the root of partition):
> 
> ```
> kernel_address=0x02000000
> ```

Removed that, but still gets stuck when kernel8.img is not present. Or, do I need to edit something in build tree and rebuild?
This what happens without kernel8.img:
```
    [LV2] VA:0x3fc00000 PA:0x3fc00000 size:0x200000 DEV-RW-XN-S
    [LV2] VA:0x3fe00000 PA:0x3fe00000 size:0x200000 DEV-RW-XN-S
  [LV1] VA:0x40000000 size:0x40000000
  [LV1] (2 invalid descriptors omitted)
NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v2.0(debug):v2.0-580-g023bc019
NOTICE:  BL1: Built : 11:48:24, Jan 29 2019
INFO:    BL1: RAM 0x100ee000 - 0x100fa000
VERBOSE: BL1: cortex_a53: CPU workaround for 826319 was not applied
VERBOSE: BL1: cortex_a53: CPU workaround for 835769 was not applied
VERBOSE: BL1: cortex_a53: CPU workaround for disable_non_temporal_hint was not applied
INFO:    BL1: cortex_a53: CPU workaround for 843419 was applied
INFO:    BL1: cortex_a53: CPU workaround for 855873 was applied
INFO:    Using crypto library 'mbed TLS'
VERBOSE: rpi3: mbox: Sending request at 0x100f2420
NOTICE:  rpi3: Detected: Raspberry Pi 3 Model B (1GB, Sony, UK) [0x00a02082]
INFO:    BL1: Loading BL2

```

igoropaniuk, 2019-01-29:
@msa2 

Probably, non-existence of `kernel8.img` isn't a root cause here. I've renamed it to `kernel55.img` and removed the line from `config.txt` and managed to U-boot. In our case the booting process is stuck on TF-A BL2 loading.

Here `fatls mmc 0` output as a proof:

```
U-Boot> fatls mmc 0
            boot/
   958028   armstub8.bin
    25374   bcm2710-rpi-3-b.dtb
    50268   bootcode.bin
       45   config.txt
    18693   COPYING.linux
     2527   fixup_cd.dat
     6635   fixup.dat
     9770   fixup_db.dat
     9768   fixup_x.dat
     1494   LICENCE.broadcom
   641252   start_cd.elf
  4974116   start_db.elf
  2840612   start.elf
  3923364   start_x.elf
    16384   uboot.env
 15071744   kernel55.img

16 file(s), 1 dir(s)
```

msa2, 2019-01-30:
I get stuck on 
```

NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v2.0(debug):v2.0-580-g023bc019
NOTICE:  BL1: Built : 11:48:24, Jan 29 2019
INFO:    BL1: RAM 0x100ee000 - 0x100fa000
VERBOSE: BL1: cortex_a53: CPU workaround for 826319 was not applied
VERBOSE: BL1: cortex_a53: CPU workaround for 835769 was not applied
VERBOSE: BL1: cortex_a53: CPU workaround for disable_non_temporal_hint was not applied
INFO:    BL1: cortex_a53: CPU workaround for 843419 was applied
INFO:    BL1: cortex_a53: CPU workaround for 855873 was applied
INFO:    Using crypto library 'mbed TLS'
VERBOSE: rpi3: mbox: Sending request at 0x100f2420
NOTICE:  rpi3: Detected: Raspberry Pi 3 Model B (1GB, Sony, UK) [0x00a02082]
INFO:    BL1: Loading BL2
```
after renaming kernel8.img to kernel55.img, even after editing the config.txt. Must be something with TBB being enabled...?
```
root@kone2:/media/boot# ls -l
total 42784
-rwxr-xr-x 1 root root  1069851 Jan 29 14:51 armstub8.bin
-rwxr-xr-x 1 root root    25374 Jan 28 14:41 bcm2710-rpi-3-b.dtb
-rwxr-xr-x 1 root root    29470 Jan 30 09:34 bcm2710-rpi-3-b-pubkey.dtb
-rwxr-xr-x 1 root root    50268 Jan 28 12:36 bootcode.bin
-rwxr-xr-x 1 root root       45 Jan 30 09:46 config.txt
-rwxr-xr-x 1 root root    18693 Jan 28 12:36 COPYING.linux
-rwxr-xr-x 1 root root     2527 Jan 28 12:36 fixup_cd.dat
-rwxr-xr-x 1 root root     6635 Jan 28 12:36 fixup.dat
-rwxr-xr-x 1 root root     9770 Jan 28 12:36 fixup_db.dat
-rwxr-xr-x 1 root root     9768 Jan 28 12:36 fixup_x.dat
-rwxr-xr-x 1 root root 15098986 Jan 30 09:34 image.fit
-rwxr-xr-x 1 root root 15071744 Jan 28 14:41 kernel55.img
-rwxr-xr-x 1 root root     1494 Jan 28 12:36 LICENCE.broadcom
-rwxr-xr-x 1 root root   641252 Jan 28 12:36 start_cd.elf
-rwxr-xr-x 1 root root  4974116 Jan 28 12:36 start_db.elf
-rwxr-xr-x 1 root root  2840612 Jan 28 12:36 start.elf
-rwxr-xr-x 1 root root  3923364 Jan 28 12:36 start_x.elf
-rwxr-xr-x 1 root root    16384 Jan 29 14:51 uboot.env
root@kone2:/media/boot# cat config.txt
enable_uart=1
device_tree_address=0x01000000
root@kone2:/media/boot# 
```

igoropaniuk, 2019-02-03:
> after renaming kernel8.img to kernel55.img, even after editing the config.txt. Must be something with TBB being enabled...?

If there is only one difference in our setups, the answer is yes.
I've never tried TBB myself, so I can't confirm if this is a root cause.

johnphilby, 2019-02-14:
> root@kone2:/media/boot# ls -l
> total 42784
> -rwxr-xr-x 1 root root  1069851 Jan 29 14:51 armstub8.bin

I am assuming you have made the following changes to the .its file right? i.e removed optee.bin and replaced it with armstub8.bin

```
tee-1 {
            description = "atf";
            data = /incbin/("armstub8.bin");

```

In which case the .fit image generated in the subsequent steps would contain TF-A. We should be able to remove "armstub8.bin" and that should not hamper the boot process. But, on removal of "armstub8.bin", the board does not boot up. So, I think it is not working as it is supposed to by following the steps in the blog blog.crysys.hu.

I too am looking on how to slove this issue. Will keep you posted. If you beat me to it, let me know.

msa2, 2019-02-14:
I must admit that I'm not very familiar with TF-A boot internals, so following is just guess work:

I enabled trusted boot, and I presume it also verifies "kernel8.img". However, as u-boot reloads the kernel, it needs to be "reverified", and for that I activated fit-image with verification in u-boot. And, the only '.its' file that I'am aware, is related to that. If I put the fit image as "kernel8.img", TBB process apparently rejects it.

*If* trusted boot actually loads and verifies kernel8.img, one could also change u-boot to start it from memory (but I'm not sure).

Alternately, the build process could generate 'kernel8.img' as fit with signing, so that it would pass TBB checks and also that u-boot could verify it.

For the time being, I just left the 'kernel8.img' untouched, and just load the fit image in u-boot using another name.

This was just a PoC test, now I'm using non-TBB and struggling with trying to get TPM chip activated for IMA in kernel (currently spi init comes too late)...

msa2, 2019-04-15:
Returning to this topic. Judging from

  https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/plat/rpi3.rst

I assume that "armstub8.bin" and "kernel8.img" are verified in TBB, and u-boot can just start the linux kernel from the load address. Thus we don't need any trickery with FIT image.

The open question is for the files

- config.txt
- device tree blob

Looks like they are not protected in any away. Question is what threats they can open? One obvious target is fiddling with the load addresses in config.txt -- no sure how much sanity checks the ROM boot code does.

Of course, this is all just PoC thought experiment, can't make it secure unless we get some real support from the ROM boot code. Actually, I don't even know whether bootcode.bin and start.elf are protected in anyway? Could hack just by changing content in them?

msa2, 2019-04-15:
And, of course... all the previous verification steps are void and null in u-boot, as you can interactively load and execute any kernel code you wish -- if goal is trust chain from bl1 to linux kernel, you must remove u-boot (at least prevent it being interrupted and also freeze/protect uboot.env :-)

msa2, 2019-04-18:
Continuing my "monology" here... I'm guessing following (correct me, if I'm wrong)

```	- BL1 - "load/verify" BL2
	- BL2 - "load/verify" BL31, BL32, BL33 and kernel8.img?
	- BL31 - ?
	- BL32 - op-tee os components
	- BL33 - u-boot
```
I'm bit unsure how "kernel8.img" fits into "BLx" scheme. From the build, it seems BL33 is "u-boot". However, my earlier experiments seem to indicate that TBB does verify kernel8.img, I just don't know where and how?

It seems that the DTB is not verified by anyone (especially with overlays enabled)? What kind of security risk that gives?

jbech-linaro, 2019-04-18:
Sorry for killing the monologue :)

Slide 7 [here](https://www.slideshare.net/linaroorg/trusted-firmware-deepdivev10), still holds true when booting with TF-A. So, yes, the BLx things you've listed are correct (BL31 is the monitor).

msa2, 2019-05-09:
I'm still uncertain about the verification of kernel8.img. Does BL2 verify it along with the other loaded components or not? There does not seem any mention of it anywhere.

If it is not verified yet, it would nice if kernel8.img could be added to be verified by same mechanism as BL3x components (and not need to do any special FIT images and u-boot signed verification).

igoropaniuk, 2019-05-17:
@msa2 
> I'm still uncertain about the verification of kernel8.img. Does BL2 verify it along with the other loaded components or not? There does not seem any mention of it anywhere.

No. If you check TF-A TBB documentation, you will see that the latest booted BLOB, which is verified in the context of TBB feature, is BL33, which is U-boot in our setup (this is what you already found out :) ), and frankly TF-A doesn't and actually shouldn't care about what is booted after BL33.

> If it is not verified yet, it would nice if kernel8.img could be added to be verified by same mechanism as BL3x components (and not need to do any special FIT images and u-boot signed verification).

But why? If FIT verified boot already makes the deal (you can put whatever you want into a FIT image, including zImage; dtbs/dtbos; ramdisk etc. )

msa2, 2019-05-21:
My previous test with FIT image verification was faulty. All I managed to achieve, was only a hash verification, but no signature verification. The reason: I didn't include the public keys into u-boot dtb, which is embedded into u-boot binary.

I'm trying to figure out if u-boot build Makefile already has the option or not (likely). That is, somewhere before uboot and it's dtb are combined into single binary (and signed for TBB), there needs to be something that adds the FIT public key to dtb file, something like

`   ./tools/mkimage -F -k ../../fit/keys/ -K dts/dt.dtb -r <fit image path>
`
Unfortunaly, I haven't found a command that would just add the public key to dtb. The above command also does signing and wants the FIT image already built.

And, perhaps, to make building simpler, could use the keys of TBB also for FIT signing?

jbech-linaro, 2019-05-21:
@msa2 , these are very old and most likely obsolete instructions. But it is something we did a long time ago. I'm not sure for how long this will stay there, since there has been discussion about delete that wiki. But maybe, there is something on the U-Boot/Verified boot page that you can re-use.
https://wiki.linaro.org/WorkingGroups/Security/Verified-U-boot

msa2, 2019-05-21:
> https://wiki.linaro.org/WorkingGroups/Security/Verified-U-boot
It creates uboot with pubkeys, but does not integrate with TF-A  boot. The u-boot with keys should be signed by the TBB process in "arm-trusted-firmware". I was hoping to see some changes into optee build process that would handle it...

msa2, 2019-05-22:
Forgetting for a moment how to combine FIT verification with TBB process, I tested the FIT without TBB.

I need to put dtb with overlays into FIT image, but it is unclear which "images" need load address specified and which not.  And, **actually what addresses can I use**? I tried with following its (not specifying addresses except for kernel):

```
/dts-v1/;
/ {
    description = "RPi FIT Image";
    #address-cells = <2>;
    images {
        kernel-1 {
            description = "default kernel";
            data = /incbin/("Image");
            type = "kernel";
            arch = "arm64";
            os = "linux";
            compression = "none";
            load =  <0x00080000>;
            entry = <0x00080000>;
            hash@1 {
                algo = "sha1";
            };
        };
        fdt-1 {
            description = "device tree";
            data = /incbin/("bcm2710-rpi-3-b.dtb");
            type = "flat_dt";
            arch = "arm64";
            compression = "none";
            hash@1 {
                algo = "sha1";
            };
        };
        fdt-2 {
            data = /incbin/("spi-bcm2835.dtbo");
            type = "flat_dt";
            arch = "arm64";
            compression = "none";
            hash@1 {
                algo = "sha1";
            };
        };
        fdt-3 {
            data = /incbin/("mcp2515-can0.dtbo");
            type = "flat_dt";
            arch = "arm64";
            compression = "none";
            hash@1 {
                algo = "sha1";
            };
        };
        fdt-4 {
            data = /incbin/("tpm-slb9670.dtbo");
            type = "flat_dt";
            arch = "arm64";
            compression = "none";
            hash@1 {
                algo = "sha1";
            };
        };
    };
    configurations {
        default = "config-1";
        config-1 {
            description = "default configuration";
            kernel = "kernel-1";
            fdt = "fdt-1", "fdt-2", "fdt-3", "fdt-4";
            signature-1 {
                algo = "sha1,rsa2048";
                key-name-hint = "dev";
[... 90 lines cut ...]
  Architecture: AArch64
  Hash algo:    sha1
  Hash value:   bf77fdf4a9beaeefd769e60e86626972da0426fd
 Image 4 (fdt-4)
  Description:  unavailable
  Created:      Wed May 22 11:27:33 2019
  Type:         Flat Device Tree
  Compression:  uncompressed
  Data Size:    803 Bytes = 0.78 KiB = 0.00 MiB
  Architecture: AArch64
  Hash algo:    sha1
  Hash value:   aea4b24abb043782ba983b0d323d36b85ae8a807
 Default Configuration: 'config-1'
 Configuration 0 (config-1)
  Description:  default configuration
  Kernel:       kernel-1
  FDT:          fdt-1
                fdt-2
                fdt-3
                fdt-4
  Sign algo:    sha1,rsa2048:dev
  Sign value:   14f4c68dad518c7d20578885de17825cc6e61c4bf2652962b50c5c12ad2eb6ac1f130c0efdf445c95230a2540dbd949b72b1572120009422cc2e728ad689d2502333acd0b4c7f43dee20a4a30cf14afc30090e049561f0d5e8a9d33fd152773b92cb560cb1f0b93d60d0d1446b5777fbf99607356e34e8bc7b6453bf78d6d978e1d0b098655584a0f265553cd32fcddf042b45b889d1861e3fed46fc1196e31d5bbd538de05996de4df92794bd2eb1f1346037003fcc973712075175e426e39123e7e67d1bb5504864e8300d3b0737ebab95841daa4f0c81be156b05bdf33031d2365ba5d87966383e4075540dfb30dde2df114a122761054dd52529a4d4816b
  Timestamp:    Wed May 22 11:27:56 2019

```

msa2, 2019-05-22:
Returning to the question of integrating TBB and signed FIT image. The missing piece is to get the public key into u-boot dtb. Somewhere in the u-boot/Makefile, after dt.dtb has been created, we need add something like
```
tools/mkimage -f <path-to-kernel.its> -k <path-to-keys> -K dts/dt.dtb -r <somepath>/kernel8.fit
```
The rule "dtbs: dts/dt.dtb" ? Any suggestions?

msa2, 2019-05-23:
**Brute force hack**: adding mkimage into u-boot/dts/Makefile manages to build TBB verified uboot.bin with public key -- proof of concept, now if someone had skills find proper so lution.

```
diff --git a/dts/Makefile b/dts/Makefile
index a7a604303c..f311c982f7 100644
--- a/dts/Makefile
+++ b/dts/Makefile
@@ -25,11 +25,13 @@ $(obj)/dt-spl.dtb: $(DTB) $(objtree)/tools/fdtgrep FORCE
 
 $(obj)/dt.dtb: $(DTB) FORCE
        $(call if_changed,shipped)
+       ./tools/mkimage -f ../../fit/kernel.its -k ../../fit/keys -K $@ -r ../../fit/kernel8.fit
 
 targets += dt.dtb dt-spl.dtb

```

Now, my remaining problem is to find "algorithm" for determining the proper load addresses into its file.

```
U-Boot> run fitboot
15305098 bytes read in 639 ms (22.8 MiB/s)
## Loading kernel from FIT Image at 02000000 ...
   Using 'config-1' configuration
   Verifying Hash Integrity ... sha1,rsa2048:dev+ OK
   Trying 'kernel-1' kernel subimage
   ...
```

msa2, 2019-05-24:
> Now, my remaining problem is to find "algorithm" for determining the proper load addresses into its file.

My bad... I had wrong "bootm" with only kernel address. The fdt's in its-file don't need "load" addresses. I just needed to do my bootm as:
`
bootm ${kernel_addr_r} - ${fdt_addr_r}
`
...and it works, overlays and all.

Time to wrap up? The u-boot hack for installing public key is ugly and requires the linux kernel has been fully built before u-boot.

msa2, 2019-05-27:
>  $(obj)/dt.dtb: $(DTB) FORCE
>         $(call if_changed,shipped)
> +       ./tools/mkimage -f ../../fit/kernel.its -k ../../fit/keys -K $@ -r ../../fit/kernel8.fit

Just a caution: above assumes existing linux image, and because optee builds u-boot before linux, the fit image will always contain the previously compile kernel, if the kernel is recompiled. Running make twice gets correct kernel in (with lots of weird error messages about modules, though....)

closed 2020-01-19 by github-actions[bot]