OP-TEE/build #339: Python 2 and Python 3 on OP-TEE Raspberry Pi 3

siddhantdash, 2019-01-28:
I have set up successfully OP-TEE on Raspberry Pi 3 (RPi3). Xtest works and returns no errors. The OP-TEE RPi3 is connected to the internet through ethernet. I am trying to run my Python script in a secure world using OP-TEE. However, there is no Python installed. I tried to install using the following.
1. apt-get: not found as it is not installed.
2. wget: I tried using wget but it says "not an HTTP or FTP URL when trying to install from Python FTP URL.

Please help how should I install and run python scripts on OP-TEE on RPi3

siddhantdash, 2019-01-28:
Thanks for the reply. 
I had developed secure algorithm on Python which I wanted to run on the secure world on OP-TEE Raspberry Pi. As you said, there is not any way to run python on op-tee then ?
Also, how do I install gcc on OP-TEE just in case I wish to run C programs?

How do I enable a buildroot configuration flag? Is it there on Buildroot documentation and not here on the OP-TEE? 

Thanks!

jbech-linaro, 2019-01-28:
> Thanks for the reply.
> I had developed secure algorithm on Python which I wanted to run on the secure world on OP-TEE Raspberry Pi. As you said, there is not any way to run python on op-tee then ?

No, it's not possible. Or ... it's just software, but I sorting out all dependencies and build a Python interpreter for secure side is nothing I would even consider doing. I imagine it would be pretty hard and complicated to make it work.

> How do I enable a buildroot configuration flag? Is it there on Buildroot documentation and not here on the OP-TEE?

Easiest to enable a BR flag with our OP-TEE setup is to 
1) Find out the name of the configuration flag (grep / search for it in the buildroot folder).
2) Add a new line for it in our common.mk file, like we've done here for example: https://github.com/OP-TEE/build/blob/master/common.mk#L236-L237

Note that this still just enables things running in normal world, i..e, you plain Linux environment.

siddhantdash, 2019-01-28:
Thanks for all the details. 
I wish to know in order to run my applications in the secure world, how do I get into that zone? Where do I place my files? 

A quick question, the Linux OS on which the OP-TEE is based, could that be used to run SenseHat module? Does it have the drivers to run it as I would like to simulate a real-time environment of a sensor node?

jbech-linaro, 2019-01-28:
> Thanks for all the details.
> I wish to know in order to run my applications in the secure world, how do I get into that zone? Where do I place my files?
> 

That is the Trusted Applications as we refer to them. Those are C applications that are loaded from flash (or from secure storage). https://github.com/OP-TEE/optee_os/blob/master/documentation/optee_design.md#12-trusted-applications

> A quick question, the Linux OS on which the OP-TEE is based, 

So, this is no specific OS in terms of Ubuntu, Debian, Arch Linux, Raspbian and whatever is out there. This is just a vanilla/upstream Linux kernel that is built to boot up on the devices making it possible to run xtest. For root file system we are using Buildroot. 

> could that be used to run SenseHat module? Does it have the drivers to run it as I would like to simulate a real-time environment of a sensor node?

From non-secure Linux side? Yes, if you have all drivers/bus protocols enabled (kernel) and necessary tools in the root fs.

From secure side? Here the answer is: maybe. We have SPI support on secure side, so in case you communicate with the SenseHat using SPI, then it should probably work. Having that said, you have to fiddle with configuration on both secure and non-secure side to make that work.

In any case, think of anything running on secure side as a stripped down "C" project (+ assembler).

siddhantdash, 2019-01-28:
Thanks, @jbech-linaro for patiently answering all my queries. I wanted to know if the kernel to the OP-TEE is open-source so I could explore modifying it to my requirements? Is there a license for that?

jbech-linaro, 2019-01-28:
> Thanks, @jbech-linaro for patiently answering all my queries. I wanted to know if the kernel or the source code to the OP-TEE is open-source so I could explore modifying it to my requirements? Is there a license for that?

Yes, there are two kernels and both are fully open source:
* Non-secure side, i.e., Linux kernel - GPLv2 --> https://github.com/linaro-swg/linux
* Secure side, i.e, OP-TEE / TEE core - BSD 2-Clause --> https://github.com/OP-TEE/optee_os

jbech-linaro, 2019-01-30:
@siddhantdash ROM, yes and no. By default the secure side have no direct access to storage, that's the reason why we piggy back on tee-supplicant in non-secure world for storage operations. Having that said. I'm sure it's possible to reconfigure the system so the secure side **can** get access. But it's probably not a good idea, since when the system is up and running, Linux kernel "owns" the storage devices.

RAM? yes

siddhantdash, 2019-01-30:
Thanks @jbech-linaro 
How can I access the RAM? Is there any specific method in OP-TEE for it to work?

jbech-linaro, 2019-01-30:
Please read about the shared memory concept in the GlobalPlatform specifications (globalplatform.org, Client API and Internal Core API) and also here: https://github.com/OP-TEE/optee_os/blob/master/documentation/optee_design.md#7-shared-memory

siddhantdash, 2019-02-26:
Hi @jbech-linaro 
I tried installing gcc using buildroot package but it never gets installed. I tried installing python in the same way and it works perfectly in the normal world. Could you let me know what is happening?

jbech-linaro, 2019-02-26:
@siddhantdash sorry, I'm not a buildroot guy, I'm just a regular user of it which means I rarely do anything special with it.

siddhantdash, 2019-02-26:
@jbech-linaro  Thanks! Could you tell me the compiler a secure Trusted application uses? I am sure it would be using some C compiler and do you have any idea that native gcc could be installed or not?

jbech-linaro, 2019-02-26:
I'm using the cross compiler that comes when setting up everything using build.git (see [this](https://optee.readthedocs.io/building/gits/build.html#get-and-build-the-solution) and [this](https://optee.readthedocs.io/building/toolchains.html) if you want to download it standalone).

siddhantdash, 2019-02-26:
I used the standalone option. But I couldn't run gcc. Can you tell me how can I invoke the cross compiler in the normal world to run a C application?

jbech-linaro, 2019-02-26:
I don't invoke it in normal world, I'm doing everything on the host PC using the cross compiler as mentioned above.
```bash
[jbech@guppy ~/devel/optee_projects/rpi3] 
$ ./toolchains/aarch32/bin/arm-linux-gnueabihf-gcc
arm-linux-gnueabihf-gcc: fatal error: no input files
compilation terminated.
```

siddhantdash, 2019-02-27:
I want to run it inside the Raspberry Pi after installing OP-TEE on it. I tried to locate the arm-linux-gnueabihf-gcc but could not find it inside the OP-TEE on Raspberry Pi 3

jbech-linaro, 2019-02-27:
It's not there by default (for more information see [this](https://optee.readthedocs.io/building/gits/build.html#root-filesystem)), to get gcc included you would need to enable it in Buildroot, however, as I've said, I'm not a Buildroot guy, but usually it's just about finding the correct config flag for Buildroot and enable it (somewhere in [this](https://github.com/OP-TEE/build/blob/master/common.mk#L200-L247) section). So,
1. Try to figure out what flag you need to enable to get arm GCC enabled and deployed in the root fs.
2. Add that config flag to the section I refer to above.
3. Rebuild and see whether it got included or not.

jbech-linaro, 2019-08-08:
I'm closing the ticket, either because it has already been answered or that it is no longer relevant or it could be lack of response from the author. Having that said, feel free to re-open the ticket if you have more to add to the ticket.

closed 2019-08-08 by jbech-linaro