OP-TEE/build #346: Modifying the device tree (rpi3)?

msa2, 2019-02-06:
I'm somewhat lost in your build process, and have no clue where to start if I need to modify the dtb for rpi? We were planning to play with TPM chip on rpi3, and one option is "LetstrustTPM"...  their instructions are for plain rpi3, and I'm wondering where to start to get equivalent of
`
$ make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- zImage modules dtbs overlays/letstrust-tpm.dtbo -j8
`
included into optee build process?

jbech-linaro, 2019-02-06:
The DTB in use will be picked configure here:
https://github.com/OP-TEE/build/blob/master/rpi3.mk#L40

And then it's put into the BOOT partition here:
https://github.com/OP-TEE/build/blob/master/rpi3.mk#L166

I think if you change those two lines and rebuild, then it should (probably) just work.

msa2, 2019-02-06:
I couldn't figure out what to put into L40 (my attempt just generated build fail). Seems I need to modify

```
@@ -63,6 +63,7 @@ dtbo-$(CONFIG_ARCH_BCM2835) += \
 	jedec-spi-nor.dtbo \
 	justboom-dac.dtbo \
 	justboom-digi.dtbo \
+	letstrust-tpm.dtbo \
 	lirc-rpi.dtbo \
 	ltc294x.dtbo \
 	mbed-dac.dtbo \
```
Now, after that build worked and I go "letstrust-tpm.dtbo", which I copied into boot/overlays

```
# cat config.txt
enable_uart=1
dtparam=spi=on
dtoverlay=letstrust-tpm
kernel_address=0x02000000
device_tree_address=0x01000000
# ls -l overlays/
total 2
-rwxr-xr-x    1 root     root           803 Feb  6  2019 letstrust-tpm.dtbo

```
Also tried to enable tpm driver in kernel. However, "/dev/tpm*" is not appearing... maybe I messed up with the kernel config somehow...

msa2, 2019-02-06:
My modified rpi3.conf
```
diff --git a/kconfigs/rpi3.conf b/kconfigs/rpi3.conf
index 07554cf..526b7bb 100644
--- a/kconfigs/rpi3.conf
+++ b/kconfigs/rpi3.conf
@@ -1,2 +1,7 @@
 CONFIG_TEE=y
 CONFIG_OPTEE=y
+CONFIG_HW_RANDOM_TPM=y
+CONFIG_TCG_TPM=y
+CONFIG_TCG_TIS_CORE=y
+CONFIG_TCG_TIS_SPI=y
+CONFIG_SECURITYFS=y
```

msa2, 2019-02-06:
Seems something is there, maybe the overlay dtb worked (maybe these are always there).. Anyway, I guess I just have to poke more,....
```
670\@0/naro-installer:/sys/firmware/devicetree/base/soc/spi@7e204000# ls -l slb96
total 0
-r--r--r-- 1 root root  4 Jan 23 11:15 #address-cells
-r--r--r-- 1 root root  4 Jan 23 11:15 #size-cells
-r--r--r-- 1 root root 17 Jan 23 11:15 compatible
-r--r--r-- 1 root root  8 Jan 23 11:15 name
-r--r--r-- 1 root root  4 Jan 23 11:15 phandle
-r--r--r-- 1 root root  4 Jan 23 11:15 reg
-r--r--r-- 1 root root  4 Jan 23 11:15 spi-max-frequency
-r--r--r-- 1 root root  5 Jan 23 11:15 status
root@linaro-installer:/sys/firmware/devicetree/base/soc/spi@7e204000# 

```

msa2, 2019-02-07:
This no more about modifying device tree (I copied the tpm-slb9640-overlay.dts raspberry linux, and it seems to build ok. I also copied drivers/char/tpm/* files...

But, after I think I'm doing all correct, build ok but still no `/dev/tpm*` appears... I even added printk statements into tpm-interface.c, and I can see that all following calls are executed without errors
```
 	tpm_class = class_create(THIS_MODULE, "tpm");
	tpmrm_class = class_create(THIS_MODULE, "tpmrm");
	rc = alloc_chrdev_region(&tpm_devt, 0, 2*TPM_NUM_DEVICES, "tpm");
```
... but, no /dev/tpm0 visible. Am I still missing some configured driver (there is "TGC_INFINEON, but that depends on PNP, so not available on rpi -- no ACPI; and TCG_TIS_I2C_INFINEON?)? Currently I have
```

CONFIG_TEE=y
CONFIG_OPTEE=y
CONFIG_HW_RANDOM_TPM=y
CONFIG_TCG_TPM=y
CONFIG_TCG_TIS_CORE=y
CONFIG_TCG_TIS_SPI=y
CONFIG_TCG_TIS=y
CONFIG_SECURITYFS=y

```

msa2, 2019-02-07:
Seems I need to activate "spi_bcm2835"...
`
modprobe spi_bcm2835
`

After that I see `/dev/tpm0`, `/dev/tpmrm0 `and `/dev/spidev0.0`

(not sure whether copying char/tpm/* from https://github.com/raspberrypi/linux/tree/rpi-3.14.y was needed, but it doesn't seem to hurt either).

msa2, 2019-02-08:
"Final report" for the time being. This issue can be closed now, however hopefully the instructions will still be found, if anyone else wants to do the same...

Today (8.2.2019) checked out the latest of optee rpi3.xml, and with following configurations the Letstrust (Infineon) TPM chip appears to become alive on boot:
```
diff --git a/kconfigs/rpi3.conf b/kconfigs/rpi3.conf
index 07554cf..bd62007 100644
--- a/kconfigs/rpi3.conf
+++ b/kconfigs/rpi3.conf
@@ -1,2 +1,8 @@
 CONFIG_TEE=y
 CONFIG_OPTEE=y
+CONFIG_HW_RANDOM_TPM=y
+CONFIG_SPI_BCM2835=y
+CONFIG_TCG_TPM=y
+CONFIG_TCG_TIS_CORE=y
+CONFIG_TCG_TIS_SPI=y
+CONFIG_SECURITYFS=y
```

The linux side is missing the "`tpm-slb9670-overlay.dts`", can be got from Raspberry git

    https://github.com/raspberrypi/linux/tree/rpi-3.14.y

and enable compilation on optee linux by (maybe this eventually will be available by default):
```
diff --git a/arch/arm/boot/dts/overlays/Makefile b/arch/arm/boot/dts/overlays/Makefile
index 969f40721..f216b2112 100644
--- a/arch/arm/boot/dts/overlays/Makefile
+++ b/arch/arm/boot/dts/overlays/Makefile
@@ -132,6 +132,7 @@ dtbo-$(CONFIG_ARCH_BCM2835) += \
        tc358743.dtbo \
        tc358743-audio.dtbo \
        tinylcd35.dtbo \
+       tpm-slb9670.dtbo \
        uart0.dtbo \
        uart1.dtbo \
        upstream.dtbo \
```
On preparing the bootable SD card, you need to create 'overlays' directory, and copy `tpm-slb9670.dtbo` there:
```
root@linaro-installer:/media/boot# ls -l overlays/
total 2
-rwxr-xr-x 1 root root 803 Feb  7 08:07 tpm-slb9670.dtbo
```
and activate it and SPI with two additional lines in config.txt
```
root@linaro-installer:/media/boot# diff -c ~linaro/config-orig.txt config.txt
*** /home/linaro/config-orig.txt	2019-02-08 10:37:41.004324741 +0000
--- config.txt	2019-02-07 08:06:18.000000000 +0000
***************
*** 1,3 ****
--- 1,5 ----
  enable_uart=1
+ dtparam=spi=on
+ dtoverlay=tpm-slb9670
  kernel_address=0x02000000
  device_tree_address=0x01000000

```
After boot, you should see the tpm and spi devices:
```
root@linaro-installer:~# ls -l /dev/{spi*,tpm*}
crw------- 1 root root 153,     0 Feb  7 14:49 /dev/spidev0.0
crw------- 1 root root  10,   224 Feb  7 14:49 /dev/tpm0
crw------- 1 root root 253, 65536 Feb  7 14:49 /dev/tpmrm0
```

jbech-linaro, 2019-02-08:
Thanks @msa2 , eventually I'll pick it up in a guide in the updated documentation. In any case it's good to have it documented here in a comment at least.

closed 2019-02-08 by jbech-linaro