OP-TEE/build #348: Compilation error with OPTEE 3.4.0 release.

pangupta, 2019-02-19:
CC      out/arm-plat-ls/ta_arm32-lib/libmbedtls/mbedtls/library/aes.o
aarch64-linux-gnu-gcc: error: unrecognized command line option -mthumb
aarch64-linux-gnu-gcc: error: unrecognized command line option -mthumb-interwork
aarch64-linux-gnu-gcc: error: unrecognized command line option -mno-unaligned-access; did you mean -Wno-aligned-new?
aarch64-linux-gnu-gcc: error: unrecognized command line option -mfloat-abi=hard
mk/compile.mk:147: recipe for target 'out/arm-plat-ls/ta_arm32-lib/libmbedtls/mbedtls/library/aes.o' failed
make: *** [out/arm-plat-ls/ta_arm32-lib/libmbedtls/mbedtls/library/aes.o] Error 1

jforissier, 2019-02-19:
Hi,

Could be due to wrong cross compiler being used. Could you check exactly which compiler is invoked? Simply append `V=1` to the make command.

jforissier, 2019-02-19:
https://github.com/OP-TEE/optee_os/blob/3.4.0/documentation/build_system.md#cross_compile-cross-compiler-selection

pangupta, 2019-02-19:
Thanks.
From Release 3.4.0 onwards, setting CROSS_COMPILE32 is a mandatory step.
After doing so, the problem got solved.

jforissier, 2019-02-19:
Well I don't think something is supposed to have changed with 3.4.0, but I'm glad you git it working!

pangupta, 2019-02-19:
Earlier, 32 bit toolchain is not used in the compilation of optee_os.
Which is now become mandatory such that mbedtls library files can be compiled for ta_arm32-lib.

jforissier, 2019-02-19:
I don't think it's true. All TA libraries (libutils.a libutee.a etc) have always been compiled in both 64 and 32-bit modes as soon as TEE core is 64 bit. Have you identified a particular commit that changed the behavior? Not that it matters that much I suppose.
One thing I can think of is https://github.com/OP-TEE/optee_os/commit/dc701d99e0d0 but it's supposed to add more flexibility and not change the default behavior.

pangupta, 2019-02-22:
Yes, you are right that the said commit id is making the difference.
In my compilation environment, CFG_USER_TA_TARGETS is empty.
That means: CFG_USER_TA_TARGETS ="ta_arm32 ta_arm64".

Setting CFG_USER_TA_TARGETS = "ta_arm64", is required.

We can close this issue.

closed 2019-02-22 by pangupta

jforissier, 2019-02-22:
@pangupta thanks for clarifying.