OP-TEE/build #354: Add support for RPI3 B Plus

johnphilby, 2019-04-02:
@jbech-linaro As per your comments [here](https://github.com/OP-TEE/build/issues/276#issuecomment-474731038) I have added a PLATFORM=rpi3-b and PLATFORM=rpi3-bplus variable to differentiate between the two targets. I could not stick to your advice of using PLATFORM=rpi-3b since this gave compile errors at optee_os/mk/subdir.mk. This could be because the script takes the string upto the '-' and this would lead to a conflict in the existing implementation. So, I stuck with using "rpi3-b" and "rpi3-bplus".

Also, since we pull the RPi3 firmware from the official repo and because the "start*" binaries there do not work for RPi3 B +, I had to create a new directory in the build/rpi3/firmware path to pick up the working "start*" binaries.

Have tested using xtest and it completed without errors on RPi3 B +.

Please find the changeset here https://github.com/johnphilby/build/tree/add-support-rpi3-bplus

Let me know your thoughts.

jbech-linaro, 2019-04-10:
I had a quick look at it, seems like a good start. I think you should just create a pull request for it and then we can take the review comments etc as we usually do, thanks!

johnphilby, 2019-04-10:
Pull request created https://github.com/OP-TEE/build/pull/358

closed 2020-01-19 by github-actions[bot]