OP-TEE/build #357: Custom TA crashing after moving to branch 3.4.0

divneil, 2019-04-08:
I worked on a TA around June 2018, so, had to come back to it now.
I tried to run after fixing the build issues (https://github.com/OP-TEE/build/issues/355), and TA crashes. Please find below the log. Can you please help in giving some pointers.

D/TC:? 0 load_elf:827 Lookup user TA ELF b56091e0-e3c2-41b9-9f7a-983e26d3bcf1 (REE)
D/TC:? 0 load_elf_from_store:795 ELF load address 0x40025000
D/TC:? 0 tee_ta_init_user_ta_session:1017 Processing relocations in b56091e0-e3c2-41b9-9f7a-983e26d3bcf1
E/TA:  assertion '!((vaddr_t)data & (POOL_ALIGN - 1))' failed at lib/libutils/ext/mempool.c:134 in mempool_alloc_pool()
E/TC:? 0 
E/TC:? 0 TA panicked with code 0xffff0000
E/TC:? 0 Status of TA b56091e0-e3c2-41b9-9f7a-983e26d3bcf1 (0x10161d30) (active)
E/TC:? 0  arch: aarch64  load address: 0x40025000 ctx-idr: 2
E/TC:? 0  stack: 0x40004000 133120
E/TC:? 0  region 0: va 0x40000000 pa 0x10100000 size 0x2000 flags ---R-X 
E/TC:? 0  region 1: va 0x40002000 pa 0x10164000 size 0x1000 flags ---RW- 
E/TC:? 0  region 2: va 0x40004000 pa 0x10977000 size 0x21000 flags rw-RW- 
E/TC:? 0  region 3: va 0x40025000 pa 0x10800000 size 0x133000 flags r-x--- [0] .ta_head .text .eh_frame .rodata .gnu.hash .dynsym .rela.dyn .got .rela.got .dynamic .dynstr .hash
E/TC:? 0  region 4: va 0x40158000 pa 0x10933000 size 0x44000 flags rw---- [0] .data .bss
E/TC:? 0  [0] b56091e0-e3c2-41b9-9f7a-983e26d3bcf1 @ 0x40025000 (optee_examples/sdo/ta/b56091e0-e3c2-41b9-9f7a-983e26d3bcf1.elf)
E/TC:? 0 Call stack:
E/TC:? 0  0x0000000040066934 _assert_break at optee_os/lib/libutee/assert.c:27 (discriminator 1)
E/TC:? 0  0x00000000400700a0 mempool_alloc_pool at optee_os/lib/libutils/ext/mempool.c:134 (discriminator 1)
E/TC:? 0  0x000000004006b2ac _TEE_MathAPI_Init at optee_os/lib/libutee/tee_api_arith_mpi.c:48
E/TC:? 0  0x000000004006a6ec init_instance at optee_os/lib/libutee/arch/arm/user_ta_entry.c:56
E/TC:? 0  0x0000000010102d88 ???
D/TC:? 0 user_ta_enter:315 tee_user_ta_enter: TA panicked with code 0xffff0000
D/TC:? 0 tee_ta_close_session:380 tee_ta_close_session(0x10161d90)
D/TC:? 0 tee_ta_close_session:399 Destroy session
D/TC:? 0 tee_ta_close_session:425 Destroy TA ctx
E/TC:? 0 tee_ta_open_session:586 Failed. Return error 0xffff3024
TEEC_Opensession failed with code 0xffff3024 origin 0x3

jforissier, 2019-04-08:
Fixed in master (and therefore in next release).
https://github.com/OP-TEE/optee_os/pull/2792

divneil, 2019-04-08:
Thanks @jforissier. It works now, so, closing this issue.

closed 2019-04-08 by divneil