OP-TEE/build #359: run xtest error on hikey960

qingyunqu, 2019-04-11:
Hi,
When I flash the optee into hikey960 board successfully, and try to run `xtest` or `optee_example_hello_world`.
There occurs error:
```
# xtest
Run test suite with level=0

TEE test application started with device [(null)]
######################################################
#
# regression
#
######################################################
 
* regression_1001 Core self tests
E/TC:4 0 tee_entry_std:551 Bad arg address 0x217f87000
regression_1000.c:246: res has an unexpected value: 0xffff000e = TEEC_ERROR_COMMUNICATION, expectedS
Segmentation fault
```
and
```
# optee_example_hello_world 
E/TC:4 0 tee_entry_std:551 Bad arg address 0x217cab000
optee_example_hello_world: TEEC_Opensession failed with code 0xffff000e origin 0x2
```
Can you find out why?

jforissier, 2019-04-11:
> Can you find out why?

Me, probably not. Because I have no idea how you have built your binaries.
But I know for sure that if you build everything from build.git, it works fine ;-)

jenswi-linaro, 2019-04-12:
My guess is that OP-TEE in secure world isn't aware of all the DDR which Linux uses. So it doesn't recognize `0x217cab000` as a valid non-secure physical address. But, there could be other causes for this error too.

qingyunqu, 2019-05-07:
@jenswi-linaro Maybe there is anything I can do to test what happened on the hikey960 borad? Except run the `xtest`.

jenswi-linaro, 2019-05-08:
You can double check the DTB used by the kernel to see that the configured memory matches the memory configuration of hikey960 in `core/arch/arm/plat-hikey/platform_config.h`

liurenju, 2019-07-07:
I also experienced the same issue. I simply download the build for hikey960 through the instructions in OP-TEE website. And it shows me this error.

vchong, 2019-07-08:
This has been fixed. Please see https://github.com/OP-TEE/optee_os/issues/3068.

closed 2019-07-08 by vchong