OP-TEE/build #360: My rpi3 b gets only one core running on linux?

msa2, 2019-04-12:
I just noticed that I get only one CPU on Raspberry PI 3 b, when I think I should get 4? Is this as it should be or do I have some configuration problem?

```
[    0.000000] Boot CPU: AArch64 Processor [410fd034]
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
...
[    0.011332] EFI services will not be available.
[    0.013144] smp: Bringing up secondary CPUs ...
[    1.065310] CPU1: failed to come online
[    1.065338] CPU1: failed in unknown state : 0x0
[    2.098310] CPU2: failed to come online
[    2.098339] CPU2: failed in unknown state : 0x0
[    3.131253] CPU3: failed to come online
[    3.131280] CPU3: failed in unknown state : 0x0
[    3.131346] smp: Brought up 1 node, 1 CPU
[    3.131371] SMP: Total of 1 processors activated.
```

jbech-linaro, 2019-04-12:
I've seen this being asked in another GitHub issue also, I cannot remember the conclusion and whether we did something about it or not. But I agree, all 4 cores should be brought to life to utilize and test the system fully.

msa2, 2019-04-12:
Yes, I actually found that issue, when searching more...

https://github.com/OP-TEE/optee_os/issues/2523

msa2, 2019-05-27:
You might want to integrate the following patch...

https://github.com/AntonioND/rpi3-arm-tf-bootstrap/blob/master/0001-rpi3-Enable-PSCI-support.patch

I applied it manually, and I think I got all 4 cores running now.

jbech-linaro, 2019-06-03:
> I applied it manually, and I think I got all 4 cores running now.

Cool! Would you mind send a pull request to the branch rpi3-optee-4.14 at https://github.com/linaro-swg/linux/tree/rpi3-optee-4.14?

jbech-linaro, 2019-06-28:
> I applied it manually, and I think I got all 4 cores running now.

I just did the same, but ...
```
[    0.013141] smp: Bringing up secondary CPUs ...
[    1.065164] CPU1: failed to come online
[    1.065192] CPU1: failed in unknown state : 0x0
[    2.098017] CPU2: failed to come online
[    2.098044] CPU2: failed in unknown state : 0x0
[    3.130865] CPU3: failed to come online
[    3.130892] CPU3: failed in unknown state : 0x0
[    3.130957] smp: Brought up 1 node, 1 CPU
[    3.130983] SMP: Total of 1 processors activated.
```
Maybe I've missed some step?

jbech-linaro, 2019-06-28:
Probably stale files, after cleaning and rebuilding it looks better
```
[    0.013153] smp: Bringing up secondary CPUs ...
[    0.032694] Detected VIPT I-cache on CPU1
[    0.032765] CPU1: Booted secondary processor [410fd034]
[    0.051874] Detected VIPT I-cache on CPU2
[    0.051923] CPU2: Booted secondary processor [410fd034]
[    0.071094] Detected VIPT I-cache on CPU3
[    0.071137] CPU3: Booted secondary processor [410fd034]
[    0.071275] smp: Brought up 1 node, 4 CPUs
[    0.071401] SMP: Total of 4 processors activated.
```
I will cherry-pick that patch to the Linaro/Rpi/kernel fork.

jbech-linaro, 2019-06-28:
Patch cherry-picked, see https://github.com/linaro-swg/linux/commits/rpi3-optee-4.14. Thanks @msa2 , closing the ticket!

closed 2019-06-28 by jbech-linaro