OP-TEE/build #364: Build the solution for RPI3 B with inline assembly statements in TA

Xinyao90, 2019-04-27:
Hi,

I'm trying to add inline assembly in ta.c file as follows:
__asm__ volatile("mov r3, #0x2710\n");
Then when I build the solution in linux ubuntu for RPI3 B with command:
$ make -j `nproc`

There shows the error message:
/tmp/ccq2KO7F.s:151: Error: operand 1 must be an integer register -- `mov r3,#0x2710'
/tmp/ccq2KO7F.s:160: Error: operand 1 must be an integer register -- `mov r2,x0'
/tmp/ccq2KO7F.s:169: Error: operand 1 must be an integer or stack pointer register -- `sub r2,r2,r3'
/tmp/ccq2KO7F.s:181: Error: operand 1 must be an integer register -- `subs r3,r3,#0x1'
/tmp/ccq2KO7F.s:177: Error: undefined symbol r2 used as an immediate value

The suggestion to solve the problem is to ask the compiler to use 32-bit mode instead of 64-bit mode by adding CFLAGS:  -march=armv7-a -mtune=cortex-a5. 
However, I cannot find the Makefile to modify the flag. 

Could anyone help me to locate the Makefile to do modification there?

Thanks a lot,
Xinyao

jenswi-linaro, 2019-04-29:
Since you're obviously compiling a 64-bit TA (you're including `.../export-ta_arm64/mk/ta_dev_kit.mk` from the makefile), how about using AArch64 assembly instead? In this case I guess the equivalent instruction would be `mov x3, #0x2710`.

If you'd like to compile a 32-bit TA you need to include `.../export-ta_arm32/mk/ta_dev_kit.mk` instead from the makefile.

Xinyao90, 2019-04-29:
@jenswi-linaro Thanks for your quick reply. I’m still wondering where the Makefile is, I‘d like to make changes inside of it. I’m using $ make -j nproc to build the solution.

Xinyao90, 2019-04-29:
@jenswi-linaro  I've used your suggested instruction mov x3, #0x2710, there is no error when building the solution on Linux Ubuntu. However, when I try to run it on raspberry pi 3B using UART, it gives translation fault related error message as below. Any suggestion would be helpful.

E/TC:0 0 User TA data-abort at address 0x2710 (translation fault)
E/TC:0 0  esr 0x92000045  ttbr0 0x2000010176000   ttbr1 0x00000000   cidr 0x0
E/TC:0 0  cpu #0          cpsr 0x60000100
E/TC:0 0  x0  0000000000000000 x1  0000000000000001
E/TC:0 0  x2  0000000000000000 x3  0000000000002710
E/TC:0 0  x4  0000000000000000 x5  000000004000e92c
E/TC:0 0  x6  0000000000000003 x7  0000000000000000
E/TC:0 0  x8  0000000000000000 x9  0000000000000003
E/TC:0 0  x10 00000000400056c0 x11 00000000d6bd0200
E/TC:0 0  x12 00000000d491fbf8 x13 0000000040004f80
E/TC:0 0  x14 00000000516f822f x15 0000000000000000
E/TC:0 0  x16 000000001017ae28 x17 e894f12d00000000
E/TC:0 0  x18 1825380200000000 x19 000000004001a150
E/TC:0 0  x20 000000004000f000 x21 0000000040004f40
E/TC:0 0  x22 0000000000000003 x23 0000000040004f80
E/TC:0 0  x24 0000000000000001 x25 00000000101618e0
E/TC:0 0  x26 000000001017b130 x27 0000000000000001
E/TC:0 0  x28 0000000040004f88 x29 0000000040004ef0
E/TC:0 0  x30 000000004000533c elr 0000000040005080
E/TC:0 0  sp_el0 0000000040004ef0
E/TC:0 0 Status of TA 8aaaf200-2450-11e4-abe2-0002a5d5c51b (0x10161b00) (active)
E/TC:0 0  arch: aarch64  load address: 0x0000000040005000 ctx-idr: 2
E/TC:0 0  stack: 0x0000000040004000 4096
E/TC:0 0  region  0: va 0x0000000040000000 pa 0x0000000010100000 size 0x002000 flags ---R-X 
E/TC:0 0  region  1: va 0x0000000040002000 pa 0x0000000010164000 size 0x001000 flags ---RW- 
E/TC:0 0  region  2: va 0x0000000040004000 pa 0x0000000010823000 size 0x001000 flags rw-RW- 
E/TC:0 0  region  3: va 0x0000000040005000 pa 0x000000001080d000 size 0x00b000 flags r-x--- [0]
E/TC:0 0  region  4: va 0x0000000040010000 pa 0x0000000010818000 size 0x00b000 flags rw---- [0]
E/TC:0 0  [0] 8aaaf200-2450-11e4-abe2-0002a5d5c51b @ 0x0000000040005000
E/TC:0 0 Call stack:
E/TC:0 0  0x0000000040005080
E/TC:0 0  0x0000000010102fb8
 of the program.D/TC:? 0 user_ta_enter:317 tee_user_ta_enter: TA panicked with code 0xdeadbeef
 start_tD/TC:? 0 tee_ta_invoke_command:673 Error: ffff3024 of 3
 = 59668

jenswi-linaro, 2019-04-30:
I don't know where that Makefile is, ta.c is not a file from the OP-TEE project.

jenswi-linaro, 2019-04-30:
Adding random inline assembly can very well introduce errors like this. Notice that `0x2710` indeed is in `x3`.

Xinyao90, 2019-05-09:
@jenswi-linaro  Thanks Jenswi, I do figured out the assembly code problem with your help. 

BTW, do you know if I follow https://optee.readthedocs.io/building/gits/build.html this build instruction for raspberry pi 3B. In step 5-Build the solution, where dose the binary file store? The reason is I'd like to use the disassembler to see the instructions.

Thanks a lot.

jenswi-linaro, 2019-05-09:
Which binary files? The optee_os stuff is under optee_os/out and the optee_test stuff is under out-br/build/optee_test-1.0/

closed 2019-07-19 by Xinyao90