OP-TEE/build #367: Build for rk3328

jonsmirl, 2019-05-10:
Is it possible to get a build for the rk3328 similar to the rpi3 one added?

jbech-linaro, 2019-05-13:
Hi @jonsmirl , I see no problem with that, having that said, none of us in the core team have any RK-boards at our hands, so it's a bit hard for us to put something together (we're relying on the external [RK maintainers](https://github.com/OP-TEE/optee_os/blob/master/MAINTAINERS#L128)). If you have a device and a working setup, then feel free to put together a manifest (manifest.git) and makefiles (build.git). If you have no experience with that, then I can guide you.

jonsmirl, 2019-05-13:
In the discussion on the other optee-os issue the Rockchip people have explained what the problem is. Upsteam optee only has support in it for the RK3228. They have optee running on all of their other CPUs, but they have not published that support into upstream optee.  That's a problem because the widely available dev boards for Rockchip use the RK3328 and RK3399.  

Also, there is no support in upstream for their four new CPUs that are starting to ship (that is what we are targeting). They have provided binaries for these CPUs.

I was confused because I though I was incorrectly building optee, now I know the code is not there.

closed 2020-01-19 by github-actions[bot]

TaihuLight, 2021-12-31:
@jonsmirl I have a Rockchip Pi 4B board that contains a RK3399 chip.  You can remotely access and control this board with my help. Now, I hope to know how to install OP-TEE on this board? 
@jbech-linaro Additionally, I hope you could makefiles (build.git) for installing OP-TEE on RK3399.

jbech-linaro, 2022-01-03:
> @jbech-linaro Additionally, I hope you could publish a makefile (build.git) for installing OP-TEE on RK3399. I can provide a RK3399 board for you!

Thanks for the offer. We do discuss RK devices from time to time. Personally I have limited time working on something like that right now, but since the RK3399 itself already is supported in OP-TEE, I think it should be fairly straight forward to make it supported from build.git. If you want to ship a board, please ping me at "joakim.bech        at l i n a r o.org".