OP-TEE/build #372: Optee - key generation and openssl

jonsmirl, 2019-05-28:
Is there a PKCS #11 implementation for optee around somewhere? Or is there another way to have user space OpenSSL use keys stored in optee?  I have optee installed on my system and I can run the samples/xtest.

My goal is to replace an existing ATECC608 encryption chip. That chip provides a PKCS #11 driver for OpenSSL. Via that driver OpenSSL uses the ATECC608 for the ECC key exchange. After it gets the DES key out of the ECC exchange the ATECC608 is no longer used.

How does initial key generation work? With the ATECC608 I tell it to generate a key pair and pass me back the public key. I have no access to the private key, there is no way to ask the chip to give it to me.

jonsmirl, 2019-05-28:
One solution to this would be to use cryptoauthlib. Cryptoauthlib is an open source PKCS #11 library provided by Microchip.  https://github.com/MicrochipTech/cryptoauthlib   It also has a mbed TLS wrapper.

When you dig down inside of it there are 24 commands that can be issued to their encryption chips.
https://github.com/MicrochipTech/cryptoauthlib/blob/master/lib/atca_command.h#L121

There is HAL driver layer for talking to the hardware:
https://github.com/MicrochipTech/cryptoauthlib/tree/master/lib/hal

An idea would be to write a HAL using libteec. Then implement those 24 commands (or as many as are relevant) inside a TA.  Implementing these command should not be too hard since you already have a crypto library and secure storage.

etienne-lms, 2019-05-28:
Hello @jonsmirl. I've started implementation of a PKCS#11 trusted application based on OP-TEE GPD TEE API for crypto. I have ported some crypto algo (AES/RSA) and PKCS#11 attributes management, but no elliptic curves yet (if that's want you need) and code is still immature: [sks ta](https://github.com/etienne-lms/optee_os/tree/sks/ta/secure_key_services) and [libsks](https://github.com/etienne-lms/optee_client/tree/sks/libsks). Feel free to look at, reuse, contribute... OP-TEE has many resources to help TA development. It's stalled work since few months, I will get back on it... later. I would happily get some feedback on both.

The package you refer to would be a great PKCS#11 software stack with OP-TEE. It would handle PKCS#11 formats/attributes while OP-TEE would bring the mbed TLS lib, secure storage and trusted application environment. Would remain to do the serialization of data, for non-secure client from/to trusted app interfacing, and for storing persistent objects (attributes/data) in OP-TEE secure storage.
However the source notice for this package is too restrictive for a flexible use, [license.txt](https://github.com/MicrochipTech/cryptoauthlib/blob/master/license.txt).

If you think of some other open source packages that could help bringing PKCS#11 into OP-TEE, please feel free to open discussion.

jonsmirl, 2019-05-28:
The license is GPLv2, that is too restrictive? All of those other ones don't matter since an optee version would not use those drivers.

Also, it seems like it would much easier to do all of the PKCS stuff in a user space library. Then only call into optee at the very lowest level.  For example only implement this api:

atcab_init
atcab_release
atcab_get_pubkey
atcab_read_pubkey
atcab_write_pubkey
atcab_genkey
atcab_genkey_base
atcab_random
atcab_sign

Is there an exsting TA that implements those?  And I do need the ECC support.

jonsmirl, 2019-05-28:
Here is the essential bit implementing the PKCS support, it is actually not that large.
https://github.com/MicrochipTech/cryptoauthlib/tree/pkcs11/lib/pkcs11

So maybe recreate that with a more permissive license, then call into an existing TA for the needed functions?

jonsmirl, 2019-05-28:
Random searching and I found this PKCS library, is the license ok? It is from IBM.

https://github.com/opencryptoki/opencryptoki

etienne-lms, 2019-05-29:
> The license is GPLv2, that is too restrictive? All of those other ones don't matter since an optee version would not use those drivers.

From the source files: 
`You may use this software and any derivatives exclusively with Microchip products.`

> Also, it seems like it would much easier to do all of the PKCS stuff in a user space library. Then only call into optee at the very lowest level.

Well... I would see 2 ways:
* Have OP-TEE doing only the crypto part (generate key, sign/verify), and have the PKCS#11 object attributes be handle in non-secure part. Then the TA part is somewhat reduced, especially if the goal is to support only signature verification. For signing and other operation expecting access to the private key, there shall be some extra code to handle access permission to the key.
* Have OP-TEE handling all PKCS#11 objects and sequences: data and attributes, object access permissions, ... are managed in secure userspace TA. Non-secure userspace library is only the wrapper/serializer for invocation of the TA. This is the scheme I've based my sks TA on. This brings quite a lot of PKCS#11 specifications into the TA implementation (session/login/object attributes/...).

jonsmirl, 2019-05-29:
Putting the minimal amount of code possible into the TA makes it easier to audit to ensure that it is doing the right thing.   It also provides for a more generalized solution, the same TA could be directly used by an OpenSSL plugs and all of PKCS#11 can be ignored.

I'm trying to see if the smart card people are interested in this....
https://github.com/OpenSC/OpenSC/issues/1690

jonsmirl, 2019-05-31:
Some comments from the smart card people.....

   -

   Yes, you could implement a "card driver" in OpenSC so that your security
   module is usable via all APIs that OpenSC implements (PKCS#11, minidriver,
   tokend).
   -

   Yes, there are other PKCS#11 implementations for different kind of
   "security modules", e.g.
   https://docs.aws.amazon.com/cloudhsm/latest/userguide/pkcs11-library-install.html
    or http://trousers.sourceforge.net/pkcs11.html and
   https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11, which you
   mentioned.
   -

   The OpenSC projects maintains libp11, which implements engine_pkcs11 for
   OpenSSL. (However, OpenSSL has started to add native PKCS#11 support
   lately.)
   -

   There is some PKCS#11 example code in pkcs11-tool.c or in src/p11test.

jbech-linaro, 2019-06-03:
> An idea would be to write a HAL using libteec. Then implement those 24 commands (or as many as are relevant) inside a TA.

Interesting idea and this is what I also had in mind when giving my Secure96 presentation a while ago ([see the third bullet](https://youtu.be/9RlCkBK4ILQ?t=1251)).

Related, but not strictly related to your question ...  the code for Secure96 is located at my GitHub account, [here](https://github.com/jbech-linaro/secure96), I did the initial work for ATSHA204A, but later on @michpappas did numerous changes and contributions which finalized the ATSHA204A implementation, he also added support for the ATECC508A.

jonsmirl, 2019-06-03:
I guess my comment over on optee-examples wasn't clear. Why don't you use the existing Android keymaster TA on Linux/buildroot to implement secure key storage?  The TA does not need to change.  What is needed is a de-Androidified plug-in for OpenSSL.  Now you only need to maintain a single TA.

https://github.com/linaro-swg/optee_examples/issues/60#issuecomment-498116327

wamserma, 2020-01-14:
This is worked on in https://github.com/OP-TEE/optee_os/pull/2732.

closed 2020-01-20 by github-actions[bot]