OP-TEE/build #376: `make run` stops with `no rule for core/include/mm/file.h` for qemu

sheepNo, 2019-06-28:
I followed this https://optee.readthedocs.io/building/devices/qemu.html and ran `make run` after ``make -j `nproc` ``. I ended up with the following error:

```bash
user@LAPTOP:~/op_tee_project/build$ make run
make -C /home/user/op_tee_project/build/../optee_os O=out/arm CROSS_COMPILE=" /home/user/op_tee_project/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-" CROSS_COMPILE_core=" /home/user/op_tee_project/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-" CROSS_COMPILE_ta_arm64=/home/user/op_tee_project/build/../toolchains/aarch64/bin/aarch64-linux-gnu- CROSS_COMPILE_ta_arm32=/home/user/op_tee_project/build/../toolchains/aarch32/bin/arm-linux-gnueabihf- CFG_TEE_CORE_LOG_LEVEL=3 DEBUG=1 CFG_TEE_BENCHMARK=n PLATFORM=vexpress-qemu_virt
make[1] : on entre dans le répertoire « /home/user/op_tee_project/optee_os »
  CHK     out/arm/conf.mk
  CHK     out/arm/include/generated/conf.h
make[1]: ***  Aucune règle pour fabriquer la cible « core/include/mm/file.h », nécessaire pour « out/arm/core/include/generated/.asm-defines.s ». Arrêt.
make[1] : on quitte le répertoire « /home/user/op_tee_project/optee_os »
common.mk:385 : la recette pour la cible « optee-os-common » a échouée
make: *** [optee-os-common] Erreur 2
```

Indeed, this is no `file.h` there:
```bash
user@LAPTOP:~/op_tee_project$ ls optee_os/core/include/mm/
tee_mm.h  tee_mmu.h  tee_mmu_types.h
```

What am I missing? Do I need to make optee_os separately beforehand?

jforissier, 2019-06-28:
Hello @sheepNo,

It could be that your `optee_os` directory is "dirty", in the sense that you have made a previous build, then updated the source code without first cleaning the build.

Try `rm -rf ../optee_os/out` and `make run` again.

sheepNo, 2019-06-28:
Thanks for the quick reply, I did make clean before seeing your message then did what you proposed too and it worked.

closed 2019-06-28 by sheepNo