OP-TEE/build #378: Perf syscall cannot measure accurate PMU reading from main.c in hello_world example

Xinyao90, 2019-07-19:
Hi,

I want to measure the number of HW instructions used by TEEC_InvokeCommand() in the hello_world example. As TEEC_InvokeCommand() calls another function, inc_value(), in hello_world_ta.c, the idea is to make changes to inc_value() so that I could get different number of HW instructions. No matter what changes I made in hello_world_ta.c, however, the number of HW instructions measured from main.c changes little. It seems to me the measured instructions do not count the instructions in inc_value().

Could you help me to figure out this issue?
Thanks a lot

[1](https://github.com/linaro-swg/optee_examples/blob/master/hello_world/host/main.c)
[2](https://github.com/linaro-swg/optee_examples/blob/master/hello_world/ta/hello_world_ta.c)

vchong, 2019-07-23:
@Xinyao90 Can you maybe include your changes here and explain how exactly you're doing the measurements?

**NOTE**: It's summer time and many people in the core team is on vacation, so expect delayed answers to both questions as well as feedback in pull requests!

Xinyao90, 2019-07-25:
Here is the changes I've made to measure function TEEC_InvokeComman() using perf_syscall. The original code is from 
https://github.com/linaro-swg/optee_examples/blob/master/hello_world/host/main.c

        ioctl(fd, PERF_EVENT_IOC_RESET, 0);
        ioctl(fd, PERF_EVENT_IOC_ENABLE, 0);
        res = TEEC_InvokeCommand(&sess, TA_HELLO_WORLD_CMD_DEC_VALUE, &op,
                                 &err_origin);
        ioctl(fd, PERF_EVENT_IOC_DISABLE, 0);
        read(fd, &count1[iter], sizeof(long long));

vchong, 2019-07-25:
@Xinyao90 `main.c`, running in normal world, has no visibility into `hello_world_ta.c`, running in secure world. That is the point of having a TEE, so that code and data loaded inside are protected from the REE wrt confidentiality and integrity.

There's some (limited) support in `optee_os` for benchmarking and profiling [1][2][3], but not sure if they're what you want.

[1] https://optee.readthedocs.io/debug/benchmark.html
[2] https://optee.readthedocs.io/debug/ftrace.html
[3] https://optee.readthedocs.io/debug/gprof.html

Xinyao90, 2019-07-26:
@vchong Thanks for your suggestion. I tried benchmark but it did not work since I kept getting the error message about failure to create mobj for timestamp buffer. 

I searched it and found
https://github.com/linaro-swg/optee_benchmark/issues/10

I should build optee_client, optee_os and linux all support benchmarks. I am very new to this area and I don't know how to build linux with benchmark support.  All I did was set the flag in `build/common.mk` by
`CFG_TEE_BENCHMARK               ?= y`
and then make the build by using
`$ cd <optee-project>/build
$ make CFG_TEE_BENCHMARK=y`

vchong, 2019-07-28:
@Xinyao90 `$ cd <optee-project>/build $ make CFG_TEE_BENCHMARK=y` should work. I'm not able to try to verify right now, but can you try from a new clean build? If you still get the error, please paste the details here.

Xinyao90, 2019-07-28:
I'm following this intruction https://optee.readthedocs.io/building/gits/optee_benchmark.html.
After $ cd <optee-project>/build $ make CFG_TEE_BENCHMARK=y, I invoke the benchmark in rpi3 buildroot by # benchmark ./optee_example_hello_world 
Listed below is the error message:

[BenchmaD/TC:? 0 tee_ta_init_session_with_context:477 Re-open TA 0b9a63b0-b4c6-4c85-a284-a228ef547b4e
rk] INFOD/TC:? 0 tee_ta_init_session_with_context:477 Re-open TA 0b9a63b0-b4c6-4c85-a284-a228ef547b4e
: 1. OpeD/TC:? 0 get_benchmark_memref:120 Benchmark: Sending back timestamp buffer paddr = 0x0
ning BenD/TC:? 0 tee_ta_close_session:398 csess 0x101618b0 id 2
chmark SD/TC:? 0 tee_ta_close_session:417 Destroy session
tatic TAE/TC:? 0 alloc_benchmark_buffer:73 Benchmark: can't create mobj for timestamp buffer
...
[BeD/TC:? 0 tee_ta_invoke_command:673 Error: ffff000c of 4
nchmark]D/TC:? 0 tee_ta_init_session_with_context:477 Re-open TA 0b9a63b0-b4c6-4c85-a284-a228ef547b4e
 INFO: 2D/TC:? 0 get_benchmark_memref:120 Benchmark: Sending back timestamp buffer paddr = 0x0
. AllocaD/TC:? 0 tee_ta_close_session:398 csess 0x101618b0 id 2
ting perD/TC:? 0 tee_ta_close_session:417 Destroy session
-core buffers, cD/TC:? 0 tee_ta_close_session:398 csess 0x10161990 id 1
ores detD/TC:? 0 tee_ta_close_session:417 Destroy session
ected = 1
[Benchmark] ERROR: TEEC_InvokeCommand: 0xffff000c

vchong, 2019-08-06:
@Xinyao90 I tried a test build (`make CFG_TEE_BENCHMARK=y`) and run (`benchmark optee_example_hello_world`) but got an `llegal instruction`. There was an error in one of the setup files [here](https://github.com/OP-TEE/build/pull/381), but that didn't help much. It could be that benchmark is broken atm. See https://github.com/OP-TEE/optee_os/issues/3073#issuecomment-516734297 for some details.

Xinyao90, 2019-08-08:
@vchong Thanks for your reply. It might be the reason.

closed 2020-01-19 by github-actions[bot]