OP-TEE/build #384: Pine A64

foreverska, 2019-09-04:
Hi,

I recognize that the platform for sunxi exists within optee-os.  Does anyone have a build script for building for the Pine A64?  If not, what is the recommendation for building to this platform?

Thanks

jforissier, 2019-09-04:
Hi @foreverska ,

I have no idea myself but perhaps @Amit-Radur can help?

```
$ ./scripts/get_maintainer.py -s ./core/arch/arm/plat-sunxi | grep A64
Amit Singh Tomar <amittomer25@gmail.com> [@Amit-Radur] (reviewer:AllWinner sun50i A64)
```

foreverska, 2019-12-12:
@jforissier 

Thanks for the lead.  I got in contact with Amit and they were instrumental in figuring out the switches for building OP-TEE.  But I am still attempting to build an image.  After much head-banging I've figured out the platform boots off SPL+UBoot so I am borrowing steps from poplar.  I inform u-boot of BL31 and ARM-TF is informed of BL32.  What I get out is a "u-boot-sunxi-with-spl.bin" which I had hoped would boot and It does but not as far as I had hoped.  Here is the output:

```
U-Boot SPL 2020.01-rc4-00244-gf39abbbc53 (Dec 12 2019 - 11:57:45 -0600)
DRAM: 2048 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.0(release):v2.0-507-g34efb683
NOTICE:  BL31: Built : 11:57:43, Dec 12 2019
NOTICE:  BL31: Detected Allwinner A64/H64/R18 SoC (1689)
NOTICE:  BL31: Found U-Boot DTB at 0x408e070, model: Pine64 LTS
NOTICE:  BL31: PMIC: Detected AXP803 on RSB.
```

It sits here forever.  What I assume is that it has attempted to jump to BL32 and fails somewhere within.  **Would OP-TEE print anything on Pine A64's UART0 if it was trying to boot?**  **Has anyone seen similar behavior out of another device?**  If I remove `SPD=opteed` from arm-tf options it continues attempting to boot into Linux so it would seem BL31 -> BL33 transition works.

foreverska, 2020-01-15:
BL32 is not currently loaded either the board or GRUB.  It can be loaded by the boards FEL mechanism but this is a human process currently.  I am working on polishing up a build script and documentation on the FEL process.

closed 2020-01-15 by foreverska

MQuaresma, 2020-04-02:
Hello @foreverska, have you had any luck booting OPTEE + Linux on the Pine A64+ using U-Boot+ATF? I've been trying to get it to work but using bootm but it fails when parsing the OPTEE header. Thanks in advance!

foreverska, 2020-04-02:
Unfortunately once I saw FEL worked that got me past what I needed from it at work and I stopped tracking the problem.  Loading BL32 via FEL is the only way I've gotten it to work.

MQuaresma, 2020-04-02:
Do you have any notes you could point me to to get that working? I need to boot OPTEE through ATF but that might be a good starting point.

MQuaresma, 2020-04-05:
@foreverska I've managed to boot OP-TEE as BL32 using U-Boot+ATF

foreverska, 2020-04-05:
Sorry for the delayed response.  This didn't feel like three days ago.  Did you get it with FEL or off a SDcard/storage?

MQuaresma, 2020-04-05:
No problem. I got it off an SDcard

foreverska, 2020-04-05:
Please do, this has some applicability to stuff farther down my work list.  Good work, thanks.

MQuaresma, 2020-04-11:
Hi again, here's the [link](https://mquaresma.github.io/pine/a64/2020/04/11/Setting-up-OPTEE-and-Linux-for-the-Pine-A64.html) to my post about the setup. If there's anything missing don't hesitate to ping me, as it might be useful to other people.