OP-TEE/build #387: Add build scripts for a new board: BD-SL-I.MX6

h1994st, 2019-10-04:
Hi,

I'm trying to add new build scripts for a development board ([BD-SL-I.MX6](https://boundarydevices.com/product/sabre-lite-imx6-sbc/)) and have some problems with "ARM Trusted Firmware".

Existing "ARM Trusted Firmware" does not support my board, which should be "imx6q". Is "ARM Trusted Firmware" required while setting up TrustZone in the development board? I remember that compiling OPTEE OS and Linux kernel should be enough for TrustZone installation. Correct me if I'm wrong.

I'm not sure whether this is the right place to ask questions about "ARM Trusted FIrmware", but I will appreciate it if anyone can provide suggestions. Thanks.

---

Besides, here are some information related to my board:

- Board: [BD-SL-I.MX6](https://boundarydevices.com/product/sabre-lite-imx6-sbc/)
- CPU:
    - CPU Name: NXP/Freescale i.MX6
    - CPU Type: ARM Cortex-A9
    - CPU Cores: 4 (Quad-Core)
    - CPU Clock (Max): 1GHz

etienne-lms, 2019-10-07:
Hello @h1994st,
Indeed one could boot up OP-TEE and a Linux kernel without going through TF-A (Arm Trusted Firmware). If so, you must use some other boot means (some other bootloader) to get OP-TEE and Linux images loaded in some executable RAM. Remember OP-TEE is not an image loader hence cannot be used to load your Linux kernel image, DTB, etc... in RAM.

Current armv7 OP-TEE boot scheme is:
* At reset, CPU executes in secure state;
* Boot stage loads OP-TEE image and non-secure world image in RAM;
* Boot stage branches to OP-TEE entry point (possibly filling in boot arguments into CPU registers)
* OP-TEE executes its initialization;
* OP-TEE switches CPU in non-secure state and branches to non-secure world entry point;

h1994st, 2019-10-11:
Hi, @etienne-lms. Thank you for your response! Right now, I am using u-boot.

BTW, what are the benefits of using TF-A? Just out of curiosity.

closed 2020-01-19 by github-actions[bot]

gagachang, 2020-04-07:
Hi, @h1994st 

I bought a BD-SL-i.MX6 board recently.
And I also noticed that there are not any build scripts related to this board in this repository.
It seems that you have ported the OP-TEE on the board successfully.
Because of some time pressure, would you mind helping me to accomplish this?

Any suggestions would be greatly appreciated.
Thank you!

h1994st, 2020-04-07:
> Hi, @h1994st
>
> I bought a BD-SL-i.MX6 board recently.
> And I also noticed that there are not any build scripts related to this board in this repository.
> It seems that you have ported the OP-TEE on the [...]

Hi @gagachang,

I did it manually, so there are no build scripts for now. : (

You can take a look at [mattlentz/secloak-kernel](https://github.com/mattlentz/secloak-kernel/). They were using a [Nitrogen6Q board](https://boundarydevices.com/product/nitrogen6x/), which is very similar to [BD-SL-i.MX6](https://boundarydevices.com/product/bd-sl-i-mx6/). Thus, their codebase is a good reference, if you want to port OP-TEE to BD-SL-i.MX6. You can start with their [commit histories](https://github.com/mattlentz/secloak-kernel/commits/master): https://github.com/mattlentz/secloak-kernel/commit/ed4de5faa1665fbabcd5085a9cb98719437c8e9d. They only have six new commits.

Hope this will help.

gagachang, 2020-04-08:
Hi, @h1994st 

Thanks for your reply! It's really a perfect tutorial.
I successfully built the uboot (from Boundary Devices), OP-TEE and linux (from linaro-swg/linux) images.
However, when I booted from the SD card, OP-TEE said that the Device Tree is invalid.

I used `imx6q-nitrogen6x.dts` device tree file from linaro-swg/linux repo.
May I ask you which linux distribution and device tree source file you used?
In uboot I also loaded the images at addresses as follows:

load mmc 0:1 0x10800000 zImage
load mmc 0:1 0x13000000 imx6q-nitrogen6x.dtb
load mmc 0:1 0x20000000 uTee

Please correct me if I was wrong.
Thank you!

h1994st, 2020-04-15:
Hi @gagachang,

I used the Linux kernel and u-boot from the Boundary Device
-
https://github.com/boundarydevices/linux-imx6/tree/boundary-imx_4.9.x_2.0.0_ga
- https://boundarydevices.com/imx-linux-kernel-4-9-for-nitrogen-platforms/
- https://boundarydevices.com/u-boot-v2018-07/

The addresses are the same as what I used. You can paste the log here so we
know what's going on.

Best,

gagachang, 2020-04-16:
Hi, @h1994st 

Thanks for suggestion.
I have successfully built the OP-TEE on my board last weekend.
For now I am using linaro-swg's upstream linux and the same u-boot as you mentioned.
If I had some problem in the future, I will try your build.

Thank you very much 😃 
Best regards.

ncpsa, 2020-05-06:
Hi @h1994st @gagachang ,
I am also using imx6dl, for me optee is booting well, but kernel is not booting.
If I disable the SMC call from kernel then kernel is booting but PSCI & secondary core enabling calls are failing. 
Please refer this https://github.com/OP-TEE/optee_os/issues/3816
Could you help?

Thanks in Advance.