OP-TEE/build #395: Build and run for QEMU failed!!!

nageshshamnur, 2019-10-18:
i followed the instructions to download and build the testsetup as explained in the below links:
https://optee.readthedocs.io/en/latest/building/gits/build.html
https://optee.readthedocs.io/en/latest/faq/faq.html#q-what-is-the-quickest-and-easiest-way-to-try-op-tee

I am installing the optee setup on Ubuntu 16.04. with the optee tag 3.5.0 (since with latest master there were compiling errors).
During "repo sync", it was unable to install some source git repos such as qemu and optee_test due to proxy issues and i so i clone those git repos manually and also i commented the lines for these git repos present in the folder .repo/manifests.xml. 

But finally during "make run", building is failing while launching QEMU with the following errors. Can someone help me with this issue. 

make[1]: Entering directory '/home/Nagesh/TEE/optee_project/build'
ln -sf /home/Nagesh/TEE/optee_project/build/../out-br/images/rootfs.cpio.gz /home/Nagesh/TEE/optee_project/build/../out/bin/

* QEMU is now waiting to start the execution
* Start execution with either a 'c' followed by <enter> in the QEMU console or
* attach a debugger and continue from there.
*
* To run OP-TEE tests, use the xtest command in the 'Normal World' terminal
* Enter 'xtest -h' for help.

Option "-t" is no longer supported in this version of gnome-terminal.
No protocol specified
Failed to connect to Mir: Failed to connect to server socket: No such file or directory
Option "-t" is no longer supported in this version of gnome-terminal.
No protocol specified
Unable to init server: Could not connect: Connection refused
Failed to parse arguments: Cannot open display:
Failed to connect to Mir: Failed to connect to server socket: No such file or directory
Unable to init server: Could not connect: Connection refused
Failed to parse arguments: Cannot open display:

jforissier, 2019-10-18:
> Cannot open display

X11 issue I suppose

jbech-linaro, 2019-10-18:
> I am installing the optee setup on Ubuntu 16.04. with the optee tag 3.5.0 (since with latest master there were compiling errors).

Highly unlikely, since the [automated tests](https://optee.mooo.com:5000) runs OP-TEE + QEMU several times every day. That is based on master on all OP-TEE gits except the one corresponding to the PR (which most often is based on master anyways).

> During "repo sync", it was unable to install some source git repos such as qemu and optee_test due to proxy issues and i so i clone those git repos manually and also i commented the lines for these git repos present in the folder .repo/manifests.xml.

More likely that the issue is here, since doing that manually means lots of manual steps and it's pretty easy to mess up things when doing it manually.

> Option "-t" is no longer supported in this version of gnome-terminal.

We have addressed that already [here](https://github.com/OP-TEE/build/commit/5dc030cd01dd6ea4e8e431c1189ef0cb9db73bb4). We cannot control behavior of gnome-terminal. So if people use the most recent version of gnome-terminal and check out an old revision of OP-TEE, then it will obviously not work. But back in the days when we created the 3.5.0 release, that flag was still supported and then it all worked.

nageshshamnur, 2019-10-18:
> > Cannot open display
> 
> X11 issue I suppose
after searching i got to know that Mir is the replacement for X11, but i couldn't just solve this issue even after trying multiple things. Any pointers would very useful. Thanks

jenswi-linaro, 2019-10-18:
It looks like you missing python-pyelftools

See https://optee.readthedocs.io/en/latest/building/prerequisites.html

jforissier, 2019-10-18:
@jenswi-linaro +1

But the error tells me this must not be optee_os master because now we have a much nicer error message thanks to https://github.com/OP-TEE/optee_os/commit/7ba36df9586.

nageshshamnur, 2019-10-18:
> > Cannot open display
> 
> X11 issue I suppose

issue got solved after running the make run gnome-windows application.

closed 2019-10-18 by nageshshamnur

PeiyanFlying, 2019-12-10:
> > > Cannot open display
> > 
> > 
> > X11 issue I suppose
> 
> issue got solved after running the make run gnome-windows application.

Excuse me, but how to 'make run' on gnome-windows application? Could you give me some detailed advice? I am really really new to linux~
Best!
Many thanks again!