OP-TEE/build #396: Seeed Studio Hikey960 boards

dkarnikis, 2019-11-04:
Hello,
I am thinking of buy this board [ARM Rock 960](https://gr.mouser.com/ProductDetail/Seeed-Studio/102110137?qs=%2Fha2pyFadujSa%252BngDKpGnz%2FeoY8EQSaoDRsS9JY9U1H8vC9uCfB2rg%3D%3D) or this [ARM HiKey 960 Dev](https://gr.mouser.com/ProductDetail/Seeed-Studio/102110117?qs=%2Fha2pyFaduhyMMoYKN0lzeIWL9pIG7a9oPHAiRXOnyeBRrWsQB5xyB9MXgmYQGsSbwr4Pi5WGGOVnbVPeFplbQ%3D%3D) Do you know if any of them are compatible with optee?

Thanks

jbech-linaro, 2019-11-04:
Both of them are. A list of all devices officially supported can be found here:
https://optee.readthedocs.io/en/latest/general/platforms.html#platforms-supported

The Rock960 (rk3399) was merged last week, the docs for it is on the way (see https://github.com/OP-TEE/optee_docs/pull/50).

dkarnikis, 2019-11-04:
Cool, thanks a lot !

closed 2019-11-04 by jbech-linaro