OP-TEE/build #400: make is giving error for Build OP-TEE for Raspberry Pi 3B+

amit22-verma, 2019-12-05:
~/devel/optee/build$ make -j2
make -C /home/advnet/devel/optee/build/../optee_os O=out/arm CFG_ARM64_core=y CROSS_COMPILE=" /home/advnet/devel/optee/build/../toolchains/aarch64/bin/aarch64-linux-gnu-" CROSS_COMPILE_core=" /home/advnet/devel/optee/build/../toolchains/aarch64/bin/aarch64-linux-gnu-" CROSS_COMPILE_ta_arm64=" /home/advnet/devel/optee/build/../toolchains/aarch64/bin/aarch64-linux-gnu-" CROSS_COMPILE_ta_arm32=" /home/advnet/devel/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-" CFG_TEE_CORE_LOG_LEVEL=3 DEBUG=0 CFG_TEE_BENCHMARK=n PLATFORM=rpi3
make[1]: Entering directory '/home/advnet/devel/optee/optee_os'
cd /home/advnet/devel/optee/build/../u-boot && \
	ARCH=arm64 \
	scripts/kconfig/merge_config.sh /home/advnet/devel/optee/build/../u-boot/configs/rpi_3_defconfig /home/advnet/devel/optee/build/kconfigs/u-boot_rpi3.conf
Using /home/advnet/devel/optee/build/../u-boot/configs/rpi_3_defconfig as base
Merging /home/advnet/devel/optee/build/kconfigs/u-boot_rpi3.conf
Value of CONFIG_SYS_TEXT_BASE is redefined by fragment /home/advnet/devel/optee/build/kconfigs/u-boot_rpi3.conf:
Previous value: CONFIG_SYS_TEXT_BASE=0x00080000
New value: CONFIG_SYS_TEXT_BASE=0x11000000

make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/home/advnet/devel/optee/u-boot'
  GEN     out/arm/core/include/generated/arm32_sysreg.h
  GEN     out/arm/core/include/generated/arm32_sysreg.S
  CHK     out/arm/conf.mk
  UPD     out/arm/conf.mk
  GEN     out/arm/core/ta_pub_key.c
Traceback (most recent call last):
  File "scripts/pem_to_pub_c.py", line 61, in <module>
    main()
  File "scripts/pem_to_pub_c.py", line 24, in main
    from Crypto.PublicKey import RSA
  File "/usr/lib/python3/dist-packages/Crypto/PublicKey/RSA.py", line 78, in <module>
    from Crypto import Random
  File "/usr/lib/python3/dist-packages/Crypto/Random/__init__.py", line 29, in <module>
    from Crypto.Random import _UserFriendlyRNG
  File "/usr/lib/python3/dist-packages/Crypto/Random/_UserFriendlyRNG.py", line 38, in <module>
    from Crypto.Random.Fortuna import FortunaAccumulator
  File "/usr/lib/python3/dist-packages/Crypto/Random/Fortuna/FortunaAccumulator.py", line 39, in <module>
    from . import FortunaGenerator
  File "/usr/lib/python3/dist-packages/Crypto/Random/Fortuna/FortunaGenerator.py", line 35, in <module>
    from Crypto.Util import Counter
  File "/usr/lib/python3/dist-packages/Crypto/Util/Counter.py", line 59, in <module>
    from Crypto.Util import _counter
ImportError: cannot import name '_counter' from 'Crypto.Util' (/usr/lib/python3/dist-packages/Crypto/Util/__init__.py)
mk/subdir.mk:160: recipe for target 'out/arm/core/ta_pub_key.c' failed
make[1]: *** [out/arm/core/ta_pub_key.c] Error 1
make[1]: Leaving directory '/home/advnet/devel/optee/optee_os'
common.mk:380: recipe for target 'optee-os-common' failed
make: *** [optee-os-common] Error 2
make: *** Waiting for unfinished jobs....
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  YACC    scripts/kconfig/zconf.tab.c
  LEX     scripts/kconfig/zconf.lex.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --alldefconfig Kconfig
#
# configuration written to .config
#
make[1]: Leaving directory '/home/advnet/devel/optee/u-boot'

jbech-linaro, 2019-12-05:
Probably missing dependency, see this thread:
https://github.com/OP-TEE/optee_os/issues/3441#issuecomment-559603379

amit22-verma, 2019-12-05:
Thanks. I installed using pip3 install crypto. Now make is successful.

closed 2019-12-05 by amit22-verma