OP-TEE/build #401: Master build broken for QEMU and QEMUv8

nageshshamnur, 2019-12-11:
the following commands when tried on either label:3.6.0 or master doesn't compile successfully. However label 3.5.0 builds and executes successfully.

command:
repo init -u https://github.com/OP-TEE/manifest.git
$ repo sync
$ cd build
$ make toolchains -j2
$ make run

Error info:
  CHK     out/arm/conf.mk
  CHK     out/arm/include/generated/conf.h
  CHK     out/arm/core/include/generated/asm-defines.h
  GEN     out/arm/core/ta_pub_key.c
Traceback (most recent call last):
  File "scripts/pem_to_pub_c.py", line 61, in <module>
    main()
  File "scripts/pem_to_pub_c.py", line 24, in main
    from Crypto.PublicKey import RSA
ModuleNotFoundError: No module named 'Crypto'
mk/subdir.mk:160: recipe for target 'out/arm/core/ta_pub_key.c' failed
make[1]: *** [out/arm/core/ta_pub_key.c] Error 1
make[1]: Leaving directory '/home/Nagesh/TEE/optee_project_64/optee_os'
common.mk:380: recipe for target 'optee-os-common' failed
make: *** [optee-os-common] Error 2

vchong, 2019-12-11:
See https://github.com/OP-TEE/optee_os/issues/3441.

If you are using latest Ubuntu, then
$ apt-get install python3-pycryptodome

If you are running Ubuntu 16.04 (and probably earlier), then
$ apt-get install python3-pip
$ python3 -m pip install --user pycryptodome

jforissier, 2019-12-11:
I think the prerequisite page in the OP-TEE documentation (https://optee.readthedocs.io/en/latest/building/prerequisites.html) is not quite right:
1. Python (2) is not needed anymore, all scripts use Python 3, so all `python-*` could go
2. `scripts/sign_encrypt.py` now uses Cryptodome not Crypto, so `python3-pycryptodome` is needed in addition to `python3-crypto`.

jbech-linaro, 2019-12-11:
Yeaps, after having ~10 issues opened about this, I think it's about time to fix the pre-reqs. I did a couple of test build in Docker from scratch ~2 weeks ago. I.e., a diff between our current 3.7.0 pre-reqs and "Ubuntu latest" (and the same for Ubuntu 16.04). To make a successful build from empty Ubuntu-docker images I had to change the following: 

**Latest Ubuntu**
```diff
--- Dockerfile-3.7.0    2019-11-27 13:05:47.212630787 +0100                                                          
+++ Dockerfile  2019-11-27 13:07:01.820430836 +0100                                                                  
@@ -1,4 +1,4 @@                                                                                                      
-FROM ubuntu:16.04                                                                                                   
+FROM ubuntu:latest                                                                                                  
 MAINTAINER Joakim Bech (joakim.bech@linaro.org)                                                                     
                                                                                                                     
 # This is needed on later Ubuntu distros to be able to install the i386                                             
@@ -42,11 +42,9 @@                                                                                                   
        make \                                                                                                       
        mtools \                                                                                                     
        netcat \                                                                                                     
-       python-crypto \                                                                                              
        python3-crypto \                                                                                             
-       python-pyelftools \                                                                                          
+       python3-pycryptodome \                                                                                       
        python3-pyelftools \                                                                                         
-       python-serial \                                                                                              
        python3-serial \                                                                                             
        rsync \                                                                                                      
        unzip \                                                                                                      
@@ -54,10 +52,13 @@                                                                                                  
        xdg-utils \                                                                                                  
        xterm \                                                                                                      
        xz-utils \                                                                                                   
-       zlib1g-dev                                                                                                   
-                                                                                                                    
-RUN python3 -m pip install --user pycryptodome                                                                      
-#RUN python3 -m pip install --upgrade pip                                                                           
+       zlib1g-dev \                                                                                                 
+       # extra for Docker only \                                                                                    
+       curl \                                                                                                       
+       cpio \                                                                                                       
+       git \                                                                                                        
+       python \                                                                                                     
+       wget 
```

**Ubuntu 16.04**
```diff
--- Dockerfile-3.7.0    2019-11-27 13:05:47.212630787 +0100                                                          
+++ Dockerfile  2019-11-27 10:13:35.924079646 +0100                                                                  
@@ -42,11 +42,9 @@                                                                                                   
        make \                                                                                                       
        mtools \                                                                                                     
        netcat \                                                                                                     
-       python-crypto \                                                                                              
+       python3-pip \                                                                                                
[... cut ...]
Not for our scripts, but for "repo" which still is a Python2 script, it is needed.

I think the conclusion is that the pre-req page can be updated with:
```diff
-       python-crypto \
-       python-pyelftools \
+      python \
+       python3-pycryptodome \
        python3-pyelftools \
+       python3-pip \
-       python-serial \
```

And an extraline saying that if you are on Ubuntu 16.04 or earlier version, then you need to run this since there is not apt package for it.
```
$ python3 -m pip install --user pycryptodome
```

jforissier, 2019-12-11:
@jbech-linaro looks good!

closed 2019-12-13 by nageshshamnur

jforissier, 2020-01-21:
@jbech-linaro 

> And an extraline saying that if you are on Ubuntu 16.04 or earlier version, then you need to run this since there is not apt package for it.
>
> ```
> $ python3 -m pip install --user pycryptodome
> ```

On 16.04 you need `pycryptodomex` (note the 'x'). Indeed, the pip3 package `pycryptodome` provides the `Crypto` module, in replacement of the default one. `pycryptodomex` provides `Cryptodome`, which is what we need now. The command that worked for me in an Ubuntu 16.04 VM is:
```
$ sudo -H pip3 install pycryptodomex
```

But, perhaps at some point we should just say we recommend Ubuntu >= 18 and not mention any workaround for older distros, since we're not using them on a daily basis so we cannot guarantee they work.

jbech-linaro, 2020-01-21:
> On 16.04 you need `pycryptodomex` (note the 'x'). Indeed, the pip3 package `pycryptodome` provides the `Crypto` module, in replacement of the default one. `pycryptodomex` provides `Cryptodome`, which is what we need now.

Alright, I used the one without 'x' when I [tried it in Docker 2 months ago](https://github.com/jbech-linaro/docker_optee/blob/cached-ubuntu-16.04/Dockerfile#L63).

> But, perhaps at some point we should just say we recommend Ubuntu >= 18 and not mention any workaround for older distros, since we're not using them on a daily basis so we cannot guarantee they work.

Indeed 2020 now, so Ubuntu 16.04 is becoming old (even though [Ubuntu 16.04 EOL](https://wiki.ubuntu.com/Releases) isn't until 2024).

jforissier, 2020-01-21:
> Alright, I used the one without 'x' when I tried it in Docker 2 months ago.

Hmmm... I suppose it must have been before https://github.com/OP-TEE/optee_os/commit/ababd72d2fd76cb2ded8e202b49db28d6545f6eb which changed `import Crypto` with `import Cryptodome`. The difference between `pycryptodome` and `pycryptodomex` is mentioned [here](https://pypi.org/project/pycryptodome/) -- well there is no difference except for the name of the module ;-)

jbech-linaro, 2020-01-21:
> Hmmm... I suppose it must have been before [OP-TEE/optee_os@ababd72](https://github.com/OP-TEE/optee_os/commit/ababd72d2fd76cb2ded8e202b49db28d6545f6eb) which changed `import Crypto` with `import Cryptodome`. T

Nope, I did the Docker stuff (on Nov 27, 2019) just to be able to sort out the issues we had with the newly Cryptodome stuff (Nov 25, 2019). But all this cryptodome stuff seems rather magic, so I don't really know what is the truth :smile: