OP-TEE/build #403: Unable to establish TCP connection with usermode networking

nageshshamnur, 2019-12-19:
Hi All,
   for my QEMU image, I am using the usermode networking (-netdev user,id=vmnic$(HOSTFWD) -device virtio-net-device,netdev=vmnic) as defined in common.mk and the QEMU come up and assign the IP address successfully. But however, if I run simple tcp_server on quest and tcp_client on the host using the server address as 127.0.0.1 with the port which is set in HOSTFWD, the packet doesn't seem to reach the TCP server and gets dropped at the guest. Any pointers would be helpful. I tried searching on optee links and google, but there was no solution to my problem.

closed 2020-01-25 by github-actions[bot]

becominginsane, 2021-03-04:
I'd like to remove the stale label and reopen the issue:
it is not currently possible to forward more ports in a straightforward manner.

I tried adding another network via netdev using the same device type, but the interface is down by default and when enabling it then it just won't connect.
I suggest changing [`HOSTFWD`](https://github.com/OP-TEE/build/blob/master/common.mk#L392) to append instead of simple assignment, like `QEMU_EXTRA_ARGS` does, so one can set `HOSTFWD` when running VM and put the forwarding they need!

jforissier, 2021-03-04:
Hi @becominginsane,

> I suggest changing [`HOSTFWD`](https://github.com/OP-TEE/build/blob/master/common.mk#L392) to append instead of simple assignment, like `QEMU_EXTRA_ARGS` does, so one can set `HOSTFWD` when running VM and put the forwarding they need!

That sounds like a good idea, would you like to create a PR for that?