OP-TEE/build #404: How to change SLiRP networking with TAP networking

nageshshamnur, 2019-12-19:
Hi Group,
   I found that the OPTEE QEMU environment uses the SLiRP user-mode networking for host connectivity. I wanted to set up network connectivity of OPTEE QEMU with host network, and so tried replacing the SLiRP with TAP networking, using various methods as defined in different blogs such as:
https://blog.elastocloud.org/2015/07/qemukvm-bridged-network-with-tap.html#comment-form
https://brezular.com/2011/06/19/bridging-qemu-image-to-the-real-network-using-tap-interface/
https://wiki.qemu.org/Documentation/Networking
https://wiki.archlinux.org/index.php/QEMU#Tap_networking_with_QEMU

but none of them seems to work, QEMU VM comes up but doesn't list the eth0. Can someone point me to the right way of replacing the SLiRP with TAP based networking? Thanks.

jbech-linaro, 2019-12-23:
This is not exactly an answer, but have you been able to get TAP based networking to work otherwise with QEMU? I.e., do you have another setup where this is working? I've been playing with TAP stuff in the past (non OP-TEE related) and I've always got stuck on something. I.e., seems complicated to get everything working.

nageshshamnur, 2019-12-23:
Hi jbech,
 Thanks for the reply. Yes, in the past, I have created QEMU for x86 arch in which I was successful in creating 2 TAP interface on the host and 2 veth on the guest and able to communicate successfully.

jbech-linaro, 2019-12-23:
Ok, so, since enabling TAP on QEMU isn't really about OP-TEE as such, then one idea might be to look into the various flags that you find Buildroot. I.e., search for "TAP" related Kconfig flags there and then see if any of those makes any difference. If you find a working combination, then we can set those flag from this (build.git) later on.

nageshshamnur, 2019-12-24:
I did try different options including the method you suggested, to my bad luck nothing has worked. Hence posted on this list.

closed 2020-01-29 by github-actions[bot]