OP-TEE/build #415: repo init failing

divneil, 2020-03-03:
Hi,

I am trying to repo init the latest optee version (3.8.0) and repo init is failing:
```
  File "/home/divneil/rpi3-optee-ref/.repo/repo/manifest_xml.py", line 1077, in _ValidateFilePaths
    '<%s> invalid "src": %s: %s' % (element, src, msg))
error.ManifestInvalidPathError: <linkfile> invalid "src": ../toolchains/aarch64/bin/aarch64-linux-gnu-gdb: bad component: ..
```

Can you please help. I was earlier on 3.5.0 and it is also now failing with the same error.

Regards,
Divneil

vchong, 2020-03-03:
This thread might help: https://github.com/OP-TEE/manifest/issues/157

jbech-linaro, 2020-03-03:
Besides that thread, the that particular issue was resolved in:
https://github.com/OP-TEE/manifest/commit/017ade90e840a9da45c775e7146f322889986ca7

Since we've had this being asked more than once, I think I'll update the docs trying to clarify this, since it's not obvious what is happening and why.

divneil, 2020-03-03:
Thanks @vchong and @jbech-linaro. For now, I have moved forward by using the information given in https://github.com/OP-TEE/manifest/issues/157#issuecomment-593272376.

Quoting from the same comment the steps which worked for me.
```
....
$ cd .repo/repo/
$ git checkout v1.13.11 
....
```

closed 2020-03-03 by divneil