OP-TEE/build #418: snapshots.linaro.org/android/android-hikey960-optee-p can not bootup

liuyunli, 2020-04-24:
i have try the below images, but none of them could boot the hikey960.
    https://snapshots.linaro.org/android/android-hikey960-optee-p/latest
    https://snapshots.linaro.org/android/android-hikey960-optee-p/236
    https://snapshots.linaro.org/android/android-hikey960-optee-p/269
    https://snapshots.linaro.org/android/android-hikey960-optee-p/226

    blocked by:
[uart.log](https://github.com/OP-TEE/build/files/4527447/uart.log)

    [  854.275632] ServiceManagement: Waited one second for android.hardware.keymaster@3.0::IKeymasterDevice/default. Waiting another...[ 

    you can check the tail of the attachment.

    Thanks

vchong, 2020-04-24:
@liuyunli Yes, this is a known issue due to failure mounting a dedicated partition for keymaster purposes. We'll try to find time/resources to get this fixed but don't have an ETA atm. In the meantime, can you try the images here [1] to see if they work? These images have some of the keymaster features disabled. If not, are you able to try building your own images based on [2]?

Btw, can you let us know please your use case(s) for OP-TEE on AOSP?

[1] https://people.linaro.org/~victor.chong/prebuilt/pie/380/hikey960
[2] https://optee.readthedocs.io/en/latest/building/aosp/aosp.html

liuyunli, 2020-04-26:
@vchong  thank you for your feedback. I have tryed the images "pie/380/hikey960" and the android bootup. but unable lunch to desktop, screen blocked in "Android" logo . adb tools is able to use.  
unable to run "xtest" and "tee-supplicant" 
hikey960:/vendor/bin # ./xtest
Run test suite with level=0

TEE test application started with device [(null)]
Failed to open TEE context: 0xffff0008
255|hikey960:/vendor/bin # ./te
tee              tee-supplicant
255|hikey960:/vendor/bin # ./tee-supplicant &
[1] 2707
hikey960:/vendor/bin # ERR [2707] TEES:main:694: failed to find an OP-TEE supplicant device

[1] + Done (1)             ./tee-supplicant

you can check the logs:
[380log.zip](https://github.com/OP-TEE/build/files/4534932/380log.zip)

i can not build my own images based on you link. because of the china-GFW, I cannot apply the patch which need the connect to android.googlesource.com. Can you tar your source code to  "https://people.linaro.org/~victor.chong/" and then i can get them and build myself?

I am working for a company of fingerprint chip used  for mobile phone. So I need a tee environment to build TA and test on aosp. I find the Hikey960 board is the only one I can choose which both closely update with optee and AOSP. Do you have any other advice about the board select?

vchong, 2020-04-29:
tee-supplicant failed to start for some reason. Let us investigate and get back to you again.

As for board selection, you're right. Afaik, the hikey620 and hikey960 boards are the only publicly available platforms that closely update with optee and AOSP. Both boards are hard to get these days, but so far there's no alternative.

vchong, 2020-05-01:
@liuyunli Sorry for the delay. I've disabled file-based disk encryption (FBE) which doesn't have all the required components to work for now. Please try https://people.linaro.org/~victor.chong/prebuilt/pie/381/hikey960 and let us know if it works. If it does, do you still need the source code for building? Not sure how long it'll take to compress and upload such a big tarball.

Thanks for describing your use case! The reason we asked was also because we wanted to make sure disabling FBE wouldn't affect your use case. Finally, please consider upstreaming your fingerprint CA/TA to https://github.com/linaro-swg/kmgk when you're done. It'll be very useful and helpful for us as well.

liuyunli, 2020-05-01:
[381.zip](https://github.com/OP-TEE/build/files/4563732/381.zip)
@vchong  
Thank you very much for the 381 link.After my test, I found that I still couldn't get into the system and **screen blocked in "Android" logo**. A good news is that, after adb  running "tee-supplicant &", can run "xtest" with result:
"24521 subtests of which 0 failed
97 test cases of which 0 failed
0 test cases were skipped
TEE test application done!"

please help to check the blocked reason continue. 

By the way,  l think i still need the source code for building,  command with "--depth=1" will save a lot of space.
Thanks.

vchong, 2020-05-06:
@liuyunli You're welcome, and sorry for the issue. We normally run and test the command line prompt only since that's all that's needed for xtest. In any case, we don't really have any good news for you atm. It looks like some new selinux rules have crept into the code base that are blocking proper boot, but even with selinux disabled, it still fails to boot to the home screen. It might be a kernel issue, but we don't plan to spend any more time on this since it doesn't seem to be really an OP-TEE issue. Moreover, the hikey960 P build has always been flaky anyway, so we're planning to upgrade to Q, but it might take some time.

Even with depth 1 we're still talking several tens of GBs. We don't know of a place where we can upload such a big file. Don't you have a vpn service you can use somewhere? Are you able to consider other sources of development in the meantime? E.g. Qemu v8 [1] which is a lot smaller in size in terms of download. Or if you really need a hikey960 build environment, you can maybe try using an older build for now until the next version is released? Please refer to [2] for details.

[1] https://optee.readthedocs.io/en/latest/building/devices/qemu.html#qemu-v8
[2] https://discuss.96boards.org/t/hikey960-4gb-ram-board-aosp-optee-booting-issue/9654

liuyunli, 2020-05-20:
@vchong  how is the progress status of  "planning to upgrade to Q" . As optee and ta&ca software developing is under the framework of android platform. i think the Q or P or R is not very necessary. developer may more care about the thing such as  "boot to the home screen".

vchong, 2020-05-20:
@liuyunli We're currently working on building optee_os and bootloader with clang, as gcc has been removed from AOSP. If you just want something that boots to the homescreen, you can try the 3.4 build as specified in [2].

liuyunli, 2020-06-21:
@vchong could you share with the progress status of "planning to upgrade to Q".
Thank you very much!

vchong, 2020-06-22:
@liuyunli We've a set of experimental patches that can build with clang without errors on AOSP master (cloned about a month ago, maybe older), but booting fails due to error starting healthd, so again this doesn't appear to be OP-TEE related. It may very well be an issue with the master branch and just require an update to the sources. That remains to be verified, but we haven't been able to do it due to other priorities. The best we can offer right now is maybe a manifest for this build updated to the latest master branch, but note that this will be very experimental, i.e. patches are work-in-progress and builds not tested. You can try building with this manifest if you want. It'll help us with verification as well. Does that sound like something you'd be willing to try?

liuyunli, 2020-06-23:
I'd like to have a try. you can push the code to a developing branch. I will test it  in leisure times.

liuyunli, 2020-07-01:
@vchong  Have you solved the problem?  Now the status of this issue is closed.

jbech-linaro, 2020-07-01:
> @vchong Have you solved the problem?

I believe @vchong is still working with it. But from what I understand things have changed in upstream AOSP that made it a bit complicated to make it work again (boot order, run-time dependencies).

> Now the status of this issue is closed.

That's the bot that closed it automatically, not @vchong .

liuyunli, 2020-07-01:
Thanks @jbech-linaro .and thanks the hard working of @vchong

vchong, 2020-07-04:
@liuyunli Sorry for the delay. Even just creating a temp build for testing purposes was a humongous boatload more work than expected! Hard work but not much to show for it. :s

https://github.com/linaro-swg/optee_android_manifest/tree/master-dirty#build-guide

- AOSP 11-12
- Kernel 4.14 (can try building with 5.4 if you want, extra steps needed, but we haven't tried so don't really know what will happen, but for sure xtest SDP tests will fail)
- Boot to home screen. Ignore the initial home screen error message about problem with the device and contacting the manufacturer. It happens with stock AOSP builds too so not sure what's going on there.
- No adb over usb, but over tcp is ok
- xtest doesn't pass everything yet. Some might just be Android.mk not being updated with newer tests, but all the PKCS11 ones also don't pass.

closed 2020-08-09 by github-actions[bot]