OP-TEE/build #422: Building for qemu_v8 failed

giuseppeguida90, 2020-05-12:
Hi all,
I followed the steps described in 
https://optee.readthedocs.io/en/latest/building/gits/build.html#get-and-build-the-solution 
and ran:

> repo init -u https://github.com/OP-TEE/manifest.git -m qemu_v8.xml 
> repo sync -j4 --no-clone-bundle
> cd <optee-project>/build
> make toolchains
> make -j2 toolchains
> make -j `nproc`

After running the last command (and waiting a very long time for compilation to complete), I received the following error:

Python build finished, but the necessary bits to build these modules were not found:
_bsddb             bsddb185           dbm             
dl                 imageop            sunaudiodev     
To find the necessary bits, look in setup.py in detect_modules() for the module's name.

Failed to build these modules:
_bisect            _codecs_iso2022    _collections    
_csv               _ctypes            _ctypes_test    
_elementtree       _functools         _hashlib        
_heapq             _hotshot           _io             
_json              _locale            _lsprof         
_multibytecodec    _multiprocessing   _random         
_socket            _struct            _testcapi       
array              audioop            binascii        
cmath              cPickle            crypt           
cStringIO          datetime           fcntl           
future_builtins    grp                itertools       
linuxaudiodev      math               mmap            
operator           parser             pyexpat         
readline           resource           select          
spwd               strop              syslog          
termios            time               unicodedata     
zlib                                                  

make[3]: *** [Makefile:544: sharedmods] Error 1
make[2]: *** [package/pkg-generic.mk:241: /home/giuseppe/optee/out-br/build/host-python-2.7.15/.stamp_built] Error 2
make[1]: *** [Makefile:23: _all] Error 2
make[1]: Leaving directory '/home/giuseppe/optee/out-br'
make: *** [common.mk:249: buildroot] Error 2
[build.log](https://github.com/OP-TEE/build/files/4616925/build.log)

Can anybody help me out?
Thanks

jforissier, 2020-05-12:
Hi @giuseppeguida90,

I know it won't help you much... but I tried a clean build and it worked fine here :open_mouth:
So there must be something wrong with your environment. Which Linux distribution are you using?

giuseppeguida90, 2020-05-12:
Hi @jforissier,
I am running Ubuntu 20.04.

jforissier, 2020-05-12:
Same here, 20.04. Do you have all prerequisite dependencies installed? Any network error when downloading stuff perhaps?

giuseppeguida90, 2020-05-12:
Yes, I have followed the building instructions step by step, i have double checked. The only missing details is that i am running Ubuntu 20.04 as a virtual machine, i don't know if this can be a problem.

etienne-lms, 2020-05-13:
For info, I've built qemu_v8 from scratch on my ubuntu 18.04. Took a while but works fine.

giuseppeguida90, 2020-05-13:
I've reinstalled everything on clean Ubuntu 18.04.4 virtual machine, it worked like charm.

closed 2020-05-13 by giuseppeguida90

etienne-lms, 2020-05-13:
good news. enjoy.