OP-TEE/build #423: Build qemu_v7 with op-tee version 3.6.0 succeed,but can not run

chenlifu2015, 2020-05-13:
hi, experts
    I have download the source of  op-tee project with version 3.6.0，and I use following command to build the project:
   make -f qemu.mk (The building process is successful)
   then I run the command  : make -f qemu.mk run-only,after that  secure world is loading normally, but I get the error message in the normal world:  
    [LV2] VA:0x79800000 PA:0x79800000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x79a00000 PA:0x79a00000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x79c00000 PA:0x79c00000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x79e00000 PA:0x79e00000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7a000000 PA:0x7a000000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7a200000 PA:0x7a200000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7a400000 PA:0x7a400000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7a600000 PA:0x7a600000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7a800000 PA:0x7a800000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7aa00000 PA:0x7aa00000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7ac00000 PA:0x7ac00000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7ae00000 PA:0x7ae00000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7b000000 PA:0x7b000000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7b200000 PA:0x7b200000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7b400000 PA:0x7b400000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7b600000 PA:0x7b600000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7b800000 PA:0x7b800000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7ba00000 PA:0x7ba00000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7bc00000 PA:0x7bc00000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7be00000 PA:0x7be00000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7c000000 PA:0x7c000000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7c200000 PA:0x7c200000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7c400000 PA:0x7c400000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7c600000 PA:0x7c600000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7c800000 PA:0x7c800000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7ca00000 PA:0x7ca00000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7cc00000 PA:0x7cc00000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7ce00000 PA:0x7ce00000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7d000000 PA:0x7d000000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7d200000 PA:0x7d200000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7d400000 PA:0x7d400000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7d600000 PA:0x7d600000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7d800000 PA:0x7d800000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7da00000 PA:0x7da00000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7dc00000 PA:0x7dc00000 size:0x200000 MEM-RW(PRIV)-NOACCESS(USER)-EXEC(PRIV)-EXEC(USER)-NS
    [LV2] VA:0x7de00000 size:0x200000
    [LV2] (16 invalid descriptors omitted)
  [LV1] VA:0x80000000 size:0x40000000
  [LV1] (1 invalid descriptors omitted)
NOTICE:  BL2: v1.5(debug):v1.5
NOTICE:  BL2: Built : 00:55:08, May 13 2020
INFO:    BL2: Doing platform setup
INFO:    BL2: Loading image id 4
VERBOSE: Using Memmap
WARNING: Firmware Image Package header check failed.
VERBOSE: Trying alternative IO
VERBOSE: Using Semi-hosting IO
INFO:    Loading image id=4 at address 0xe100000
INFO:    Image id=4 loaded: 0xe100000 - 0xe10001c
INFO:    OPTEE ep=0xe100000
INFO:    OPTEE header info:
INFO:          magic=0x4554504f
INFO:          version=0x2
INFO:          arch=0x0
INFO:          flags=0x0
INFO:          nb_images=0x1
INFO:    BL2: Loading image id 21
VERBOSE: Using Memmap
WARNING: Firmware Image Package header check failed.
VERBOSE: Trying alternative IO
VERBOSE: Using Semi-hosting IO
INFO:    Loading image id=21 at address 0xe100000
INFO:    Image id=21 loaded: 0xe100000 - 0xe15d298
INFO:    BL2: Skip loading image id 22
INFO:    BL2: Loading image id 5
VERBOSE: Using Memmap
WARNING: Firmware Image Package header check failed.
VERBOSE: Trying alternative IO
VERBOSE: Using Semi-hosting IO
INFO:    Loading image id=5 at address 0x60000000
INFO:    Image id=5 loaded: 0x60000000 - 0x600053ac
NOTICE:  BL1: Booting BL32
INFO:    Entry point address = 0xe100000
INFO:    SPSR = 0x1d3
VERBOSE: Argument #0 = 0xe300000
VERBOSE: Argument #1 = 0x0
VERBOSE: Argument #2 = 0x40000000
VERBOSE: Argument #3 = 0x0
Using QEMU provided DTB at 0x40000000
Relocating DTB for kernel use at 0x483bb000
Check "r < 0": bios/main.c:155

Any reply is appreciated，thanks

jbech-linaro, 2020-05-13:
That's weird, QEMU-v7 is part of our CI/regression testing on all patches sent here to GitHub. Also QEMU-v7 is my go-to setup and I cannot recall seeing these issues. Any particular reason why you want to use 3.6.0? Can't you use master? (default.xml manifest).

etienne-lms, 2020-05-13:
@chenlifu2015, did you fetch the release from a `repo init ...` command specific to qemu-v7? default.xml as @jbech-linaro suggests, or another qemu-v7 compliant manifest?

jforissier, 2020-05-13:
Works for me, using the official `3.6.0` tag for the manifest. Please see the commands below.

Note that the `repo` tool has to be patched in order to be able to clone version 3.6.0, since Google apparently decided it was OK to break other people's build scripts without providing a fallback :open_mouth: -- patch attached: [repo.patch.txt](https://github.com/OP-TEE/build/files/4622275/repo.patch.txt)

```
$ mkdir optee_repo_qemu-3.6.0
$ cd optee_repo_qemu-3.6.0
$ curl https://storage.googleapis.com/git-repo-downloads/repo >repo
$ chmod +x repo
$ ./repo init -u https://github.com/OP-TEE/manifest.git -b 3.6.0
...
error.ManifestInvalidPathError: <linkfile> invalid "src": ../toolchains/aarch32/bin/arm-linux-gnueabihf-gdb: bad component: ..
$ cd .repo/repo/
$ patch -p1 <../../repo.patch
patching file manifest_xml.py
patching file project.py
$ cd ../..
$ ./repo init -u https://github.com/OP-TEE/manifest.git -b 3.6.0
...
repo has been initialized ...
$ ./repo sync -j10
$ cd build
$ make -j2 toolchains
$ make -j$(nproc)
...
$ echo $?
0
$ make check-only
...
Starting QEMU... done, guest is booted.
Running: xtest...
##################################################
##################################################
##################################################
##################################################
##################################################
##################################################
##################################################
##################################################
##################################################
##################################################
############################################
Status: PASS (544 test cases)
$ cd ..
$ grep "OP-TEE version" out/bin/serial1.log 
I/TC: OP-TEE version: 3.6.0 #1 Wed 13 May 2020 01:26:50 PM UTC arm
```

chenlifu2015, 2020-05-14:
Thanks a lot,   @jbech-linaro I will try the master branch，@etienne-lms  yes，I use the default.xml when I init the repo project，@jforissier，I will follow your steps and try again，thanks

closed 2020-06-20 by github-actions[bot]