OP-TEE/build #424: build regression for Qemu v8?

vchong, 2020-05-19:
There seems to be a build regression on current master branch?

```
  LD      out/5b9e0e40-2636-11e1-ad9e-0002a5d5c51b.elf
/<src_root>/build/../toolchains/aarch64/bin/aarch64-linux-gnu-ld.bfd: cannot find -los_test
/<src_root>/optee_os/out/arm/export-ta_arm64/mk/link.mk:103: recipe for target 'out/5b9e0e40-2636-11e1-ad9e-0002a5d5c51b.elf' failed
make[3]: *** [out/5b9e0e40-2636-11e1-ad9e-0002a5d5c51b.elf] Error 1
package/pkg-generic.mk:238: recipe for target '/<src_root>/out-br/build/optee_test-1.0/.stamp_built' failed
make[2]: *** [/<src_root>/out-br/build/optee_test-1.0/.stamp_built] Error 2
Makefile:23: recipe for target '_all' failed
make[1]: *** [_all] Error 2
make[1]: Leaving directory '/<src_root>/out-br'
common.mk:244: recipe for target 'buildroot' failed
make: *** [buildroot] Error 2
```

jbech-linaro, 2020-05-19:
FYI, I'm doing a test build from scratch, I'll let you know once done. I think the stamp-failures I've seen in the past disappeared if I threw away the out-br folder.

jenswi-linaro, 2020-05-19:
I just did a build from scratch. Builds OK here.

jbech-linaro, 2020-05-19:
I've tried from scratch (rm -rf && repo ...) and I've seen two issues so far:
- Missing "texlive" package (probably doesn't block and an apt-get fixed it).
- QEMU compiler issues (seen on both tries)
```bash
In file included from /media/jbech/TSHB_LINUX/devel/optee_projects/qemu_v8/qemu/include/qemu/timer.h:5,
                 from /media/jbech/TSHB_LINUX/devel/optee_projects/qemu_v8/qemu/include/qemu/timed-average.h:29,
                 from /media/jbech/TSHB_LINUX/devel/optee_projects/qemu_v8/qemu/include/block/accounting.h:28,
                 from /media/jbech/TSHB_LINUX/devel/optee_projects/qemu_v8/qemu/include/block/block_int.h:27,
                 from block/file-posix.c:29:
/usr/include/linux/swab.h: In function ‘__swab’:
/media/jbech/TSHB_LINUX/devel/optee_projects/qemu_v8/qemu/include/qemu/bitops.h:20:34: warning: "sizeof" is not defined, evaluates to 0 [-Wundef]
 #define BITS_PER_LONG           (sizeof (unsigned long) * BITS_PER_BYTE)
                                  ^~~~~~
/media/jbech/TSHB_LINUX/devel/optee_projects/qemu_v8/qemu/include/qemu/bitops.h:20:41: error: missing binary operator before token "("
 #define BITS_PER_LONG           (sizeof (unsigned long) * BITS_PER_BYTE)
                                         ^
make[1]: *** [/media/jbech/TSHB_LINUX/devel/optee_projects/qemu_v8/qemu/rules.mak:69: block/file-posix.o] Error 1
make[1]: Leaving directory '/media/jbech/TSHB_LINUX/devel/optee_projects/qemu_v8/qemu'
make: *** [Makefile:111: qemu] Error 2
```

So we have three different states:
- Jens: OK
- Vee: stamp issues
- Joakim: QEMU errors.

jenswi-linaro, 2020-05-19:
I'm building on Ubuntu 20.04.

jbech-linaro, 2020-05-19:
> I'm building on Ubuntu 20.04.
> 

I'm on Debian. Will give a try on my Ubuntu laptop also.

> ```shell
> %gcc --version
> gcc (Ubuntu 9.3.0-10ubuntu2) 9.3.0
> ```

That's not the toolchain version we get when running `make toolchains`. I.e., you're running a modified setup. This is what you get from `make toolchains`.

```
jbech@goby /media/jbech/TSHB_LINUX/devel/optee_projects/qemu_v8 
$ ./toolchains/aarch64/bin/aarch64-linux-gnu-gcc --version
aarch64-linux-gnu-gcc (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36)) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
```

jforissier, 2020-05-19:
FWIW, good here on Ubuntu 19.10
```
cd /tmp
mkdir optee-qemuv8 && cd optee-qemuv8
repo init -u https://github.com/OP-TEE/manifest.git -m qemu_v8.xml
repo sync -j10
cd build
make -j2 toolchains
make -j10 check
...
Starting QEMU... done, guest is booted.
Running: xtest...
##################################################
##################################################
##################################################
##################################################
##################################################
##################################################
##################################################
##################################################
##################################################
##################################################
##################################################
#####################################
Status: PASS (587 test cases)
```

jenswi-linaro, 2020-05-19:
> That's not the toolchain version we get when running make toolchains. I.e., you're running a modified setup. This is what you get from make toolchains.

QEMU is compiled using the host gcc, not a cross compiler.

jbech-linaro, 2020-05-19:
> QEMU is compiled using the host gcc, not a cross compiler.

You're right, I didn't pay attention to the output from you. But as said, I'm gonna try it using my Ubuntu machine, since that is what we mention in the pre-reqs.

jbech-linaro, 2020-05-19:
On my Ubuntu machine (`Ubuntu 18.04.4 LTS`) our QEMU v8 also works with no issues. So my QEMU issues seems to be related to the Debian tools/versions.

vchong, 2020-05-20:
So in my case, cmake is building the TAs in alphabetical order, so `os_test` is built first before `os_test_lib`, hence the error `cannot find -los_test`. I managed to trick buildroot into building `os_test_lib` first by symlinking it to `nos_test_lib` under `out-br/build/optee_test-1.0/ta/`, so that made the error go away, but not really sure what the underlying problem is. : ( The build seemed fine for Qemu the last time I tried, though that might be a while ago.

vchong, 2020-05-20:
https://github.com/OP-TEE/optee_test/issues/307#issuecomment-542265329 and https://github.com/OP-TEE/build/pull/353 seem related.

closed 2020-06-25 by github-actions[bot]

jforissier, 2020-08-31:
@vchong tentative fix in https://github.com/OP-TEE/build/pull/437.