OP-TEE/build #431: OP-TEE Support for Qemu on Mingw

antonionehme, 2020-07-01:
Hi,

I wonder if OP-TEE has support for Qemu running over windows (MinGW).
My aim is to boot my versatile board through Qemu in a trusted environment (TrustZone).

Thanks!

jforissier, 2020-07-01:
Hi @antonionehme,

I have never tried working with OP-TEE with MinGW. While it might be possible in theory, you can expect to meet **many** issues before reaching a somewhat working state.
So, if your computer is running Windows natively, I would strongly recommend you boot into Linux VM and do everything from there. VMWare Player or Oracle VirtualBox with Ubuntu 20.04 should be super easy to setup and should work flawlessly.

antonionehme, 2020-07-01:
Many thanks for the quick response!

closed 2020-07-01 by antonionehme