OP-TEE/build #442: Why OP-TEE running on rpi3 is insecure?

FainBleeze, 2020-11-28:
Hello! I saw a disclaimer when reading the[ Device Specific  Information](https://optee.readthedocs.io/en/latest/building/devices/rpi3.html) of rpi3.
> Warning
> 
> This port of Trusted Firmware A and OP-TEE to Raspberry Pi 3 IS NOT SECURE! Although the Raspberry Pi3 processor provides ARM TrustZone exception states, the mechanisms and hardware required to implement secure boot, memory, peripherals or other secure functions are not available. Use of OP-TEE or TrustZone capabilities within this package does not result in a secure implementation. This package is provided solely for educational purposes and prototyping.

It seems that some mechanisms or hardware needed for ARM TrustZone are not implemented on rpi3. What does it refer to, memory，cache or something else?

Besides, since the CPU  architecture seems not the only thing needed for a secure implementation, how can I confirm if OP-TEE can run on my own board securely? For example, I'm now trying to port OP-TEE to Nvidia Jetson TK1. But its documentation doesn't mention anything about ARM TrustZone. Are there any tests I can do to confirm it?

gagachang, 2020-11-30:
ARM TrustZone requires some hardware controllers to ensure security of the system. One of the important controllers is TZASC (TrustZone Address Space Controller), which ensures DRAM isolation between normal world and secure world (non-secure memory / secure memory).

AFAIK, rpi3 doesn't have TrustZone hardware controllers including TZASC, so its memory isolation is not that secure. 

Below is the official documentation about TZASC from ARM:
https://developer.arm.com/documentation/ddi0431/c/introduction/about-the-tzasc

The TZASC driver in optee_os:
https://github.com/OP-TEE/optee_os/blob/master/core/drivers/tzc400.c

jenswi-linaro, 2020-11-30:
Skimming through https://developer.arm.com/documentation/genc009492/latest/ might help also.

closed 2020-11-30 by FainBleeze