OP-TEE/build #443: OP-TEE build for Hikey(Lemaker) error

SmallPond, 2020-12-01:
When I executed "make boot-img", an error occurred.

```
mformat -i ....../build/../out/boot-fat.uefi.img -n 64 -h 255 -T 131072 -v "BOOT IMG" -C ::

disk full

```
```
$ du -h ../linux/arch/arm64/boot/Image 
74M	../linux/arch/arm64/boot/Image
```

The size of boot-image（74MB） exceeds 64MB (131072 x 512 B), so I modified the statement as follows.
```
mformat -i $(BOOT_IMG) -n 64 -h 255 -T 198144 -v "BOOT IMG" -C ::
```

There is no problem with the compilation, but the image cannot be burned.

```
fastboot flash ptable /home/db/ATC21/build/../l-loader/ptable-linux-8g.img
< waiting for any device >
target reported max download size of 134217728 bytes
sending 'ptable' (17 KB)...
OKAY [  0.011s]
writing 'ptable'...
OKAY [  0.041s]
finished. total time: 0.053s
fastboot flash fastboot /home/db/ATC21/build/../trusted-firmware-a/build/hikey/release/fip.bin
target reported max download size of 134217728 bytes
sending 'fastboot' (1551 KB)...
OKAY [  0.199s]
writing 'fastboot'...
OKAY [  0.209s]
finished. total time: 0.409s
fastboot flash nvme /home/db/ATC21/build/../out/nvme.img
target reported max download size of 134217728 bytes
sending 'nvme' (128 KB)...
OKAY [  0.023s]
writing 'nvme'...
OKAY [  0.070s]
finished. total time: 0.093s
fastboot flash boot /home/db/ATC21/build/../out/boot-fat.uefi.img
target reported max download size of 134217728 bytes
sending 'boot' (99072 KB)...
OKAY [ 12.117s]
writing 'boot'...
FAILED (remote: Error flashing partition.)
finished. total time: 12.122s
hikey.mk:360: recipe for target 'flash' failed
make: *** [flash] Error 1

```

So if anyone here could help regarding this.

Thanks.

jbech-linaro, 2020-12-01:
Two comments:
- If you increase the size of the boot-image, then you must also make sure to change the partition table accordingly (see https://github.com/96boards-hikey/l-loader/blob/master/generate_ptable.sh#L73 for example).
- HiKey is in our CI/automatic testing and that seems to work as expected. That is a plain HiKey setup according to the instructions in the OP-TEE documentation.

SmallPond, 2020-12-01:
Thank you. @jbech-linaro

closed 2020-12-01 by SmallPond