OP-TEE/build #459: HiKey: build requires Python 2

jforissier, 2021-03-04:
The version of EDK2 included in the HiKey build requires Python 2, which is deprecated.
On a build host with Python 3 only:

```
$ make
[...]
make[2]: Entering directory '/home/jerome/work/optee_repo_hikey/edk2/BaseTools/Tests'
Traceback (most recent call last):
  File "RunTests.py", line 22, in <module>
    import TestTools
  File "/home/jerome/work/optee_repo_hikey/edk2/BaseTools/Tests/TestTools.py", line 94
    print description, '(base64 encoded):'
          ^
SyntaxError: invalid syntax
make[2]: *** [GNUmakefile:17: test] Error 1
make[2]: Leaving directory '/home/jerome/work/optee_repo_hikey/edk2/BaseTools/Tests'
make[1]: *** [GNUmakefile:25: Tests] Error 2
make[1]: Leaving directory '/home/jerome/work/optee_repo_hikey/edk2/BaseTools'
make: *** [Makefile:138: edk2] Error 2
```

jforissier, 2021-06-29:
I spent a few hours on this, trying to update EDK2, and I must admit I failed :-/

First, the Python 3 commits in upstream EDK2 are not completely trivial to identify. But more importantly the EDK2 HiKey/HiKey960 branch is very different from upstream (it is 3 years old and has not been rebased). So, backporting commits from upstream is very difficult. The other way around (applying HiKey/HiKey960 patches onto upstream) is equally hard. Not mentioning the crazyness^Wcomplexity of the EDK2 build system which also rules out another approach: try and fix issues as they appear.

So, my plan is: continue using the current code while possible, that is while Python 2 can still be run. Then perhaps switch to a more reasonable bootloader such as U-Boot (which apparently as received support for HiKey* recently: https://source.denx.de/u-boot/u-boot/-/tree/master/board/hisilicon).

jbech-linaro, 2021-06-29:
> Then perhaps switch to a more reasonable bootloader such as U-Boot (which apparently as received support for HiKey* recently: https://source.denx.de/u-boot/u-boot/-/tree/master/board/hisilicon).

You definitely get my +1 for that (and this might be useful as well https://github.com/jbech-linaro/u-boot-hikey ... but if there is official support, then maybe limited use the the stuff I did a long time ago).

jforissier, 2021-06-29:
Apparently there is HiKey/HiKey960 support in upstream https://github.com/tianocore/edk2-platforms (along with the main repository https://github.com/tianocore/edk2), and we already are using those Gits for FVP and Synquacer. So I will give that a try first.

MartinSchoenstedt, 2022-11-03:
I also came in contact with this issue on a new machine without python2. Maybe for now just add python2 to the prerequisites of OP-TEE or at least in the device specific information for the hikey board would be a temporary solution, as I searched quite some time until I found this open issue.

jforissier, 2022-11-03:
@MartinSchoenstedt I also learned that upstream EDK2 was lacking support for some stuff that are present in the fork (UFS I believe), at least it was the case back when I worked on this. So it could be more work than just a matter of Python2 vs Python3.

One other thing to consider: the OP-TEE CI loop used to run tests on HiKey, it doesn't anymore(new machine without Python2). I am in the process of introducing RockPi4B in the build environment (build.git / manifest.git) to replace it. That being said, I have no intention to remove `plat-hikey` from the `optee_os` source tree at the moment, and I will continue to run tests once in a while (at release time). So if you are targeting HiKey and can manage to deal with the Python2 issue, then you should be fine.

MartinSchoenstedt, 2022-11-03:
@jforissier 
> So if you are targeting HiKey and can manage to deal with the Python2 issue, then you should be fine.

For me, just installing python2 solved the issue. I did not change EDK2 from the op-tee fork at all or modified anything else.