OP-TEE/build #468: Error trying to build optee os for zcu106

tebina, 2021-04-19:
Hi,

I am trying to compile optee os for the zcu106 (PLATFORM=zcu106) and I get the error above.
I tried different optee os versions still doesn't work.
I am running ubuntu 16.04 x64 on vm.
Do you have any idea on what could be the issue ? 
Thank you.
```
DEBUG: Executing shell function do_compile
NOTE: make -j 8 comp-cflagscore=--sysroot=/home/tebina/petalinux-optee/zcu106-2018.2/build/tmp/work/aarch64-xilinx-linux/optee-os/2.4.0+gitAUTOINC+702609a7a5-r0/recipe-sysroot CROSS_COMPILE=aarch64-xilinx-linux- CROSS_COMPILE_core=aarch64-xilinx-linux- CROSS_COMPILE_ta_arm64=aarch64-xilinx-linux- PLATFORM=zynqmp-zcu106 CFG_ARM64_core=y CFG_ARM32_core=n CFG_USER_TA_TARGETS=ta_arm64 CFG_TEE_CORE_LOG_LEVEL=2 CFG_TEE_CORE_DEBUG=n DEBUG=0
  GEN     out/arm-plat-zynqmp/core/ta_pub_key.c
  CHK     out/arm-plat-zynqmp/conf.mk
  GEN     out/arm-plat-zynqmp/ta_arm32-lib/libutee/tui/default_bold.c
  GEN     out/arm-plat-zynqmp/ta_arm32-lib/libutee/tui/default_regular.c
  GEN     out/arm-plat-zynqmp/ta_arm64-lib/libutee/tui/default_bold.c
  UPD     out/arm-plat-zynqmp/conf.mk
[...]
  CC      out/arm-plat-zynqmp/core/arch/arm/kernel/thread.o
  CC      out/arm-plat-zynqmp/core/arch/arm/kernel/abort.o
  CC      out/arm-plat-zynqmp/core/arch/arm/kernel/vfp.o
In file included from core/arch/arm/kernel/thread.c:29:0:
core/arch/arm/plat-zynqmp/./platform_config.h:69:2: error: #error "Unknown platform flavor"
 #error "Unknown platform flavor"
  ^~~~~
  AS      out/arm-plat-zynqmp/core/arch/arm/kernel/vfp_a64.o
  CC      out/arm-plat-zynqmp/core/arch/arm/kernel/trace_ext.o
make: *** [mk/compile.mk:146: out/arm-plat-zynqmp/core/arch/arm/kernel/thread.o] Error 1
make: *** Waiting for unfinished jobs....
  AS      out/arm-plat-zynqmp/core/arch/arm/kernel/misc_a64.o
ERROR: oe_runmake failed
WARNING: exit code 1 from a shell command.
ERROR: Function failed: do_compile
```

jforissier, 2021-04-19:
Hi @tebina,

`zcu106` is unknown in the upstream code. Only `zcu102` is defined in `core/arch/arm/plat-zynqmp/conf.mk`.

tebina, 2021-04-19:
It appears to be the case. I was able to compile for the zcu102.
I also had an error after that ``` aarch64-xilinx-linux-ld.bfd: cannot find libgcc.a ``` but i was able to solve it using the compiler provided by the toolchain.mk.
Thank you for the fast reply and your help.

closed 2021-04-19 by tebina

tebina, 2021-04-20:
Hi @jforissier 
Is there a way to make it compile for the zcu106 ?

jforissier, 2021-04-20:
Perhaps someone who knows the differences between zcu102 and zcu106 can propose a patch.

tebina, 2021-04-20:
I've tried to compile for a zedboard as well since it is defined in ```core/arch/arm/plat-zynq7k/conf.mk```by using  ```PLATFORM=zynq7k-zc702```  and ```PLATFORM=zc702```. I am getting an error :
```
ERROR: Nothing PROVIDES 'virtual/optee-os' (but /home/tebina/petalinux-optee/zc702-2018.2/project-spec/meta-user/recipes-apps/optee-test/optee-test.bb, /home/tebina/petalinux-optee/zc702-2018.2/project-spec/meta-user/recipes-apps/optee-client/optee-client.bb DEPENDS on or otherwise requires it)
optee-os PROVIDES virtual/optee-os but was skipped: incompatible with machine zc702-zynq7 (not in COMPATIBLE_MACHINE)
```
Is the zedboard not supported as well ?

Thanks again I really appreciate your help

jforissier, 2021-04-20:
`optee_os` does build successfully with `make PLATFORM=zynq7k-zc702` so I suppose the error you are getting is related to your build environment (petalinux). I do not know this environment so I'm afraid I can't help further.