OP-TEE/build #476: Booting MCIMX8M-EVK from a rootfs on the SD card

JamesMenetrey, 2021-04-28:
Dear OP-TEE developers,

I have noticed that @jenswi-linaro added a build for MCIMX8M-EVK that works flawlessly! Many thanks to him!

I am trying to modify the [U-boot script file](https://github.com/OP-TEE/build/blob/master/imx/u-boot_boot_script) in order to boot from a RootFS that would be on the SD card instead of being a RAMDisk. My rudimentary patch is the removal of the two last lines:

```
fatload mmc ${mmcdev}:${mmcpart}  ${initrd_addr} ramdisk.img;
booti ${loadaddr} ${initrd_addr} ${fdt_addr};
```

To be replaced using the built-in function `run mmcboot;`, which seems [to be configured to boot on the second partition by default](https://github.com/Xilinx/u-boot-xlnx/blob/master/include/configs/imx8mq_evk.h). Afterward, I formatted the second partition (already created in the image resulting of the building process) as ext4 and copied the RootFS there. Basically, I executed the following steps:

```
$ sudo dd if=boot.img of=/dev/sdc bs=1M conv=fsync
$ sudo mkfs.ext4 /dev/sdc2
$ sudo mount /dev/sdc2 p2
$ cd p2
$ gunzip -cd ~/rootfs.cpio.gz | sudo cpio -idmv
```

The board is booting on Linux, but it seems that the driver of OP-TEE is not loaded (which causes the tee supplicant to complain, obviously). A log of the whole booting process can be observed [here](https://justpaste.it/1mgva).

I am sure I am not far from a working solution, but I fail to see the issue. From my understanding, the RootFS that is copied on the second partition is the one present in the RAMDisk file, so I suspect an issue with my patch.

As @jenswi-linaro worked on it, maybe you would have a better insight?

If you are interested to have this capability as part of the building process, I would submit a PR once solved.

Many thanks!
Cheers,
Jämes

jenswi-linaro, 2021-04-28:
Hi @JamesMenetrey,

It might be the something with the DTB, perhaps you're somehow getting one without the optee node added. Check if you have `/sys/firmware/devicetree/base/firmware/optee/` from the prompt.

JamesMenetrey, 2021-04-28:
Hey @jenswi-linaro,

Thanks for your prompt reply! You are right, the issue was bound to the DTB! The built-in `mmcboot` function has an internal call to `loadfdt` that overrides the provided DTB. As a result, one should patch the u-boot script by removing the previously mentioned lines:

```
fatload mmc ${mmcdev}:${mmcpart}  ${initrd_addr} ramdisk.img;
booti ${loadaddr} ${initrd_addr} ${fdt_addr};
```
And replace them by:
```
run mmcargs;
booti ${loadaddr} - ${fdt_addr};
```

where:
- `mmcargs` sets up the correct bootargs;
- `booti` removes the reference to the optional *initrd* argument.

I compared the results of `xtest` before and after that change and this leads to the same result, while maintaining a persistent RootFS. Therefore, I suppose it properly worked.

Regarding the PR, would it make sense to add a compilation flag such as *PERSISTENT_ROOTFS*, so the RootFS is then added as a second partition in the image?

I close the issue in the meanwhile.

Cheers

closed 2021-04-28 by JamesMenetrey

jenswi-linaro, 2021-04-28:
> Regarding the PR, would it make sense to add a compilation flag such as PERSISTENT_ROOTFS, so the RootFS is then added as a second partition in the image?

Sure, feel free to propose a PR.