OP-TEE/build #478: QEMU_PSS_AUTOMOUNT seems to be only working for uid=1000

cgellner, 2021-05-07:
It seems that enabling persistent TEE storage for qemu (QEMU_PSS_AUTOMOUNT=y) will only work properly for the primary user of the PC (having uid=1000).
When I try as a different user (uid!=1000) it doesn't work out of the box.

In order to reproduce, please try to switch to a user not having uid=1000 and run
```mkdir -p /tmp/qemu-data-tee && make QEMU_PSS_AUTOMOUNT=y run``` (as described in common.mk)
inside qemu userspace execute optee_example_secure_storage.
For me it simply gets stuck.
```
# optee_example_secure_storage 
Prepare session with the TA
```
It seems this is due to tee-supplicant not being started automatically

In startup I see below logs
```
Running sysctl: OK
Saving random seed: random: dd: uninitialized urandom read (512 bytes read)
OK
Set permissions on /dev/tee*: OK
Set permissions on /dev/ion: OK
Create/set permissions on /data/tee: chown: /data/tee: Operation not permitted
chown: /data/tee: Operation not permitted
FAIL
Starting network: OK
Starting network (udhcpc): OK

Welcome to Buildroot, type root or test to login
buildroot login: root
# pidof tee-supplicant
```
When using strace I can see EPERM error for chown using UID/GID 1000 on file-handle below /tmp/qemu-data-tee
```
...
29645 fchownat(16, ".", 1000, 1000, AT_SYMLINK_NOFOLLOW <unfinished ...>
...
29645 <... fchownat resumed> )          = -1 EPERM (Operation not permitted)
...
```
Note: 1000 is the default uid of "tee"
```
# id tee
uid=1000(tee) gid=1000(tee) groups=1000(tee)
```

When I manually start tee-supplicant daemon it seems to be working.
At least secure-storage example could be executed.

jforissier, 2021-05-26:
Hi @cgellner,

That's an interesting finding. Perhaps `security_model=mapped-xattr` would help here (see https://qemu.readthedocs.io/en/latest/system/invocation.html `-fsdev local,...` parameter)?

cgellner, 2021-05-27:
Hi @jforissier,

thank you for the feedback

I did a quick test changing security_model for QEMU_PSS... to mapped-xattr in common.mk.
Your proposal seems to work.
- tee-supplicant is started automatically
- optee_example_secure_storage is executed without any obvious issues
- xtest did pass

When I create a file /data/tee/foo as root from QEMU normal-world console I see the expected credentials (root/root)
When checking /tmp/qemu-data-tee I see the file is owned by my user.
Same is true for dirf.db and keystore entries - owned by tee/tee when looking from QEMU normal-world console.
Owned by my user when looking into /tmp/qemu-data-tee.

I didn't check if the modification did have any impact on the performance.
But this might not be really relevant for QEMU OP-TEE environment.

jforissier, 2021-05-27:
@cgellner I'm glad it worked. Your observations are consistent with what we could expect from the option. Would you mind creating a PR with this fix?

> I didn't check if the modification did have any impact on the performance.
> But this might not be really relevant for QEMU OP-TEE environment.

Agreed.

cgellner, 2021-05-27:
I will try to find the time to create the corresponding PR.
But it might take a few days.

@jforissier 
As you proposed the way to fix I would like to give you the credits in the commit.
Is there any commonly used tag e.g. "Proposed-by:"... for such a case ?
Any preference ?

jforissier, 2021-05-27:
That is fine. You may add: `Suggested-by: Jerome Forissier <jerome@forissier.org>`. Thanks!

closed 2021-06-25 by jforissier