OP-TEE/build #485: Running Qemu-v8 issue

saleh-unikie, 2021-05-28:
I tried to build and run the project for Qemu-v8 as was described here:
https://optee.readthedocs.io/en/latest/building/devices/qemu.html#qemu-v8

But during the build process, this error poped up:
`./brotli/c/common/constants.h: No such file or directory`

Then I found the sub modules of edk2 are not cloned and it needs to be cloned by a separate command:
cd edk2
git submodule update --init
described here: https://github.com/tianocore/edk2#submodules

so I assume the document should be updated to this instruction:
$ repo init -u https://github.com/OP-TEE/manifest.git -m qemu_v8.xml
$ repo sync
$ cd edk2
$ git submodule update --init
$ cd ../build
$ make toolchains
$ make run

jforissier, 2021-05-28:
Hi @saleh-unikie,

That is weird, this is supposed to be taken care of by the `sync-s="true"` attribute: https://github.com/OP-TEE/manifest/blob/3.13.0/qemu_v8.xml#L24.

jforissier, 2021-05-28:
Check `repo --version`, you might be running an older version that doesn't support `sync-s`? (I am speculating here).

saleh-unikie, 2021-05-31:
I installed it about two weeks ago, it is the result of repo --version:

... A new version of repo (2.15) is available.
... You should upgrade soon:
    cp /home/saleh/Projects/op-tee-v8/.repo/repo/repo /home/saleh/bin/repo

repo version v2.15.2
       (from https://gerrit.googlesource.com/git-repo)
       (tracking refs/heads/stable)
       (Thu, 20 May 2021 06:15:38 +0000)
repo launcher version 2.14
       (from /home/saleh/bin/repo)
       (currently at 2.15.2)
repo User-Agent git-repo/2.15.2 (Linux) git/2.25.1 Python/3.8.5
git 2.25.1
git User-Agent git/2.25.1 (Linux) git-repo/2.15.2
Python 3.8.5 (default, Jan 27 2021, 15:41:15) 
[GCC 9.3.0]
OS Linux 5.8.0-53-generic (#60~20.04.1-Ubuntu SMP Thu May 6 09:52:46 UTC 2021)
CPU x86_64 (x86_64)
Bug reports: https://bugs.chromium.org/p/gerrit/issues/entry?template=Repo+tool+issue

jforissier, 2021-05-31:
OK, you could upgrade as suggested (the `cp` command) but it is probably not the cause of the issue you are seeing. Maybe you had checked out the code before the EDK2 update and `repo` is not smart enough to detect it needs to update the submodules?

Can you please try `repo sync --fetch-submodules`?

saleh-unikie, 2021-05-31:
what I did is as follow:
I made a directory and project was fetched to build the Qemu-v7, it was successful and worked.

Then I made a new folder (outside the last one, so independent) and again I cloned the project for Qemu-v8, exactly as what described in the document (https://optee.readthedocs.io/en/latest/building/devices/qemu.html#qemu-v8), nothing extra. the only thing that I didn't, was installing the prerequisites and desired tools, because it was installed beforehand.

Anyway, I called `repo sync --fetch-submodules` now, and it is fetching the sub-modules of edk2.

saleh-unikie, 2021-05-31:
NOTE: `repo sync --fetch-submodules` failed with this message:

NotADirectoryError: [Errno 20] Not a directory: '../../../../../../.repo/projects/edk2.git/subprojects/ArmPkg/Library/ArmSoftFloatLib/berkeley-softfloat-3.git/logs' -> '/home/saleh/Projects/op-tee-v8/edk2/ArmPkg/Library/ArmSoftFloatLib/berkeley-softfloat-3/.git/logs'

jforissier, 2021-05-31:
Yeah just got the same message... run `rm -rf edk2` and try again. Same for any other project that would cause a similar error (`qemu`).

jenswi-linaro, 2021-06-02:
Just a few days ago stuff worked well at my end, but now it seems that something has changed and I need to try quite a few times before I finally get a full checkout.

jenswi-linaro, 2021-06-02:
Looking at https://gerrit.googlesource.com/git-repo/ it there's indeed some problem with repo and submodules.
I'm testing a bit to see what works.

jenswi-linaro, 2021-06-03:
Another update:
I found the problem I was experiencing. I expect that everyone with Ubuntu and possibly Debian has the same problem. Git cannot clone from git.qemu.org.

``` bash
git clone https://git.qemu.org/git/qemu.git
Cloning into 'qemu'...
fatal: unable to access 'https://git.qemu.org/git/qemu.git/': server certificate verification failed. CAfile: none CRLfile: none
zsh: exit 128   git clone https://git.qemu.org/git/qemu.git
```
Git on Ubuntu is compiled to use gnutls for the ssl stuff and that doesn't work with git.qemu.org at the moment. The only workaround I could find was to build git on my machine configured to use openssl instead. Apparently it has something to do with in which order the certificates are presented from the server and gnutls is very picky about that order.

jforissier, 2021-06-03:
@jenswi-linaro hopefully "they" will address this issue on the server side...

There is a very simple workaround: use the git protocol instead:
```
$ git clone git://git.qemu.org/qemu.git
Cloning into 'qemu'...
remote: Counting objects: 566705, done.
remote: Compressing objects: 100% (99865/99865), done.
remote: Total 566705 (delta 467703), reused 563800 (delta 465210)
Receiving objects: 100% (566705/566705), 206.30 MiB | 26.26 MiB/s, done.
Resolving deltas: 100% (467703/467703), done.
```

jenswi-linaro, 2021-06-03:
OK, how do you convince the submodules in qemu to do that?

jforissier, 2021-06-03:
> OK, how do you convince the submodules in qemu to do that?

Good question :) It seems the submodules do not cause any problem.

```
$ git clone --recurse-submodules git://git.qemu.org/qemu.git
[...]
$ cd qemu
$ git submodule status
 f8b1b833015a4ae47110ed068e0deb7106ced66d capstone (4.0.1-548-gf8b1b833)
 85e5d839847af54efab170f2b1331b2a6421e647 dtc (v1.6.0-4-g85e5d83)
 776acd2a805c9b42b4f0375150977df42130317f meson (0.55.3)
 90c488d5f4a407342247b9ea869df1c2d9c8e266 roms/QemuMacDrivers (heads/master)
 33a7322de13e9dca4b38851a345a58d37e7a441d roms/SLOF (qemu-slof-20190703-81-g33a7322)
 06dc822d045c2bb42e497487935485302486e151 roms/edk2 (edk2-stable201903-2377-g06dc822d04)
 4bd064de239dab2426b31c9789a1f4d78087dc63 roms/ipxe (v1.20.1-70-g4bd064de)
 4a0041107b8ef77e0e8337bfcb5f8078887261a7 roms/openbios (heads/master)
 234ed8e427f4d92903123199f6590d144e0d9351 roms/opensbi (v0.9)
 a5300c4949b8d4de2d34bedfaed66793f48ec948 roms/qboot (heads/master)
 bf0e13698872450164fa7040da36a95d2d4b326f roms/qemu-palcode (heads/master)
 155821a1990b6de78dde5f98fa5ab90e802021e0 roms/seabios (rel-1.14.0)
 73b740f77190643b2ada5ee97a9a108c6ef2a37b roms/seabios-hppa (hppa-qemu-5.2-2-6-g73b740f7)
 cbaee52287e5f32373181cff50a00b6c4ac9015a roms/sgabios (cbaee52)
 3a6fdede6ce117facec0108afe716cf5d0472c3f roms/skiboot (v6.4)
 b46dd116ce03e235f2a7d4843c6278e1da44b5e1 roms/u-boot (v2021.04)
 60b3916f33e617a815973c5a6df77055b2e3a588 roms/u-boot-sam460ex (heads/master)
 0c37a43527f0ee2b9584e7fb2fdc805e902635ac roms/vbootrom (heads/master)
 8f43a99191afb47ca3f3c6972f6306209f367ece slirp (v4.2.0-26-g8f43a99)
 b64af41c3276f97f0e181920400ee056b9c88037 tests/fp/berkeley-softfloat-3 (heads/master)
 5a59dcec19327396a011a17fd924aed4fec416b3 tests/fp/berkeley-testfloat-3 (heads/master)
 6119e6e19a050df847418de7babe5166779955e4 ui/keycodemapdb (6119e6e)
```

jforissier, 2021-06-04:
Another workaround:
```
git clone -c http.sslverify=false --recurse-submodules https://git.qemu.org/git/qemu.git
```

Not applicable to manifest.git, so we should probably update to use `git://`.

jforissier, 2021-06-04:
@jenswi-linaro 

> Another update:
> I found the problem I was experiencing. I expect that everyone with Ubuntu and possibly Debian has the same problem. Git cannot clone from git.qemu.org.
> 
> ```shell
> git clone https://git.qemu.org/git/qemu.git
> ```

I can't see qemu being cloned from `git.qemu.org` in our [manifests](https://github.com/OP-TEE/manifest) (it comes from `github.com`). So in the end it seems nothing needs to be fixed on our side, right?

jenswi-linaro, 2021-06-04:
Yes, after looking even further into this it seems to have been a problem with QEMU v5.1.0. So nothing we need to do.

When building QEMU v6.0.0 it says:
```
Using './build' as the directory for build output
Submodule 'capstone' (https://gitlab.com/qemu-project/capstone.git) registered for path 'capstone'
Submodule 'dtc' (https://gitlab.com/qemu-project/dtc.git) registered for path 'dtc'
```

But QEMU v5.1.0 starts those urls with https://git.qemu.com/ instead. I was using some old branch of the manifest.

saleh-unikie, 2021-06-04:
> rm -rf edk2

> Yeah just got the same message... run `rm -rf edk2` and try again. Same for any other project that would cause a similar error (`qemu`).

sorry for the late answer, I tried it and it worked for me. it cloned the sub-modules.
Anyway without running this extra command and only following the official document, building the Qemu-v8 was unsuccessful.

closed 2021-07-11 by github-actions[bot]