OP-TEE/build #486: Toolchain deploy process improvements

Aleksey-kk, 2021-06-04:
There is no clear indication in case of toolchain download and unpack errors. Instead of this build processes goes on and fails later with unclear error output.

Could you make deploy process more reliable and predictable. Please see patch prototype:

```
diff --git a/toolchain.mk b/toolchain.mk
index df85216..129fcb9 100644
--- a/toolchain.mk
+++ b/toolchain.mk
@@ -22,8 +22,9 @@ define dltc
        @if [ ! -d "$(1)" ]; then \
                mkdir -p $(1); \
                echo "Downloading $(3) ..."; \
-               curl -s -L $(2) -o $(TOOLCHAIN_ROOT)/$(3).tar.xz; \
-               tar xf $(TOOLCHAIN_ROOT)/$(3).tar.xz -C $(1) --strip-components=1; \
+               curl --retry-all-errors --retry 128 -s -L $(2) -o $(TOOLCHAIN_ROOT)/$(3).tar.xz; \
+               (tar xf $(TOOLCHAIN_ROOT)/$(3).tar.xz -C $(1) --strip-components=1 ||
+               (rm $(TOOLCHAIN_ROOT)/$(3).tar.xz; exit 1)) \
                (cd $(1)/bin && for f in *-none-linux*; do ln -s $$f $${f//-none} ; done;) \
        fi
 endef
```

Note: it was not tested. This patch is only demonstration of this suggestion idea.

jforissier, 2021-06-04:
Hi @Aleksey-kk,

I think it is a good idea.

```
diff --git a/toolchain.mk b/toolchain.mk
index df85216..129fcb9 100644
--- a/toolchain.mk
+++ b/toolchain.mk
@@ -22,8 +22,9 @@ define dltc
        @if [ ! -d "$(1)" ]; then \
                mkdir -p $(1); \
                echo "Downloading $(3) ..."; \
-               curl -s -L $(2) -o $(TOOLCHAIN_ROOT)/$(3).tar.xz; \
-               tar xf $(TOOLCHAIN_ROOT)/$(3).tar.xz -C $(1) --strip-components=1; \
+               curl --retry-all-errors --retry 128 -s -L $(2) -o $(TOOLCHAIN_ROOT)/$(3).tar.xz; \
```
128 is a bit too much IMO. It could cause the command to take a very long time (for example if the server name is not resolvable or the server is not reachable at all). I think we should only try to deal with transient errors, `--retry 5` should be good enough.

```
+               (tar xf $(TOOLCHAIN_ROOT)/$(3).tar.xz -C $(1) --strip-components=1 ||
+               (rm $(TOOLCHAIN_ROOT)/$(3).tar.xz; exit 1)) \
```
I think `$(1)` needs to be removed too, or the next `make toolchains` will report nothing to do.
```
                (cd $(1)/bin && for f in *-none-linux*; do ln -s $$f $${f//-none} ; done;) \
        fi
 endef
```

Aleksey-kk, 2021-07-12:
Please reopen this issue.

jforissier, 2021-07-12:
PR welcome :)

closed 2021-07-18 by github-actions[bot]