OP-TEE/build #487: Build with GICv3 interrupt controller support for QEMU

Aleksey-kk, 2021-06-04:
For now project is built with GICv2 interrupt controller support by default.
QEMU, ARM TF-A and OP-TEE OS support working with GICv3.

Is it possible to built with v3 support by default for QEMU?

Here is tested repo diff output of change for ARMv8 and ARM qemu configuration:

```
project build/
diff --git a/qemu.mk b/qemu.mk
index 038e982..195a9e9 100644
--- a/qemu.mk
+++ b/qemu.mk
@@ -141,6 +141,7 @@ linux-cleaner: linux-cleaner-common
 ################################################################################
 # OP-TEE
 ################################################################################
+OPTEE_OS_COMMON_FLAGS += DEBUG=$(DEBUG) CFG_ARM_GICV3=y
 optee-os: optee-os-common
 optee-os-clean: optee-os-clean-common

@@ -176,6 +177,7 @@ run-only:
        -serial tcp:localhost:54320 -serial tcp:localhost:54321 \
        -smp $(QEMU_SMP) \
        -s -S -machine virt,secure=on -cpu cortex-a15 \
+       -machine gic-version=3,secure=on \
        -d unimp -semihosting-config enable,target=native \
        -m 1057 \
        -bios bl1.bin \
diff --git a/qemu_v8.mk b/qemu_v8.mk
index 8a7ee48..9044306 100644
--- a/qemu_v8.mk
+++ b/qemu_v8.mk
@@ -165,7 +165,7 @@ linux-cleaner: linux-cleaner-common
 ################################################################################
 # OP-TEE
 ################################################################################
-OPTEE_OS_COMMON_FLAGS += DEBUG=$(DEBUG)
+OPTEE_OS_COMMON_FLAGS += DEBUG=$(DEBUG) CFG_ARM_GICV3=y
 optee-os: optee-os-common

 optee-os-clean: optee-os-clean-common
@@ -202,6 +202,7 @@ run-only:
        -serial tcp:localhost:54320 -serial tcp:localhost:54321 \
        -smp $(QEMU_SMP) \
        -s -S -machine virt,secure=on -cpu cortex-a57 \
+       -machine gic-version=3,secure=on \
        -d unimp -semihosting-config enable,target=native \
        -m 1057 \
        -bios bl1.bin \

project trusted-firmware-a/
diff --git a/plat/qemu/qemu/platform.mk b/plat/qemu/qemu/platform.mk
index 1bf4e08..60859e6 100644
--- a/plat/qemu/qemu/platform.mk
+++ b/plat/qemu/qemu/platform.mk
@@ -5,7 +5,7 @@
 #

 # Use the GICv2 driver on QEMU by default
-QEMU_USE_GIC_DRIVER    := QEMU_GICV2
+QEMU_USE_GIC_DRIVER    := QEMU_GICV3

 ifeq (${ARM_ARCH_MAJOR},7)
 # ARMv7 Qemu support in trusted firmware expects the Cortex-A15 model.

```
For ARMv8 tests in /usr/bin/ passed normally with this change; ARM was not tested.

Aleksey-kk, 2021-07-12:
Please, reopen this issue.

jforissier, 2021-07-12:
> For now project is built with GICv2 interrupt controller support by default.
> QEMU, ARM TF-A and OP-TEE OS support working with GICv3.
> 
> Is it possible to built with v3 support by default for QEMU?

Personally, I don't see any issue with doing so.

> 
> Here is tested repo diff output of change for ARMv8 and ARM qemu configuration:

We would need PRs in the relevant repositories.

> ```
> project build/
> diff --git a/qemu.mk b/qemu.mk
> index 038e982..195a9e9 100644
> --- a/qemu.mk
> +++ b/qemu.mk
> @@ -141,6 +141,7 @@ linux-cleaner: linux-cleaner-common [...]

Why not make this the default in `optee_os`? And same for v8.

> ```
>  optee-os: optee-os-common
>  optee-os-clean: optee-os-clean-common
> 
> @@ -176,6 +177,7 @@ run-only:
>         -serial tcp:localhost:54320 -serial tcp:localhost:54321 \
>         -smp $(QEMU_SMP) \
>         -s -S -machine virt,secure=on -cpu cortex-a15 \
> +       -machine gic-version=3,secure=on \
> ```

OK, similar change required in `qemu-check.exp`.

> ```
>         -d unimp -semihosting-config enable,target=native \
>         -m 1057 \
>         -bios bl1.bin \
> diff --git a/qemu_v8.mk b/qemu_v8.mk
> index 8a7ee48..9044306 100644
> --- a/qemu_v8.mk
> +++ [...]

closed 2021-07-18 by github-actions[bot]