OP-TEE/build #489: Problem with building OPTEE with benchmark

cti0ncep, 2021-06-11:
Hello,

I built a fresh copy of OPTEE with repo to test benchmark. By modifiying `common.mk` and set `CFG_TEE_BENCHMARK=y`, the building process stopps with error:
```
optee_benchmark_ext 1.0 Syncing from source dir
ERROR: does not exist
package/pkg-generic.mk:185: recipe for target '/optee/out-br/build/optee_benchmark_ext-1.0/.stamp_rsynced' failed
make[2]: *** [/optee/out-br/build/optee_benchmark_ext-1.0/.stamp_rsynced] Error 1
Makefile:23: recipe for target '_all' failed
make[1]: *** [_all] Error 2
make[1]: Leaving directory '/optee/out-br'
common.mk:289: recipe for target 'buildroot' failed
make: *** [buildroot] Error 2
```

I checked the folder `/optee/out-br/build/optee_benchmark_ext-1.0` is empty.
The same error is showed also with `make CFG_TEE_BENCHMARK=y`.

optee_benchmark repository info: _commit 875be7f1959f19ed601ef37501f1cf0bfbee9da4 (HEAD, tag: 3.13.0-rc1, tag: 3.13.0, tag: 3.12.0-rc1, tag: 3.12.0, m/master, github/master)_.

Any idea of the problem?

cti0ncep, 2021-06-14:
Update: Tried to build benchmark separately (like [2551](https://github.com/OP-TEE/optee_os/issues/2551)), but it is not a clean solution and causes other problems.
According to the [docs](https://optee.readthedocs.io/en/latest/building/gits/optee_benchmark.html), it should be enough to set a switch at compile time, but it doesn't work.

Any help appreciated.

jbech-linaro, 2021-06-14:
@cti0ncep , I'm able to reproduce the same error as you're seeing. I think what you've tried is the correct things to do, however, I'm afraid that the benchmark has bit rot and would need some attention again to get it back on track. I'm not sure who that person is, but meanwhile I'm marking this as "bug".

Sadiq1811, 2022-02-10:
@jbech-linaro i was able to solve this via https://github.com/OP-TEE/build/pull/553

jbech-linaro, 2022-02-11:
Thanks a lot @Sadiq1811 , closing this issue since the fix has been merged.

closed 2022-02-11 by jbech-linaro