OP-TEE/build #501: BL31 error when initializaing runtime service opteed_fast on zynqmp-zcu102 encryption disabled board rev1.1

RavenSpear, 2021-06-30:
Petalinux ver: 2021.1
Board: ZynqMP zcu102 rev1.1 encrytion disabled(It seems for rev1.1, old version of fsbl generate by petalinux 2018.2 is not available)
build: zynqmp.mk for 2020.2, the petalinux version check is disabled.

I followed the instruction for petalinux 2018.2: https://optee.readthedocs.io/en/latest/building/devices/zynqmp.html
For the new version of zynqmp.mk, BOOT.BIN is not generated after make, I generate it manually using the following command:
```
$petalinux-package --boot --u-boot --format BIN
```

I copied image.ub, BOOT.BIN and boot.scr(is this necessary?) to SD card FAT partition, using e2image to write the rootfs on ext4 partition.

power on and the following error is displayed: 
```
Xilinx Zynq MP First Stage Boot Loader 
Release 2021.1   Jun  6 2021  -  07:07:32
NOTICE:  ATF running on XCZU9EG/silicon v4/RTL5.1 at 0x1000
WARNING: BL31: invalid exception level (3)
NOTICE:  BL31: v2.4(release):v1.1-7609-g851523ea2
NOTICE:  BL31: Built : 08:27:07, Apr 28 2021
INFO:    ARM GICv2 driver initialized
INFO:    BL31: Initializing runtime services
INFO:    BL31: PM Service Init Complete: API v1.1
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x10080000
INFO:    SPSR = 0x3c9
```
uboot and linux loaded successfully after this error.
Could you please provide any suggestion? thanks in advance.

RavenSpear, 2021-07-06:
This issue is fixed as I used petalinux-v2020.2 to regenerate the whole images dir, and BOOT.bin works well.

RavenSpear, 2021-07-06:
The following command is wrong:
```
$petalinux-package --boot --u-boot --format BIN
```
use the following command instead: 
```
petalinux-package --boot --pmufw --fpga --u-boot --add images/linux/tee.elf --cpu a53-0 --file-attribute "load=0x60000000, startup=0x60000000, exception_level=el-1, trustzone" --force
```

RavenSpear, 2021-07-06:
However, I encountered a new issue when initializing BL32(OPTEE):
```
Xilinx Zynq MP First Stage Boot Loader 
Release 2020.2   Jul  1 2021  -  16:20:34
NOTICE:  ATF running on XCZU9EG/silicon v4/RTL5.1 at 0x1000
WARNING: BL31: invalid exception level (3)
NOTICE:  BL31: v2.2(release):xilinx_rebase_v2.2_2020.1-10-ge6eea88b1
NOTICE:  BL31: Built : 16:12:37, Jul  1 2021
INFO:    ARM GICv2 driver initialized
INFO:    BL31: Initializing runtime services
INFO:    BL31: PM Service Init Complete: API v1.1
INFO:    BL31: Initializing BL32
I/TC: 
I/TC: OP-TEE version: 3.11.0-dev (gcc version 9.2.0 (GCC)) #1 Thu Jul  1 16:13:13 UTC 2021 aarch64
I/TC: Primary CPU initializing
E/TC:0 0 
E/TC:0 0 Core undef-abort at address 0x60003d34
E/TC:0 0  esr 0x02000000  ttbr0 0x60086000   ttbr1 0x00000000   cidr 0x0
E/TC:0 0  cpu #0          cpsr 0x400003c4
E/TC:0 0  x0  0000000060078058 x1  00000000600780bc
E/TC:0 0  x2  0000000000000000 x3  0000008000000000
E/TC:0 0  x4  000000006005d768 x5  0000000000000040
E/TC:0 0  x6  0000000000000007 x7  00000000ffffffff
E/TC:0 0  x8  0000000060003f70 x9  0000000060078068
E/TC:0 0  x10 0000000000000000 x11 0000000000000000
E/TC:0 0  x12 000000006008d7a8 x13 000000000000000a
E/TC:0 0  x14 00000000ffffffff x15 0000000000000020
E/TC:0 0  x16 000000006002b3dc x17 0000000000000000
E/TC:0 0  x18 0000000000000000 x19 0000000000000007
E/TC:0 0  x20 0000000060078058 x21 000000006007807c
E/TC:0 0  x22 0000000000000001 x23 0000000060078058
E/TC:0 0  x24 0000000000000008 x25 0000000050000004
E/TC:0 0  x26 0000000000000000 x27 0000000000000000
E/TC:0 0  x28 0000000000000000 x29 000000006008d890
E/TC:0 0  x30 000000006000cf8c elr 0000000060003d34
E/TC:0 0  sp_el0 000000006008d890
E/TC:0 0 Call stack:
E/TC:0 0  0x0000000060003d34
E/TC:0 0 Panic '[abort] undefined abort (trap CPU)' at core/arch/arm/kernel/abort.c:464 <get_fault_type>
E/TC:0 0 TEE load address @ 0x60000000
E/TC:0 0 Call stack:
E/TC:0 0  0x000000006000b0b8
E/TC:0 0  0x00000000600196b8
E/TC:0 0  0x0000000060009a34
E/TC:0 0  0x0000000060006a28
```
This undefined abort occurs when optee is initializaing the optee service ```tee_crypt_init()```.
After debugging, it is found that the command ```sha256su0``` in src file ```sha256_armv8_ce_a64.S``` caused this issue.

RavenSpear, 2021-07-06:
According to ```ARM® Cortex®-A53 MPCore Processor Cryptography Extension Technical Reference Manual```
, ```sha256su0``` is a command provided by ARM Cryptography Extension. If this extension is disabled, it will throw an undefined exception. Intutively, since my board is an ```Encryption Disabled``` version, I guess Xilinx disables the CE on my ZCU102 board. To fix this issue, I compiled optee with ```CFG_CRYPTO_WITH_CE=n``` to avoid calling of CE commands, and optee runs well.

closed 2021-07-06 by RavenSpear

jenswi-linaro, 2021-07-06:
Thanks for reporting the solution.