OP-TEE/build #507: sign_encrypt.py of optee-os running failed using pycryptodomex==3.4.7 on ubuntu18.04.

carloscn, 2021-07-14:
sign_encrypt.py of optee-os running failed.

There is problem on rls page https://optee.readthedocs.io/en/latest/building/prerequisites.html, the rls page recommands apt install python3-pycryptodome. In ubuntu18.04, python3-pycryptodome is pycryptodomex=3.4.7 and is installed /usr/lib/python3/dist-packages path.  The 3.4.7 verison pycryptodomex can't be supported by sign_encrypt.py. As a follow pic:
[image]

I have used pip3 install pycryptodome=3.10.1, my host includes two version pycryptodome. But the pycryptodomex=3.4.7 is imported first by sign_encrypt.py.
[image]

Unless I use apt remove python3-pycryptodome. The 3.10.1 pycryptodome supported by sign_encrypt.py perfectly.

jforissier, 2021-07-14:
@carloscn thanks for sharing your findings. 18.04 is quite old now but it may be a good idea to add your comments to the OP-TEE build documentation. If you want to create a pull request.

carloscn, 2021-07-19:
> @carloscn thanks for sharing your findings. 18.04 is quite old now but it may be a good idea to add your comments to the OP-TEE build documentation. If you want to create a pull request.

OK. The PR : https://github.com/OP-TEE/optee_docs/pull/134

jbech-linaro, 2021-08-02:
Since I've merged the PR for this, I'm closing this particular issue. Thanks @carloscn !

closed 2021-08-02 by jbech-linaro