OP-TEE/build #509: qemu_v8: there is no hvmloader in xen

dakexiaoying, 2021-07-29:
I am a novice and have recently been learning the virtualization part of OP-TEE. 
I got the latest version of OP-TEE, set XEN_BOOT = y and build the whole project successfully. However,  when I tried to create the virtual machine, it failed. 
[image]
Do you know what might cause this problem? I can't find the answer on the Internet.

ruchi393, 2021-07-29:
What we use at our end guest.cfg which you can find in build/qemu_v8/xen folder on host :

Contents of guest.cfg given below - 
kernel="/mnt/host/linux/arch/arm64/boot/Image"
memory=128
vcpus=1
command="console=hvc0 earlycon=xenboot"
ramdisk = "/mnt/host/out-br-domu/images/rootfs.cpio.gz"
name="domu"
tee="optee"

Does this work for you ?

dakexiaoying, 2021-07-29:
Thank you for your reply!
Yes, it works.
So, is it normal that there is no 'boot' folder in 'usr/lib/xen' ?

ruchi393, 2021-07-30:
> Thank you for your reply!
> Yes, it works.
> So, is it normal that there is no 'boot' folder in 'usr/lib/xen' ?

Sorry, but I am not sure about this boot folder.

closed 2021-09-04 by github-actions[bot]