OP-TEE/build #511: Custom soc_term vs telnet

b49020, 2021-08-03:
OP-TEE build currently uses custom [soc_term](https://github.com/linaro-swg/soc_term) application to access two terminals corresponding to secure world and normal world. I think similar behaviour can be achieved using `telnet` as follows:

In Qemu cmdline use:

```
-serial telnet:localhost:54320,server,nowait \
-serial telnet:localhost:54321,server,nowait \
```

And while launching new terminals, we can simply execute:

```
$ telnet localhost 54320
$ telnet localhost 54321
```
Thoughts?

jforissier, 2021-08-03:
I think it's a good idea. I remember trying `telnet` out of curiosity but I didn't know about the QEMU `telnet:` option so that didn't work...

b49020, 2021-08-03:
Thanks @jforissier, I will create a PR for this feature.

b49020, 2021-08-03:
https://github.com/OP-TEE/build/pull/512

closed 2021-09-09 by github-actions[bot]