OP-TEE/build #513: Building OP-TEE 3.12 for QEMU fails during linking

Hussain1811, 2021-08-04:
Below are the logs:
/home/saddy/project/optee_qemu/build/../toolchains/aarch64/bin/aarch64-linux-gnu-ld: drivers/phy/rockchip/phy-rockchip-typec.o: bad reloc symbol index (0xff1a1a1a >= 0x7a) for offset 0xff1a1a1aff1a1a1a in section `.debug_ranges'
/home/saddy/project/optee_qemu/build/../toolchains/aarch64/bin/aarch64-linux-gnu-ld: drivers/phy/rockchip/phy-rockchip-typec.o: bad reloc symbol index (0xff1a1a1a >= 0x7a) for offset 0xff1a1a1aff1a1a1a in section `.debug_ranges'
/home/saddy/project/optee_qemu/build/../toolchains/aarch64/bin/aarch64-linux-gnu-ld: final link failed: bad value
Makefile:1162: recipe for target 'vmlinux' failed
make[1]: *** [vmlinux] Error 1
make[1]: Leaving directory '/home/saddy/project/optee_qemu/linux'
common.mk:326: recipe for target 'linux-common' failed
make: *** [linux-common] Error 2

I followed all the steps mentioned in https://optee.readthedocs.io/en/latest/building/devices/qemu.html
Build Host: ubuntu 18.04 on virtual box 6.1.26
I am currently building for OP-TEE 3.12.0 for QEMU

Hussain1811, 2021-08-04:
could i please get some help w.r.t this aspect?

jforissier, 2021-08-04:
Hi @Hussain1811,

I have never seen this error. Maybe try a different, more recent toolchain?

Hussain1811, 2021-08-04:
i just used "make -j2 toolchains"
So, you mean i should try with some latest toolchain?

jforissier, 2021-08-04:
Which toolchain version are you using?

>  Build Host: ubuntu 18.04 on virtual box 6.1.26

This is a bit old now, I would recommend upgrading to 20.04 or 21.04 especially since you are in a VM so that's very easy to do.

jbech-linaro, 2021-08-04:
@Hussain1811, something is definitely wrong in your setup. Since reading this I've tried rebuild my own OP-TEE QEMU setup with the 3.12 tag. 

```
jyx@tuxedo ~/devel/optee_projects/qemu/linux [(HEAD detached at optee-3.12.0)]
$ git lg -1
* 436d15c681f4 - (HEAD, tag: optee-3.12.0) upstream-tee-subsys-patches.txt: fix wrong commit ID (7 months ago) <Jerome Forissier>
```

I didn't have any issues building it **and** my kernel log doesn't even mention the `phy-rockchip-typec.o` file at all (my build log can be found [here](http://ix.io/3uXJ)).
```
jyx@tuxedo ~/devel/optee_projects/qemu/build [(HEAD detached at 4a5b617) ?]
$ make -j16 linux 2>&1 | tee build-optee-3.12.0.log
...
jyx@tuxedo ~/devel/optee_projects/qemu/build [(HEAD detached at 4a5b617) ?]
$ echo $?
0
jyx@tuxedo ~/devel/optee_projects/qemu/build [(HEAD detached at 4a5b617) ?]
$ grep phy-rockchip-typec build-optee-3.12.0.log
```

It's also worth pointing out that our CI / regression testing does build QEMU several times every day and if there would have been an issue somewhere during the 3.12 tag, then I think that would have caught it.

> @jforissier: Which toolchain version are you using?
> 
> > Build Host: ubuntu 18.04

IBART still uses Ubuntu 18.04, but yes running something more recent is a good advice.

Hussain1811, 2021-08-04:
There is a user-interactive menu coming up which asks whether you want so and so feature enabled or not.
i gave 'Y' for all.
Maybe that could be causing the issue?

i had successfully built optee for qemu previously when such user-interactive menus weren't there (maybe it's some new addition in the build script?!?)

jbech-linaro, 2021-08-04:
> maybe it's some new addition in the build script?!?

To my knowledge this typically happens when moving between different version compiler/kernel version. We've recently stepped up gcc to 10.x from a 9.x. Is there a specific reason to stick to 3.12? If not, then I'd recommend to step up to the latest available (tracking master). If you still need a 3.12, then I'd recommend to check out a tagged/stable 3.12 setup (https://optee.readthedocs.io/en/latest/building/gits/build.html#stable-releases). With that you'll get a snapshot of when we did the 3.12 release where the all gits should place nicely together.

Mixing and matching new and old tags is always challenging, since there are always dependencies between gits, toolchains etc.

Hussain1811, 2021-08-07:
it was indeed an environment setup issue.
thanks a lot @jforissier  and @jbech-linaro  for the suggestions.

closed 2021-08-07 by Hussain1811