OP-TEE/build #531: Question: Compiling sources from a newly added folder to the build

Hussain1811, 2021-10-31:
Hello,

I have an existing optee qemu 3.4 (i have to use this particular optee version because of some internal use-case that i have) build working perfectly fine.
Now, i would like to add another folder (say "optee_test_extensions" which internally has some source files, CMakefiles, Makefiles) to the build.

I have added this folder path also the required makefile targets in common.mk
Then i have issued a "make run" which would eventually fire up the secure and normal world consoles.

But, my binary (say "xtest_ext") isn't available in /usr/bin of the RFS.
Maybe because my new folder isn't visible to the build system.

Please let me know if i am missing something here?

jforissier, 2021-10-31:
Hi, the build system for optee_test involves Buildroot. Perhaps you haven't properly defined the additional package(s) you want to build (BR2_PACKAGE_*)? Or you need to remove out-br and rebuild?

Hussain1811, 2021-11-01:
Thank you @jforissier 
So, without going in to the gory details, i just grepped for "BR2_PACKAGE_OPTEE_TEST" and mimicked the same settings for my new folder.
i even removed the out-br folder and built it again.

I still don't see my CA and .ta binaries in the RFS.

Files changed/added:
build/common.mk --> defined the paths to my new folder and also echoed configs to ../out-br/extra.conf
build/br-ext/configs/optee_test_extensions --> i have set `BR2_PACKAGE_OPTEE_TEST_EXTENSIONS=y` here
build/br-ext/package/optee_test_extensions --> contains the files Config.in and optee_test_extensions.mk

one observation: setting `OPTEE_TEST_EXTENSIONS_SITE_METHOD = local` in optee_test_extensions.mk results in the below error:
> *** optee_test_extensions has local site method, but `OPTEE_TEST_EXTENSIONS_SITE` is not defined.  Stop.

OPTEE_TEST_EXTENSIONS_SITE is assigned to BR2_PACKAGE_OPTEE_TEST_EXTENSIONS and BR2_PACKAGE_OPTEE_TEST_EXTENSIONS is configured as below:

```
config BR2_PACKAGE_OPTEE_TEST_EXTENSIONS
	bool "optee_test_extensions"
	select BR2_PACKAGE_OPTEE_CLIENT
	help
	  some help
```

jbech-linaro, 2021-11-01:
@Hussain1811 , did you source your new files from `br-ext/Config.in`?
```

Hussain1811, 2021-11-01:
oh, i had no clue about this. Thanks @jbech-linaro 
after sourcing; rm -rf out-br; make clean; make run......i  now get the below error:
```
/home/saddy/project/optee_qemu_3.4/build/br-ext/package/optee_test_extensions/Config.in:27: 'endif' in different file than 'if'
/home/saddy/project/optee_qemu_3.4/build/br-ext/package/optee_test_extensions/Config.in:9: location of the 'if'
/home/saddy/project/optee_qemu_3.4/out-br/build/.br2-external.in:13: 'endmenu' in different file than 'menu'
/home/saddy/project/optee_qemu_3.4/build/br-ext/package/optee_test_extensions/Config.in:9: location of the 'menu'
Makefile:929: recipe for target 'defconfig' failed
make[2]: *** [defconfig] Error 1
Makefile:79: recipe for target '_all' failed
make[1]: *** [_all] Error 2
make[1]: Leaving directory '/home/saddy/project/optee_qemu_3.4/buildroot'
common.mk:203: recipe for target 'buildroot' failed
make: *** [buildroot] Error 2
```

Hussain1811, 2021-11-01:
ok - i have fixed this issue.
This is because of a missing newline after the endif

Hussain1811, 2021-11-01:
i missed setting the ...SITE_METHOD variable in the .mk file. That was causing the issue.

@jforissier @jbech-linaro Just an off-topic question (if it's fine to ask in this issue). How are we compiling the optee_test TA's via cmake? i don't see a  `set (SRC .......)` sort of syntax in optee_test/ta/CMakeLists.txt

However, for optee_examples we indeed have this syntax.

jforissier, 2021-11-02:
@Hussain1811 the "regular" Makefiles are used for TAs. See https://github.com/OP-TEE/build/blob/3.15.0/br-ext/package/optee_test_ext/optee_test_ext.mk#L31-L38.

Hussain1811, 2021-11-02:
Thanks for the clarification

closed 2021-11-02 by Hussain1811