OP-TEE/build #534: Build for QEMU_v8 in a docker container causes errors: iasl not found

shantanoo-desai, 2021-11-14:
## Host Information

- `x86_64` Machine
- `Docker version 20.10.9, build c2ea9bc90b`

## Reproduction Steps

I am using @jbech-linaro's [docker-optee](https://github.com/jbech-linaro/docker-optee) Image which provides me the base image for trying out OP-TEE.

I did the following from within the container:
(current PWD is `build`)

- `make -j4 run all` which does produce any errors
- `make run` causes the following error `iasl not found`

### Error Logs

```
cp -f /home/optee/qemu-optee/edk2/Build/ArmVirtQemuKernel-AARCH64/RELEASE_GCC5/AARCH64/OvmfPkg/SmbiosPlatformDxe/SmbiosPlatformDxe/OUTPUT/SmbiosPlatformDxe.efi /home/optee/qemu-optee/edk2/Build/ArmVirtQemuKernel-AARCH64/RELEASE_GCC5/AARCH64/SmbiosPlatformDxe.efi
make[1]: Leaving directory '/home/optee/qemu-optee/edk2/Build/ArmVirtQemuKernel-AARCH64/RELEASE_GCC5/AARCH64/ArmVirtPkg/FdtClientDxe/FdtClientDxe'
"iasl"  -p/home/optee/qemu-optee/edk2/Build/ArmVirtQemuKernel-AARCH64/RELEASE_GCC5/AARCH64/MdeModulePkg/Universal/Disk/RamDiskDxe/RamDiskDxe/OUTPUT/./RamDisk.aml /home/optee/qemu-optee/edk2/Build/ArmVirtQemuKernel-AARCH64/RELEASE_GCC5/AARCH64/MdeModulePkg/Universal/Disk/RamDiskDxe/RamDiskDxe/OUTPUT/./RamDisk.iiii
/bin/sh: 1: iasl: not found
make[1]: *** [GNUmakefile:417: /home/optee/qemu-optee/edk2/Build/ArmVirtQemuKernel-AARCH64/RELEASE_GCC5/AARCH64/MdeModulePkg/Universal/Disk/RamDiskDxe/RamDiskDxe/OUTPUT/RamDisk.aml] Error 127
make[1]: Leaving directory '/home/optee/qemu-optee/edk2/Build/ArmVirtQemuKernel-AARCH64/RELEASE_GCC5/AARCH64/MdeModulePkg/Universal/Disk/RamDiskDxe/RamDiskDxe'
GenFfs -t EFI_FV_FILETYPE_DRIVER -g B601F8C4-43B7-4784-95B1-F4226CB40CEE -o /home/optee/qemu-optee/edk2/Build/ArmVirtQemuKernel-AARCH64/RELEASE_GCC5/FV/Ffs/B601F8C4-43B7-4784-95B1-F4226CB40CEERuntimeDxe/B601F8C4-43B7-4784-95B1-F4226CB40CEE.ffs -oi /home/optee/qemu-optee/edk2/Build/ArmVirtQemuKernel-AARCH64/RELEASE_GCC5/FV/Ffs/B601F8C4-43B7-4784-95B1-F4226CB40CEERuntimeDxe/B601F8C4-43B7-4784-95B1-F4226CB40CEESEC1.1.dpx -oi /home/optee/qemu-optee/edk2/Build/ArmVirtQemuKernel-AARCH64/RELEASE_GCC5/FV/Ffs/B601F8C4-43B7-4784-95B1-F4226CB40CEERuntimeDxe/B601F8C4-43B7-4784-95B1-F4226CB40CEESEC2.1.pe32 -oi /home/optee/qemu-optee/edk2/Build/ArmVirtQemuKernel-AARCH64/RELEASE_GCC5/FV/Ffs/B601F8C4-43B7-4784-95B1-F4226CB40CEERuntimeDxe/B601F8C4-43B7-4784-95B1-F4226CB40CEESEC3.ui
Building ... /home/optee/qemu-optee/edk2/OvmfPkg/AcpiPlatformDxe/QemuFwCfgAcpiPlatformDxe.inf [AARCH64]

build.py...
 : error 7000: Failed to execute command
	make tbuild [/home/optee/qemu-optee/edk2/Build/ArmVirtQemuKernel-AARCH64/RELEASE_GCC5/AARCH64/MdeModulePkg/Universal/Disk/RamDiskDxe/RamDiskDxe]

build.py...
 : error F002: Failed to build module
	/home/optee/qemu-optee/edk2/MdeModulePkg/Universal/Disk/RamDiskDxe/RamDiskDxe.inf [AARCH64, GCC5, RELEASE]

- Failed -
Build end time: 00:25:58, Nov.15 2021
Build total time: 00:00:44

make: *** [common.mk:410: edk2-common] Error 1
```

## Additional Information

it turns out when executing the following:

```bash
make run-only
```

which in turn mentions there is no `out/bin` directory.

I create the following directory `out/bin` in the root directory and it does run for a bit, which opens Two xterms on my host machine however it causes the following error:

```bash
ln -sf /home/optee/qemu-optee/build/../out-br/images/rootfs.cpio.gz /home/optee/qemu-optee/build/../out/bin/

* QEMU is now waiting to start the execution
* Start execution with either a 'c' followed by <enter> in the QEMU console or
* attach a debugger and continue from there.
*
* To run OP-TEE tests, use the xtest command in the 'Normal World' terminal
* Enter 'xtest -h' for help.

cd /home/optee/qemu-optee/build/../out/bin && /home/optee/qemu-optee/build/../qemu/build/aarch64-softmmu/qemu-system-aarch64 \
	-nographic \
	-serial tcp:localhost:54320 -serial tcp:localhost:54321 \
	-smp 2 \
	-s -S -machine virt,secure=on,gic-version=3,virtualization=false \
	-cpu cortex-a57 \
	-d unimp -semihosting-config enable=on,target=native \
	-m 1057 \
	-bios bl1.bin		\
	-initrd rootfs.cpio.gz \
	-kernel Image -no-acpi \
	-append 'console=ttyAMA0,38400 keep_bootcon root=/dev/vda2 ' \
	 \
	-object rng-random,filename=/dev/urandom,id=rng0 -device virtio-rng-pci,rng=rng0,max-bytes=1024,period=1000 -netdev user,id=vmnic -device virtio-net-device,netdev=vmnic
/bin/bash: line 1: /home/optee/qemu-optee/build/../qemu/build/aarch64-softmmu/qemu-system-aarch64: No such file or directory
make: *** [Makefile:391: run-only] Error 127
```

What could be the potential problem here?

jbech-linaro, 2021-11-15:
Hi @shantanoo-desai , it's been quite a while since I used my own docker image and I know that there has been changes to the packages needed to build OP-TEE. So, it's probably just the list of packages that needs to be updated in the Dockerfile. I think we can close the issue here, since this has nothing to do with `build.git`. That git works with the official tree's, my docker images are the culprit here.

shantanoo-desai, 2021-11-15:
Sounds good. For posterity, would you suggest I open the issue in the docker repository since it still is an issue which I am unable to troubleshoot?
@jbech-linaro

closed 2021-11-15 by shantanoo-desai

jbech-linaro, 2021-11-15:
> Sounds good. For posterity, would you suggest I open the issue in the docker repository since it still is an issue which I am unable to troubleshoot?
> @jbech-linaro

Sure.

jforissier, 2021-11-15:
@shantanoo-desai FYI:
```
$ dpkg -S `which iasl`
acpica-tools: /usr/bin/iasl
```
Try `apt install acpica-tools`, that should help.

jforissier, 2021-11-15:
And, note that the Docker image we use for CI (here: https://github.com/OP-TEE/optee_os/blob/3.15.0/.azure-pipelines.yml#L229) has this `acpica-tools` installed: https://github.com/jforissier/docker_optee_os_ci/blob/0b0a1cf/Dockerfile#L42 (this Dockerfile is in the `qemuv8_check` branch). You may want to check the package list, it should always be up to date. Note the Docker image itself is quite big though because it contains the whole OP-TEE tree pre-built to speed up CI, so it might not be a good idea to use this image for your purposes. In any case the package list should be helpful.