OP-TEE/build #536: Query to add Arm64 host as a test machine for OP-TEE Qemu CI

b49020, 2021-12-06:
Hi Team,

Would it be possible to add Arm64 host as a test machine for OP-TEE Qemu CI?

-Sumit

jbech-linaro, 2021-12-06:
> Would it be possible to add Arm64 host as a test machine for OP-TEE Qemu CI?

As QEMU v8? Yes, but that would bump the build time +25 minutes more. Some days recently there has been a long queue ([see the 2021-12-03 for example](https://optee.mooo.com:5000/)). Possible yes, but stretching the current machine doing testing.

b49020, 2021-12-06:
> As QEMU v8?

Actually I meant to build OP-TEE on Arm64 host along with x86 host. I guess it should be doable via replacing build machine for either hikey or Qemu targets. So it won't be another target to test to add on build time but rather a different build machine to be added that could run in parallel as well. You can consider it as another use-case of Arm-on-Arm development.

jbech-linaro, 2021-12-06:
I see, yes. I think that could be doable. We actually have a Developerbox in Lund. IBART as written today would require you to run that on another instance at a different port (if on the same IP), since it only supports a single host as the build environment. To actually get this done, I think we'd need to have a discussion internally, to figure out how to best proceed. What machine (Developbox)? Where (Lund office or?)? Who (has spare cycles)?

jforissier, 2021-12-06:
Another option would be to use Azure or similar assuming we can find an arm64 host. I could not find anything serious about arm64 support on Azure (only [this](https://azure.microsoft.com/en-us/updates/azure-devops-pipelines-introduces-support-for-linuxarm64) mention, no details).
On the other hand, GitHub Actions seems more promising: https://docs.github.com/en/actions/hosting-your-own-runners/about-self-hosted-runners#architectures. Perhaps we could translate the QEMUv8_check job to run on an Arm64 GitHub Actions runner?

b49020, 2021-12-06:
> Perhaps we could translate the QEMUv8_check job to run on an Arm64 GitHub Actions runner?

Yeah that should suffice the use-case that we support Arm64 host as a build machine as part of OP-TEE CI.

b49020, 2021-12-16:
So I gave OP-TEE build a try on my Developerbox running Ubuntu 20.04 and the build is stuck with below error:

```
<snip>
  CC      out/init.o
  CC      out/os_test.o
  CC      out/ta_entry.o
  CXX     out/cxx_tests.o
  CC      out/user_ta_header.o
  CPP     out/ta.lds
  LD      out/5b9e0e40-2636-11e1-ad9e-0002a5d5c51b.elf
/home/sumit/optee_br/build/../toolchains/aarch64/bin/aarch64-linux-ld.bfd: /home/sumit/optee_br/toolchains/aarch64/bin/../lib/gcc/aarch64-buildroot-linux-gnu/10.3.0/../../../../aarch64-buildroot-linux-gnu/lib/../lib64/libstdc++.a(cp-demangle.o): in function `d_append_num':
cp-demangle.c:(.text+0x830): undefined reference to `__sprintf_chk'
make[3]: *** [/home/sumit/optee_br/optee_os/out/arm/export-ta_arm64/mk/link.mk:117: out/5b9e0e40-2636-11e1-ad9e-0002a5d5c51b.elf] Error 1
make[2]: *** [package/pkg-generic.mk:271: /home/sumit/optee_br/out-br/build/optee_test_ext-1.0/.stamp_built] Error 2
make[1]: *** [Makefile:23: _all] Error 2
make[1]: Leaving directory '/home/sumit/optee_br/out-br'
make: *** [common.mk:324: buildroot] Error 2
```

It looks like cxx_tests are failing to build with buildroot toolchain, @jforissier any ideas?

b49020, 2021-12-16:
https://github.com/OP-TEE/build/pull/537

jforissier, 2021-12-16:
@b49020 what version of GCC is this? Could you try an older toolchain perhaps? My analysis is that by using a Linux/Glibc toolchain, using only the libstdc++ and our own libc (libutils), we are chasing a moving target. Here it seems we would need to implement `__sprintf_chk`, perhaps with a simple stub calling `sprintf` I don't know.

b49020, 2021-12-16:
> what version of GCC is this?

sumit@developerbox:~/optee_br/build$ ../toolchains/aarch64/bin/aarch64-buildroot-linux-gnu-gcc --version
aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot 2021.08) 10.3.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

> Could you try an older toolchain perhaps?

I think it is tied to buildroot version which we are currently using. So it requires us to stick with a particular buildroot version if it works with any older buildroot toolchain. Would that be preferred approach?

jforissier, 2021-12-16:
Ah yes, compiler is tied to buildroot. The interesting thing is that this 10.3.0 is problematic whereas the Arm cross compiler that we use which is also 10.x (10.2.1) doesn't have the problem.

The best option is perhaps to try and implement what's missing, that is the approach we have always been following.

jenswi-linaro, 2021-12-16:
> The best option is perhaps to try and implement what's missing, that is the approach we have always been following.

Yes, I believe we have cut some corners with the C++ support and I guess that's what has come back to bite us now.

b49020, 2021-12-17:
@jforissier 

> The best option is perhaps to try and implement what's missing, that is the approach we have always been following.
> 
> Edit: Or maybe we need to disable some compile time feature in order to not use the checking functions?

It looks like I am missing some background here. Can you share links to earlier related PR or issues?

jforissier, 2021-12-17:
> @jforissier
>
>
> It looks like I am missing some background here. Can you share links to earlier related PR or issues?

I mean C++ support was implemented by adding whatever the GCC C++ runtime has required from libc (libutils in OP-TEE) and the OS (the ldelf loader). For example if you search for `git log --grep=C++` you will find:

https://github.com/OP-TEE/optee_os/commit/eacabbbceb433cede5676d7b5869f20cfa4b9def
https://github.com/OP-TEE/optee_os/commit/36e784f621bf5d5be9183beba35f39426277c110
https://github.com/OP-TEE/optee_os/commit/7bc927fa5192f4af4f59072f38822e0082e4bddb
etc.

So following the same "philosophy" we could just add `__sprintf_chk()` to libutils, and see how it flies...

On the other hand (the second part of my reply), if it is enough to just add some compile flag to say "I do not want the checked versions of the functions" then it's easier, but I'm afraid it could be a toolchain compile-time issue.

b49020, 2021-12-21:
Thanks @jforissier for the background.

> So following the same "philosophy" we could just add __sprintf_chk() to libutils, and see how it flies...

I have tried it here: https://github.com/OP-TEE/optee_os/pull/5073, it gets rid of the error with xtest passing.

> On the other hand (the second part of my reply), if it is enough to just add some compile flag to say "I do not want the checked versions of the functions" then it's easier, but I'm afraid it could be a toolchain compile-time issue.

Indeed, its a toolchain compile time issue as we rely on `libstdc++.a` provided by toochain which has been built using `-fstack-protector-strong` as the default GCC flag for buildroot toolchain. Hence it requires `__sprintf_chk` to be implemented.

b49020, 2021-12-23:
> > Perhaps we could translate the QEMUv8_check job to run on an Arm64 GitHub Actions runner?
> 
> Yeah that should suffice the use-case that we support Arm64 host as a build machine as part of OP-TEE CI.

So [this](https://github.com/OP-TEE/optee_os/commit/b224894f4c344bd7340c51cf268e8797d8730fd4) fix has shown that OP-TEE build was already broken for Arm64 host. IMO, adding QEMUv8_check test on an Arm64 host in CI would ensure that we don't break Arm64 host support in future.

closed 2022-01-28 by github-actions[bot]