OP-TEE/build #547: Make recovery failing for HiKey 620

cgshep, 2022-01-17:
Hello OP-TEE team,

I'm attempting to run `make recovery` using the HiKey 620 target (`hikey.mk`), but the recovery binary is missing from l-loader in the OP-TEE build directory. 

The binaries from the [96boards support site](https://www.96boards.org/documentation/consumer/hikey/hikey620/installation/board-recovery.md.html) redirect to various "Page Not Found" errors, so I'm assuming that the issues are related.  

Is there an up-to-date repository where I can grab them?

I hope you can help and thanks in advance.

Makefile message:
```
...
/home/carlton/optee/build/../burn-boot/hisi-idt.py --img1=/home/carlton/optee/build/../l-loader/recovery.bin
+----------------------+
 Serial:  
 Image1:  /home/carlton/optee/build/../l-loader/recovery.bin
 Image2:  
+----------------------+

Image don't exists: /home/carlton/optee/build/../l-loader/recovery.bin
make: *** [Makefile:350: recovery] Error 1
```

jforissier, 2022-01-17:
Hi @cgshep,

I don't remember the details, but it looks like `recovery.bin` is a generated file generated from TF-A's `bl1.bin`. Try `make lloader`.

jbech-linaro, 2022-01-17:
Hi @cgshep ,

I you do a normal Hikey620 setup as explains in the [docs](https://optee.readthedocs.io/en/latest/building/devices/hikey620.html#build-instructions), then it *should* be possible to simply run `make recovery` and follow the instructions presented in the console (not that pins etc needs to be set correctly).

We're testing HiKey620 several times every day in [IBART](https://optee.mooo.com:5000/) and besides some [automated tricks to set the correct pins](https://github.com/jbech-linaro/ibart/blob/master/jobdefs/examples/hikey6220.yaml#L42-L48) for recovery mode, it uses such a standard setup and runs `make recovery` (like [here](https://github.com/jbech-linaro/ibart/blob/master/jobdefs/examples/hikey6220.yaml#L49-L61)).

cgshep, 2022-01-17:
Thanks, @jbech-linaro and @jforissier, for the quick replies. 

I solved the issue after finding `make lloader` failed during the build process because of the `python-is-python3` convenience package on my system. This was causing `gen_loader_hikey.py` to fail. I removed the package and it worked.

Thanks again!

closed 2022-01-17 by cgshep

jforissier, 2022-01-17:
> Thanks, @jbech-linaro and @jforissier, for the quick replies.
> 
> I solved the issue after finding `make lloader` failed during the build process because of the `python-is-python3` convenience package on my system. This was causing `gen_loader_hikey.py` to fail. I removed the package and it worked.

Oh, that means troubles ahead once Python 2 is not available :-/ Thanks for sharing the root cause.