OP-TEE/build #551: Conditional assignment of mandatory options

jenswi-linaro, 2022-02-02:
At quite a few places in the makefiles we have assignments like:
https://github.com/OP-TEE/build/blob/6a868e301304a7b3cdf8c10b684331bf81b6af1d/qemu.mk#L54-L66

Having these particular assignment conditional doesn't make much sense. Overriding `TF_A_FLAGS` here seems error prone.

I'm not sure these assignments:
https://github.com/OP-TEE/build/blob/6a868e301304a7b3cdf8c10b684331bf81b6af1d/qemu.mk#L22-L26
make much sense either. They are all tied closely to our build setup. Trying to override these default values must be asking for troubles.

Im my opinion it would make more sense to save conditional assignment (`?=`) to the case where it's useful to be able to override the default value. That way it would be more clear which variables are feasible to try to override and which shouldn't be touched really.

What does others think?

jbech-linaro, 2022-02-02:
First: paths: In theory you should be able to replace `TF_A_PATH` with another path to another copy of TF-A for example. But, I'm not sure that works in reality, since I'm pretty sure we have relative references in various places. Also, lines like 58, should leverage line 24 for example. Otherwise changing paths to other locations will for sure not work.

Secondly: `TF_A_FLAGS`: It'd be pretty tedious to write that separately on the command line, but if someone have a need to fully replacing it from a script or another makefile, why not?

I don't see an urgent issue that needs to be fixed. It's more that there are probably issues lurking if someone tries certain things. OTH no strong opinion.

jenswi-linaro, 2022-02-02:
> First: paths: In theory you should be able to replace `TF_A_PATH` with another path to another copy of TF-A for example. But, I'm not sure that works in reality, since I'm pretty sure we have [...]

Sure, these has good odds to work and my only need smaller tweaks to keep functional.

> 
> Secondly: `TF_A_FLAGS`: It'd be pretty tedious to write that separately on the command line, but if someone have a need to fully replacing it from a script or another makefile, why not?

Let's take a bit more complicated example of `TF_A_FLAGS`:
https://github.com/OP-TEE/build/blob/6a868e301304a7b3cdf8c10b684331bf81b6af1d/qemu_v8.mk#L153-L174

What happens if you try to override here? Not easy to use and may suddenly break.

> I don't see an urgent issue that needs to be fixed. It's more that there are probably issues lurking if someone tries certain things. OTH no strong opinion.

Agreed, this isn't urgent and it even works to a degree. But I would have preferred if there was an easy way of adding a few parameters for especially the build flags passed when building OP-TEE.

jbech-linaro, 2022-02-02:
> Let's take a bit more complicated example of `TF_A_FLAGS`:

Indeed, getting that right would be ... tricky 😃 

> But I would have preferred if there was an easy way of adding a few parameters for especially the build flags passed when building OP-TEE.

Yes, I'm certainly not disagreeing on that. The fact is that `build.git` is solely used for our setup. I cannot imagine that companies making products with OP-TEE actually use `build.git` other than potentially individual developers. So, I guess that the ability to use "other" locations for gits is pretty rare, i.e., what you said above. So, the key question would be to make sure it's easy to enable and disable various features. Could you throw something together just as an example of your thinking? Doesn't have to be a complete and working example, just something that shows the idea.

jforissier, 2022-02-02:
@jbech-linaro 
> So, the key question would be to make sure it's easy to enable and disable various features.

@jenswi-linaro 
> But I would have preferred if there was an easy way of adding a few parameters for especially the build flags passed when building OP-TEE.

I'm with you on both points. Perhaps use `:=` rather than `?=` and add some `$(xyz_additional_flags)` variable here and there?

jenswi-linaro, 2022-02-02:
Here's one attempt at the OP-TEE OS CFG flags. It turned out a bit different compared to what I was thinking of when I first created this issue.
https://github.com/OP-TEE/build/pull/552

jbech-linaro, 2022-02-02:
> Here's one attempt at the OP-TEE OS CFG flags. It turned out a bit different compared to what I was thinking of when I first created this issue.
> #552

Can't see what would wrong with something like that.

closed 2022-03-10 by github-actions[bot]