OP-TEE/build #555: host-flex 2.6.4 Downloading

AiNotArtificialIntelligence, 2022-03-23:
I used qemu to build the optee environment, and I put "XEN_BOOT? = n" Changed to "XEN_BOOT? = y" in build/Makefile to support XEN, but it gets stuck  “host-flex 2.6.4 Downloading” every time. I have tried make clean and other methods, but they are useless. Can someone help me, please?
The following is a partial log.

>>> host-gettext  Extracting
mkdir -p /home/wfs/qiubiye/out-br/per-package/host-gettext/host
mkdir -p /home/wfs/qiubiye/out-br/per-package/host-gettext/target
>>> host-gettext  Patching
>>> host-gettext  Configuring
mkdir -p /home/wfs/qiubiye/out-br/per-package/host-gettext/host
rsync -a --link-dest=/home/wfs/qiubiye/out-br/per-package/host-ccache/host/ /home/wfs/qiubiye/out-br/per-package/host-ccache/host/ /home/wfs/qiubiye/out-br/per-package/host-gettext/host
rsync -a --link-dest=/home/wfs/qiubiye/out-br/per-package/host-gettext-tiny/host/ /home/wfs/qiubiye/out-br/per-package/host-gettext-tiny/host/ /home/wfs/qiubiye/out-br/per-package/host-gettext/host
rsync -a --link-dest=/home/wfs/qiubiye/out-br/per-package/host-skeleton/host/ /home/wfs/qiubiye/out-br/per-package/host-skeleton/host/ /home/wfs/qiubiye/out-br/per-package/host-gettext/host
mkdir -p /home/wfs/qiubiye/out-br/per-package/host-gettext/target
rsync -a --link-dest=/home/wfs/qiubiye/out-br/per-package/host-ccache/target/ /home/wfs/qiubiye/out-br/per-package/host-ccache/target/ /home/wfs/qiubiye/out-br/per-package/host-gettext/target
rsync -a --link-dest=/home/wfs/qiubiye/out-br/per-package/host-gettext-tiny/target/ /home/wfs/qiubiye/out-br/per-package/host-gettext-tiny/target/ /home/wfs/qiubiye/out-br/per-package/host-gettext/target
rsync -a --link-dest=/home/wfs/qiubiye/out-br/per-package/host-skeleton/target/ /home/wfs/qiubiye/out-br/per-package/host-skeleton/target/ /home/wfs/qiubiye/out-br/per-package/host-gettext/target
>>> host-gettext  Building
>>> host-gettext  Installing to host directory
mkdir -p /home/wfs/qiubiye/out-br/per-package/host-flex/host
mkdir -p /home/wfs/qiubiye/out-br/per-package/host-flex/target
>>> host-flex 2.6.4 Downloading

jforissier, 2022-03-23:
Hello @AiNotArtificialIntelligence,

I cannot reproduce the error here. It looks like you are having some network access issue. Can you please try rebuilding with `BR2_PER_PACKAGE_DIRECTORIES=n`? This will disable top-level parallel build so that we can make sure which package is causing troubles (flex it seems).
```
build$ rm -rf ../out-br
build$ make XEN_BOOT=y BR2_PER_PACKAGE_DIRECTORIES=n
```

AiNotArtificialIntelligence, 2022-03-23:
> Hello @AiNotArtificialIntelligence,
>
> I cannot reproduce the error here. It looks like you are having some network access issue. Can you please try rebuilding with `BR2_PER_PACKAGE_DIRECTORIES=n`? [...]
Oh, I tried. It wasn't the build problem. It was a network error in my terminal. I didn't have such a problem when I executed make on a new terminal. Thank you！

jforissier, 2022-03-23:
OK, good, then I'm closing this issue.

closed 2022-03-23 by jforissier